espressif/esp-idf #7224: ULP RISCV unexpectedly wakes main CPU with ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG

bbrother, 2021-07-02:
## Environment

- Development Kit:      SP32-S2-Kaluga-1 Kit v1.3
- Kit version (for WroverKit/PicoKit/DevKitC): v1.3
- Module or chip used:  ESP32-S2-WROVER
- IDF version: v4.3 and v4.3-194-g5b5e46971-dirty
- Build System:     idf.py
- Compiler version: 8.4.0
- Operating System:     Linux (using v4.3 docker container)
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
I am trying to get a ULP example working where the ULP runs on a wakeup period and then wakes up the main CPU when certain conditions are satisfied. If you wait long enough ~10-100 seconds without a wakeup of the main processor eventually there is a a ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG

### Expected Behavior
The ULP RISCV can run for an indefinite amount of time without waking up the app processor

### Actual Behavior
The main processor wakes unexpectedly with a ESP_SLEEP_WAKEUP_COCPU_TRAP_TRIG
### Steps to reproduce

1. start with ulp_riscv example code
2. Replace ulp_riscv_example_main.c with code from gist
3. Replace ulp/main.c with code from gist

### Code to reproduce this issue
https://gist.github.com/bbrother/58b871e9861b2a443e9a94292d8aabc6

## Debug Logs

```
Running idf_monitor in directory /project                                                                                                                                                                   
Executing "/opt/esp/python_env/idf4.3_py3.6_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p /dev/ttyUSB1 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- /project/build/ulp-riscv-example.elf -m '/opt/
esp/python_env/idf4.3_py3.6_env/bin/python' '/opt/esp/idf/tools/idf.py'"...                                                                                                                                 
--- idf_monitor on /dev/ttyUSB1 115200 ---                                                                                                                                                                  
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---                                                                                                                                       
ESP-ROM:esp32s2-rc4-20191025                                                                                                                                                                                
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)                                                                                                                                                            
SPIWP:0xee                                                                                                                                                                                                  
mode:DIO, clock div:1                                                                                                                                                                                       
load:0x3ffe6100,len:0x12a8                                                                                                                                                                                  
load:0x4004c000,len:0x8f0                                                                                                                                                                                   
load:0x40050000,len:0x2984                                                                                                                                                                                  
entry 0x4004c1b0                                                                                                                                                                                            
W (72) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.                                                                   
Wakeup info 00000000 00000000 00000000 00000000 00000000                                                                                                                                                    
Not a ULP-RISC-V wakeup, initializing it! 0000 -1614824625 44A480B7                                                                                                                                         
Entering in deep sleep 549881                                                                                                                                                                               
                                                                                                                                                                                                            
ESP-ROM:esp32s2-rc4-20191025                                                                                                                                                                                
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x5 (DSLEEP),boot:0xb (SPI_FAST_FLASH_BOOT)                                                                                                                                                             
SPIWP:0xee                                                                                                                                                                                                  
mode:DIO, clock div:1                                                                                                                                                                                       
load:0x3ffe6100,len:0x12a8                                                                                                                                                                                  
load:0x4004c000,len:0x8f0                                                                                                                                                                                   
load:0x40050000,len:0x2984                                                                                                                                                                                  
entry 0x4004c1b0                                                                                                                                                                                            
W (70) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.                                                                   
Wakeup info 00100000 00000000 00000000 00000800 00000000                                                                                                                                                    
ULP-RISC-V woke up the main CPU! 0006 103 00000FEC test 1                                                                                                                                                   
Entering in deep sleep 3256675                                                                                                                                                                              
                                                                                                                                                                                                            
ESP-ROM:esp32s2-rc4-20191025                                                                                                                                                                                
Build:Oct 25 2019                                                                                                                                                                                           
rst:0x5 (DSLEEP),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x12a8
load:0x4004c000,len:0x8f0
load:0x40050000,len:0x2984
entry 0x4004c1b0
W (71) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Wakeup info 00100000 00000000 00000000 00002000 00000000
Not a ULP-RISC-V wakeup, initializing it! 000B 688 00000FEC
Entering in deep sleep 17223353
```

bbrother, 2021-07-02:
Also verified this problem still exists on master 8e3e65a

ESP-Marius, 2021-07-05:
Hi @bbrother

I wasn't able to reproduce your exact issue, but I had similar issues when working on the ULP lately. W'e merged an riscv-ulp related update to our internal master, but it hasnt been synced to Github yet.

Would you mind testing and see if you have the same issue after applying this patch?

[0001-RISCV-ULP-Add-DS18B20-1wire-RISCV-ULP-example.txt](https://github.com/espressif/esp-idf/files/6761527/0001-RISCV-ULP-Add-DS18B20-1wire-RISCV-ULP-example.txt)

bbrother, 2021-07-07:
Hi @ESP-Marius ,

Sorry for the slow response, I have been swamped by a board bring up this week.  I tested your patch on v4 4-dev-1849-g8e3e65a47-dirty and it seems to fix the problem.  Thanks again.

ESP-Marius, 2021-07-09:
@bbrother Nice hear that this fixed it. I'll backport parts of that commit to 4.3 as well.

ESP-Marius, 2021-11-16:
Closing this as the fix has already been merged.

closed 2021-11-16 by ESP-Marius