espressif/esp-idf #7227: Incorrect ethernet PHY LAN8720 initialization sequence in IDF v4.2.1 environment.

charon22, 2021-07-02:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version :    v4.2.1-64-gd9ec7df39
- Build System:         [idf.py]
- Compiler version :  (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
We are testing using LAN8720. I am trying to use the internal clock of the ESP as the clock of the RMII. 
Since we use ESP-AT, GPIO16,17 cannot be used, so the internal clock is assigned to GPIO0.
When connecting the ESP with USB power, an error occurs in finding the PHY.
Below is the log.

> [2021-06-29 13:41:25.055] V (3309) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
> [2021-06-29 13:41:25.063] V (3315) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
> [2021-06-29 13:41:25.068] D (3323) intr_alloc: Connected src 38 to int 17 (cpu 0)
> [2021-06-29 13:41:25.073] E (3331) esp_eth.phy: No PHY device detected
> [2021-06-29 13:41:25.083] E (3333) lan8720: lan8720_init(380): Detect PHY address failed
> [2021-06-29 13:41:25.085] E (3339) esp_eth: esp_eth_driver_install(199): init phy failed
> [2021-06-29 13:41:25.092] regist ethernet cmd fail

As a result of checking the signal, I was able to confirm that the same problem as the link below is still there.
The following problem occurred in ESP-IDF Release v3.3.5.

https://github.com/espressif/esp-idf/issues/6821
I am using v4.2.1-64-gd9ec7df39.

Below is a capture of the signal.
Clock is not being generated while LAN8720's RESET pin is still working.
[image]

Question 1) I would like to know which one is better and which method is recommended to use an external clock or an internal clock.
Using the internal clock makes the circuit simpler, but I want to know what is better(internal or external).

Question 2) I want to know why LAN8720 does not work according to the reset document.

Tested on this issue multiple times. When resetting the USB power, a problem occurs every time, but when the EN button on the ESP board is pressed, there is no problem. I want to know why this is.

### Expected Behavior

> D (7582) esp_netif_lwip: starting dhcp client
> D (7587) esp_netif_lwip: call api in lwip: ret=0x0, give sem
> D (7592) event: running post ETH_EVENT:2 with handler 0x400e3c90 and context 0x3ffb5594 on loop 0x3ffbb134
> D (7602) emac_esp32: receive len= 60
> D (7606) emac_esp32: receive len= 342
> D (7610) emac_esp32: receive len= 342
> D (7747) emac_esp32: receive len= 192
> D (8311) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3ffd6c3c
> D (8311) esp_netif_lwip: if0x3ffd6708 ip changed=1
> D (8311) event: running post IP_EVENT:4 with handler 0x4014911c and context 0x3ffd6d80 on loop 0x3ffbb134
> D (8320) esp_netif_handlers: esp_netif action got_ip with netif0x3ffd6708 from event_id=4
> I (8329) esp_netif_handlers: eth ip: 10.0.28.177, mask: 255.255.255.0, gw: 10.0.28.1
> D (8337) event: running post SYSTEM_EVENT:25 with handler 0x4014b83c and context 0x3ffbbe80 on loop 0x3ffbb134

### Actual Behavior
> [2021-06-29 13:41:25.055] V (3309) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
> [2021-06-29 13:41:25.063] V (3315) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
> [2021-06-29 13:41:25.068] D (3323) intr_alloc: Connected src 38 to int 17 (cpu 0)
> [2021-06-29 13:41:25.073] E (3331) esp_eth.phy: No PHY device detected
> [2021-06-29 13:41:25.083] E (3333) lan8720: lan8720_init(380): Detect PHY address failed
> [2021-06-29 13:41:25.085] E (3339) esp_eth: esp_eth_driver_install(199): init phy failed
> [2021-06-29 13:41:25.092] regist ethernet cmd fail
### Steps to reproduce

1. step1 : Rework the LAN8720 board to use the internal clock.
2. step2 : Connect ESP32 to LAN8720.
3. step3 : Connect LAN to RJ45.
4. step4 : After connecting the USB port of the ESP, check the message of the debug UART.

charon22, 2021-07-08:
Hi ESP-IDF,
We compared the signal when the USB power is reconnected and the EN button is pressed.
1. USB Power is reconnected
[image]
  > After 1)reset, before the 2)rx signals go low, 3)the PHY address is requested. In this case, an error occurs.

2. EN button is pressed
[image]
> After reset, the rx signals go low almost at the same time, and then the PHY address is requested. In this case, the PHY address is successfully obtained.

Comparing the above two cases, when the USB power is reconnected, the RX signal changes from High to Low very late. As a result, it seems that the PHY address cannot be obtained.

I would like to make a few suggestions on this.
> 1. When initializing PHY, put a delay of about 15msec before obtaining the PHY address.
> 2. Retry when getting PHYaddress fails
Below is the related example code.

> static esp_err_t lan8720_init(esp_eth_phy_t *phy)
> {
>     phy_lan8720_t *lan8720 = __containerof(phy, phy_lan8720_t, parent);
>     esp_eth_mediator_t *eth = lan8720->eth;
>     // Detect PHY address
>     if (lan8720->addr == ESP_ETH_PHY_ADDR_AUTO) {
> 		ets_delay_us(15000); // insert min input assert time
> 		//PHY_CHECK(esp_eth_detect_phy_addr(eth, &lan8720->addr) == ESP_OK, "Detect PHY address failed", err); // org
> 		int checkcnt = 0;
> 		for ( ; checkcnt < 10 ; checkcnt++ )
> 		{
>         	if ( esp_eth_detect_phy_addr(eth, &lan8720->addr) == ESP_OK)
> 				break;
> 			ESP_LOGE(TAG, "%s(%d): Detect PHY address failed(%d)", __FUNCTION__, __LINE__,checkcnt);
> 			if ( checkcnt == 9)
> 			{
> 				goto err;
> 			}
> 		}
>     }
>     /* Power on Ethernet PHY */
>     PHY_CHECK(lan8720_pwrctl(phy, true) == ESP_OK, "power control failed", err);
>     /* Reset Ethernet PHY */
>     PHY_CHECK(lan8720_reset(phy) == ESP_OK, "reset failed", err);
>     /* Check PHY ID */
>     phyidr1_reg_t id1;
>     phyidr2_reg_t id2;
>     PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IDR1_REG_ADDR, &(id1.val)) == ESP_OK,
>               "read ID1 failed", err);
>     PHY_CHECK(eth->phy_reg_read(eth, lan8720->addr, ETH_PHY_IDR2_REG_ADDR, &(id2.val)) == ESP_OK,
>               "read ID2 failed", err);
>     PHY_CHECK(id1.oui_msb == 0x7 && id2.oui_lsb == 0x30 && id2.vendor_model == 0xF, "wrong chip ID", err);
>     return ESP_OK;
> err:
>     return ESP_FAIL;
> }
This is the log captured when this code is applied. Even if the RX signal goes low late, we can get PHY address later.

[image]

Below is the result of testing with the 15msec delay removed from the code above, leaving only the retry code. On the second attempt, the PHY value is normally obtained. 
However, there is a significant delay between the first and second request times, and I am curious as to why this is so.
[image]

I'm not sure what caused this problem and which one to apply.
I would like to hear your comments on this.

Thanks

kostaond, 2021-07-23:
@charon22 a fix has been pushed to master branch. Could you please test if the issue has been fixed? You created nice analysis when reported it so it would be great if you confirm it was fixed using the same testbed :)

Alvin1Zhang, 2021-08-18:
@charon22 Thanks for reporting, and will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2021-08-18 by Alvin1Zhang