espressif/esp-idf #7228: install.sh fails if libusb-1.0-0-dev is not present

Zibri, 2021-07-04:
solution:

add
apt install libusb-1.0-0-dev

in the install script.

quickfix:
just install it before running install.sh
with: apt install libusb-1.0-0-dev

Alvin1Zhang, 2021-07-05:
@Zibri Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

dobairoland, 2021-07-05:
Hi @Zibri. Yes, you are right. However, the required `libusb-1.0-0` package on the list of packages which has to be run before running the installer. Please see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#install-prerequisites

dobairoland, 2021-07-13:
I'm closing this for lack of activity. Please notify me if this is still an issue.

closed 2021-07-13 by dobairoland