espressif/esp-idf #7229: unexpected abort is resulting for allocating big buffers

formus14, 2021-07-04:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.2.1
- Build System:       CMake
- Operating System:     Windows
- Using an IDE?: Yes (Visual GDB)
- Power Supply:         USB

## Problem Description

While allocating a big array (Bigger than the available free memory), an abort is being executed.

despite the SDK config options:

- CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE=y

- CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS is not set
[image]

### Expected Behavior
Just return null without aborting as configured in the SDK config.

### Actual Behavior
returning null followed by aborting.

### Steps to reproduce

1. Flash the code below with the sdk config attached

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void app_main(void)
{
    printf("Hello world!\n");
    printf("Free heap: %d\n", esp_get_free_heap_size());
    
    // allocating 10 mb buffer , buffer should return null 
    int* buffer = malloc(10*1000*1000);

   
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6760467/sdkconfig.txt)
printf("Free heap: %d\n", esp_get_free_heap_size());
   assert(buffer!= NULL);
    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
I (28) boot: ESP-IDF v4.2.1-dirty 2nd stage bootloader
I (28) boot: compile time 20:48:58
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05890 ( 22672) map
I (120) esp_image: segment 1: paddr=0x000158b8 vaddr=0x3ffb0000 size=0x021e4 (  8676) load
I (124) esp_image: segment 2: paddr=0x00017aa4 vaddr=0x40080000 size=0x00404 (  1028) load
I (130) esp_image: segment 3: paddr=0x00017eb0 vaddr=0x40080404 size=0x08168 ( 33128) load
I (153) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x12a28 ( 76328) map
I (182) esp_image: segment 5: paddr=0x00032a50 vaddr=0x4008856c size=0x01938 (  6456) load
I (191) boot: Loaded app from partition at offset 0x10000
I (191) boot: Disabling RNG early entropy source...
I (191) cpu_start: Pro cpu up.
I (195) cpu_start: Application information:
I (200) cpu_start: Project name:     hello-world
I (205) cpu_start: App version:      1
I (210) cpu_start: Compile time:     Jul  4 2021 20:47:56
I (216) cpu_start: ELF file SHA256:  0c0626c634a8241f...
I (222) cpu_start: ESP-IDF:          v4.2.1-dirty
I (227) cpu_start: Starting app cpu, entry point is 0x400815e4
I (220) cpu_start: App cpu up.
I (238) heap_init: Initializing. RAM available for dynamic allocation:
I (245) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (251) heap_init: At 3FFB2A28 len 0002D5D8 (181 KiB): DRAM
I (257) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (263) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (270) heap_init: At 40089EA4 len 0001615C (88 KiB): IRAM
I (276) cpu_start: Pro cpu start user code
I (294) spi_flash: detected chip: generic
I (295) spi_flash: flash io: dio
W (295) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
Free heap: 299632
Free heap: 299632
assertion "buffer!= NULL" failed: file "../../../main/hello_world_main.c", line 24, function: app_main

abort() was called at PC 0x400d5153 on core 0
Setting breakpoint at 0x40083eee and returning...

```

## Other items if possible

- [[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6760470/sdkconfig.txt)] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2021-07-04:
`assert(buffer!= NULL);`

You have the wrong test if you expect buffer to be null

projectgus, 2021-07-05:
Hi @formus14,

Yes, the abort is happening because the assertion fails not because the allocation fails. If `CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS` was set then it would abort before malloc() returned at all.

I think the confusion is because of the option [CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-assert). This option only applies to FreeRTOS `configASSERT()` macros and not the normal libc `assert()` macros. There is no "print and continue" option for standard asserts, just [these options](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-optimization-assertion-level).

(We realise this is a bit confusing, and in ESP-IDF V5.0 we're considering removing the FreeRTOS-only options and having the behaviour follow the standard assert behaviour.)

formus14, 2021-07-05:
@projectgus Thanks, now it's clear for me!

closed 2021-07-05 by formus14