espressif/esp-idf #7230: esp_reset_reason_get_hint() always returns zero (0)

T94T, 2021-07-05:
I'm trying to set the reset reason hint before a SW reset. The `esp_reset_reason_get_hint()` method always returns `0` after a reboot though.

I tested both an ESP32 and an ESP32-S2 dev module - also updated to the latest version of arduino-esp32 (#5323).

Calling `esp_reset_reason_get_hint()` before restart returns the correct value. After a SW reset the value is `0` again.

This is a minimized sketch that I use:
```
#include "esp_private/system_internal.h"

void setup() {
  Serial.begin(115200);
  
  Serial.print("Reset reason: ");
  Serial.println(esp_reset_reason(), HEX);
  Serial.print("Reset reason hint: ");
  Serial.println(esp_reset_reason_get_hint(), HEX);

  Serial.println("rebooting now...");
  delay(2000);

  esp_reset_reason_set_hint(static_cast<esp_reset_reason_t>(0x1234));
  esp_restart();
}

void loop() {
  
}

```

TOLDOTECHNIK, 2021-07-05:
Ok, thanks. Hope there is a solution for this.

igrr, 2021-07-05:
Hi @T94T, both esp_reset_reason_set_hint and esp_reset_reason_get_hint are ESP-IDF internal functions, as indicated by the header file name (esp_private/system_internal.h) and by comments in the header file. Unfortunately we don't support calling them directly from the application.

If you need to persist some values during reset, you can define a static variable with `RTC_NOINIT_ATTR` attribute. This variable will not be modified during startup and will keep its value after software reset.

TOLDOTECHNIK, 2021-07-06:
The `esp_reset_reason_set_hint()` method worked some time ago. We need a way to hold a value during resets between two different applications stored in flash side by side.

T94T, 2021-07-06:
The `esp_reset_reason_set_hint()` method worked some time ago. We need a way to hold a value during resets between two different applications stored in flash side by side.

suda-morris, 2021-07-06:
@T94T The log behind is, at app start up stage, `esp_reset_reason_init` got invoked, where we called `esp_reset_reason_get_hint` for the first time, if you checked that function, your hint `0x1234` can be reflected there. But after that, since the hint is `0x1234` and it's not `ESP_RST_UNKNOWN`, so we cleared the hint by `esp_reset_reason_clear_hint`, where writing zero directly to one of the RTC register. That's why, you always get `0` when calling `esp_reset_reason_get_hint` from application.

T94T, 2021-07-06:
Thank you very much for the comments. We will now solve this with NVS.

closed 2021-07-06 by T94T