espressif/esp-idf #7231: ESP32-S3 Marlin 9 1B became incompatible

VaderMester, 2021-07-05:
## Environment

- Development Kit:     ESP32-S3-Addax
- Kit version (for WroverKit/PicoKit/DevKitC): V1.0
- Module or chip used:  ESP32-S3
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-1849-g8e3e65a47
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Win10
- (Windows only) environment type: VScode (cmd and PS)

## Problem Description

I think after the newest update of the master branch, Addax boards became incompatible?

I'm getting this in console while the system bootloops.
```
ESP-ROM:esp32s3beta3-20201106
Build:Nov  6 2020
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Invalid chip id. Expected 6 read 9. Bootloader for wrong chip?
ets_main.c 332
```
I just updated the master branch on my end a few hours ago and also did a tool upgrade. I had a project running on it yesterday compiled before I updated, so I guess the compatibility has been removed?

I also see that /esp-idf/Kconfing has 0x0009 as default if S3 is the target chip, same as in Sdkconfig.

Lines 58-64 in /esp-idf/Kconfig:
```
    config IDF_FIRMWARE_CHIP_ID
        hex
        default 0x0000 if IDF_TARGET_ESP32
        default 0x0002 if IDF_TARGET_ESP32S2
        default 0x0005 if IDF_TARGET_ESP32C3
        default 0x0009 if IDF_TARGET_ESP32S3
        default 0xFFFF
```

Does this mean that the S3 Addax boards I have are no longer supported?

VaderMester, 2021-07-05:
After some searching, I see in esptool.py that ESP32-S3(beta 3) chips have become unsupported. I'm curious why, because the sample boards for these have not been out there for long, and I don't see the official S3 available yet anywhere.
I though there would be a transition period for Addax V1.0 boards, until official ones become available?

igrr, 2021-07-05:
Hi @VaderMester, you are right that ESP32-S3 beta3 is not supported on the latest master branch. We are sorry for the inconvenience.

Have you received the "ESP32-S3 Engineering sample notes" along with the development board? If I am not mistaken, that document explains that the support of this engineering sample will not be maintained on the master branch once the support for the final production chip is added there.

Maintaining support for two versions of the chip (beta3 and the final production version) unfortunately would result in a very significant overhead for us, in part because the final production version requires a different cross-compiler toolchain.

You may continue working with the development board you have based on an earlier ESP-IDF commit. You can also request the sample of the production version of ESP32-S3, using the same contact as you used to get the engineering sample.

VaderMester, 2021-07-06:
> Hi @VaderMester, you are right that ESP32-S3 beta3 is not supported on the latest master branch. We are sorry for the inconvenience.
>
> Have you received the "ESP32-S3 Engineering sample notes" along [...]

Thanks for the definitive answer!

closed 2021-07-06 by VaderMester