espressif/esp-idf #7232: Can't connect to softAP. Clients do not receive IP address

GeorgeFlorian, 2021-07-05:
Development Kit: ESP32-DevKit-V4
Module or chip used: ESP-WROVER-B
IDF version: v4.3
Build System: [Make|CMake|idf.py]
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux
Using an IDE?: VS Code 1.56
Power Supply: USB

## Problem Description

My issue is similar to this one: https://github.com/espressif/esp-idf/issues/6108

I am trying to implement this project: https://github.com/DroneBridge/ESP32 and I can't connect to the ESP's AP mode.
I have compiled it for my board.

When I am trying to connect to the softAP, all I get is this:
```
I (103324) DB_ESP32: Wifi AP started!
I (110754) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (110754) wifi:station: 58:96:1d:be:d4:30 join, AID=1, b, 20
I (114814) wifi:station: 58:96:1d:be:d4:30 leave, AID = 1, bss_flags is 655392, bss:0x3ffbccc4
I (114814) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (115974) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (115974) wifi:station: 58:96:1d:be:d4:30 join, AID=1, b, 20
I (162364) wifi:station: 58:96:1d:be:d4:30 leave, AID = 1, bss_flags is 655393, bss:0x3ffbccc4
I (162364) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (162364) DB_ESP32: Wifi AP started!
I (741444) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (741454) wifi:station: 2e:fa:09:ac:34:43 join, AID=1, b, 20
I (759734) wifi:station: 2e:fa:09:ac:34:43 leave, AID = 1, bss_flags is 655409, bss:0x3ffbccc4
I (759734) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (759734) DB_ESP32: Wifi AP started!
```

It rarely works. I can connect 1 in 10 times. I have to reboot the ESP a bunch of times.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6764110/sdkconfig.txt)

VaderMester, 2021-07-05:
I experienced something simillar, after I changed my code to use the new netif implementation instead of the tcpip_adpter API.

GeorgeFlorian, 2021-07-05:
> I experienced something simillar, after I changed my code to use the new netif implementation instead of the tcpip_adpter API.

I have no experience using ESP-IDF APIs. That project is not made by me. It would be almost impossible for me to modify it.

GeorgeFlorian, 2021-07-08:
I am still having this problem.

https://github.com/DroneBridge/ESP32/issues/15

xueyunfei998, 2021-07-09:
hi @GeorgeFlorian 

You use the SoftAP example in IDF to test.
Example location：esp_idf/example/wifi/getting_started/softAP/

GeorgeFlorian, 2021-07-13:
> hi @GeorgeFlorian
> 
> You use the SoftAP example in IDF to test.
> Example location：esp_idf/example/wifi/getting_started/softAP/

The softAP example works just fine.

The problem is that whenever I have GPIO18 and 19 (using them as RX and TX) connected to a Flight-Controller the softAP won't assign IP addresses.
The FC operates on 3v3.
[image]

I have measured the voltages on the FlightController RX and TX and they are 3v3.

I've created a thread on the ESP32 forum: https://www.esp32.com/viewtopic.php?f=12&t=21964

It doesn't make sense at all. If I connect pins to the ESP, the softAP won't assign IP addresses. I've also tried other pins.
As soon as the FC is powerd up, the ESP won't assign IP addresses.

What does the GPIOs 18,19,21,22 have to do with the WiFi ?

dainiusrid, 2021-07-14:
Hi,

same problem, forgot:

esp_netif_create_default_wifi_ap();

IP's are assigned now.

Regards.

GeorgeFlorian, 2021-07-14:
> Hi,
> 
> same problem, forgot:
> 
> esp_netif_create_default_wifi_ap();
> 
> IP's are assigned now.
> 
> Regards.

I've literally no idea what you said.
Can you explain more, please ?

dainiusrid, 2021-07-14:
Hello,

dronebridge project uses tcpip_adapter, which is deprecated:
...ESP-NETIF component is a successor of the tcpip_adapter, former network interface abstraction, which has become deprecated since IDF v4.1....
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html
That could be a reason. 
New ESP-IDF example for softap uses netif.

So maybe it would be good to try to build the project with IDF version lower than 4.1.

Regards

GeorgeFlorian, 2021-07-14:
> Hello,
>
> dronebridge project uses tcpip_adapter, which is deprecated:
> ...ESP-NETIF component is a successor of the tcpip_adapter, former network interface abstraction, which has become deprecated [...]

I've tried with 4.0 and it's the same.

The problem is that whenever I have GPIOs acting as RX/TX connected to a Cube Orange Flight Controller, the softAP won't assign IP.
I've opened another issue documenting more: https://github.com/espressif/esp-idf/issues/7268

xueyunfei998, 2021-08-05:
hi @GeorgeFlorian 

I think you can close this problem. Later, we will use this ticket to track this problem.
https://github.com/espressif/esp-idf/issues/7268

closed 2021-08-05 by GeorgeFlorian