espressif/esp-idf #7234: How do i solve this problem?  tool idf-python-wheels has no installed versions

19fchen, 2021-07-06:
Setting IDF_PATH: C:\Desktop\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.19.8.
WARNING: tool idf-python-wheels found in path, but returned non-zero exit code (1) with error message:
ϵͳҲָ·

WARNING: tool idf-python-wheels version idf4.3_py3.9_2021-01-07 is installed, but returned non-zero exit code (1) with error message:
ϵͳҲָ·

ERROR: tool idf-python-wheels has no installed versions. Please run 'C:\Desktop\esp-idf\install.bat' to install it.

When I install Esp-idf on my win10 , this problem arises.

georgik, 2021-07-14:
Installation on Windows should be performed by Windows Installer. It contains several additional checks for integrity of environemnt.
Please use Windows Installer to update your environment https://dl.espressif.com/dl/esp-idf/

idf-python-wheels are no longer necessary in tools/tools.json and was removed, because on Windows wheels installation is handled by Windows Installer. Possible solutions:
- update git repo to release/v4.3
- run Windows Installer and let it update the installation
- manually edit tools/tools.json and remove the package

georgik, 2021-07-19:
Closing the issue, because solution exists. Feel free to reopen the issue in case of futher problems discovered.

closed 2021-07-19 by georgik