espressif/esp-idf #7235: secure_element i2c mutex protection

iscapla, 2021-07-06:
Hello,

I would want to use the library secure_element atecc608_ecdsa at the same time of some I2C sensors.
Most of them could be find on this repo: [esp-idf-lib](https://github.com/UncleRus/esp-idf-lib).

¿Could you add mutex usage by the library?

A quick fix is to modify hal_esp32_i2c.c functions but, as this library consists to build secure elements, could be better if it can be updated without making this changes every time.

I wrote a post some weeks ago but without any answer from Espressif: [link](https://esp32.com/posting.php?mode=reply&f=13&t=21124&sid=ce689fd526d54a342e02734bee6177d8)

Thanks.

mahavirj, 2021-07-14:
@iscapla 

It is not very clear from your report as where do you expect mutex to be added. Can you please elaborate, please point to specific code pointers for same?

Alvin1Zhang, 2021-07-19:
@iscapla Thanks for reporting, would you please help share if any more details for the feature request? Thanks.

iscapla, 2021-07-19:
Hello @mahavirj @Alvin1Zhang,

In my custom board development I have ESP32 all possible pins connected to other peripherals so It is impossible to use 2 independent I2C buses: one for sensors, other for atecc608 secure element.

As the code is implemented using different threads is will be problems to access to the same I2C bus for "sensor thread" and "netif thread". For that reason I need a mutex to implement an atomic use of the bus.

As [sensors library](https://github.com/UncleRus/esp-idf-lib) implements a common mutex for all sensors, my request refers to use that same mutex to block I2C usage for 'secure element'.

I ignore where could be best location to use the mutex inside the "secure element" code without causing any troubles.

To sum up my request.....
As atecc608 I2C connection could be shared with other devices from different threads, it is needed to provide a previously created mutex (same for all devices on the same bus). These mutex has to be reserved before I2C transmission and release after all.

Thanks so much.

mahavirj, 2021-07-19:
@iscapla 

> As sensors library implements a common mutex for all sensors, my request refers to use that same mutex to block I2C usage for 'secure element'.

I am not sure how we can help here, as this is third party library. Probably you can modify application to have higher layer synchronization between using "secure element" access APIs and "sensor library" APIs. This way you should be able to handle this issue entirely in application layer code itself.

If you still think that I2C driver provided in ESP-IDF ("secure_element" I2C HAL is based on top of this) is missing out on some synchronization then please point specific case, we will evaluate further.

iscapla, 2021-07-20:
Hello @mahavirj,
Sorry if I'm not been clear enough for the request.

The third party sensor library already implements mutex access for every I2C bus but 'secure element' didn't (or at least I couldn't find the way how to use it).

As you provably know, on esp-cryptoauthlib, implements HAL functions for esp32 architecture, on file "_esp-cryptoauthlib/cryptoauthlib/third_party/hal/esp32/hal_esp32_i2c.c_". On the same file, on functions _hal_i2c_send_ and _hal_i2c_receive_ there is no mutex usage to block bus access.
My request for Espressif is ... Would you please add mutex usage on this file or in upper layers to be able to use 'secure element' with other I2C devices without causing any corrupted data or a bus blockage?

Could it be possible to provide a previous mutex initialized on application thread and only reserve and release functions be used by third-party code (sensor library or secure element)?

iscapla, 2021-07-20:
A possible solution could be:

(on file: hal_esp32_i2c.c)
```C
ATCA_STATUS hal_i2c_send(ATCAIface iface, uint8_t address, uint8_t *txdata, int txlength)
{
    ATCAIfaceCfg *cfg = iface->mIfaceCFG;
    esp_err_t rc;

    if (!cfg)
    {
        return ATCA_BAD_PARAM;
    }
	
    //---------- RESERVE MUTEX
    if (!xSemaphoreTake(cfg->atcai2c.lock, pdMS_TO_TICKS(CONFIG_I2CDEV_LOCK_TIMEOUT))){
        return ESP_ERR_TIMEOUT; //or ATCA_COMM_FAIL
    }
    //------------------------

    //ESP_LOGD(TAG, "txdata: %p , txlength: %d", txdata, txlength);
    //ESP_LOG_BUFFER_HEXDUMP(TAG, txdata, txlength, 3);

    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    (void)i2c_master_start(cmd);
    (void)i2c_master_write_byte(cmd, address | I2C_MASTER_WRITE, ACK_CHECK_EN);
    (void)i2c_master_write(cmd, txdata, txlength, ACK_CHECK_EN);
    (void)i2c_master_stop(cmd);
    rc = i2c_master_cmd_begin(cfg->atcai2c.bus, cmd, 10);
    (void)i2c_cmd_link_delete(cmd);

   //---------- RELEASE MUTEX
   xSemaphoreGive(cfg->atcai2c.lock);
   //------------------------

    if (ESP_OK != rc)
    {
        return ATCA_COMM_FAIL;
    }
    else
    {
        return ATCA_SUCCESS;
    }
}
```

Here we expect that _cfg->atcai2c_ has another component of type mutex initialized by User application.

I can't handle this scenario on application layer because in most of the cases, user hasn't got the option to know when data is sent or received over the socket. Example: when MQTT (esp32 implementation) send the PINGREQ message.

AdityaHPatwardhan, 2021-07-20:
Hi @iscapla, The `I2C` driver in the esp-idf handles the `i2c` bus access with the `cmd_mux` which is locked when `i2c_master_cmd_begin` API is called.  so I think adding a separate mutex for `I2C` bus in secure element shouldn't be required.

Also in your suggested solution you have added the semaphore in the `hal_esp32_i2c.c`. Does it solve your problem ? I am not quite sure how it would be useful to synchronise the `I2C` bus access of a third party sensor which doesn't use this semaphore.

Can you provide any sample code with which I would be able to reproduce the issue on my side ?, I think that would help in fixing it.

iscapla, 2021-07-20:
Hi @AdityaHPatwardhan,
Thanks so much for the clarification.

Just to confirm it, does that mean I2C will always thread safe?

Due to component problems I still didn't receive the prototype so I can't handle this by code. After knowing the scenario, i tried to figure out which will be the behavior of I2C bus when it has more that 1 device connected in a thread environment (I didn't know that cmd_mux had this protection).

On my suggested solution, the third party sensor has to use the same semaphore to work properly. As you mention, it has no sense if they use different ones.

AdityaHPatwardhan, 2021-08-30:
@iscapla Sorry for reverting to this very late. A discussion had been going on internally about the cause of this issue.
As per this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv420i2c_master_cmd_begin10i2c_port_t16i2c_cmd_handle_t10TickType_t The I2C API are not thread safe.
I think the I2C requires the application to handle it.
But since adding a seperate lock in esp-cryptoauthlib would be a problem if it is not used by other peripherals as well!.
Have you already found any solution to this ? or some code to reproduce this problem.

iscapla, 2021-08-30:
Hello @AdityaHPatwardhan, thanks so much for your update about I2C thread safe limitations.
I don't understand why could be a problem if esp-cryptoauthlib use a mutex witch is not used externally by other libraries.

I didn't try to solve this scenario by code because our current projects doesn't use secure_element (others will) but I did some search over the libraries. Did you explore the idea to extend esp32 with mutex usage?

I don't think, as I wrote in some messages before, hard coding mutex implementation directly on I2C hal function is the best or correct option.

Please, correct me if I am wrong about what am going to write next.

This library use pkcs11 as one of the Public-Key Cryptography methods, where it is declared a control structure called 'pkcs11_lib_ctx'. On this struct, there is a mutex usage and looks connected to freeRTOS function over "ATCA_STATUS hal_lock_mutex(void * pMutex)" on cryptoauthlib/lib/hal/hal_freertos.c file.

Furthermore, function C_Initialize could initialize all the pkcs11 stuff using a CK_C_INITIALIZE_ARGS parameter type where a custom mutex usage could be added.

Is there any configuration macro or initialization point from a esp32 app code to use all this hierarchy?

AdityaHPatwardhan, 2021-09-07:
Hi @iscapla What I meant was,
To synchronise multiple third party variables, we would need all of them to use the same semaphore for blocking the I2C access. Since currently it is not handled in the I2C driver. We may have to handle it in the application layer.

I don't think there is anything similar in case of esp32 that is present at the moment.
I am having discussion with internal team and see if we find any alternate way to resolve your issue.

iscapla, 2021-09-08:
Hi @AdityaHPatwardhan,
Thank you for the clarification. Please, let me know if you achieve any solution to this scenario. I will continue this thread you if I find any proposal.

ivanroberto, 2022-05-15:
Hi @AdityaHPatwardhan, 
I'm implementing now a product with LCD via SPI, touch panel via I2C, and IO expansion MCP23017 via I2C. 
For touch, I'm using espressif new touch bsp driver [(https://github.com/espressif/esp-bsp/tree/master/components/esp_lcd_touch)](url), and MCP23017 share the same I2C port because I don't have more spare pins. 
So my team and I do not think that is a good idea to modify the espressif component to implement a lock in an external semaphore to MCP23017 and F5X06 works nicely (they are in different tasks), because espressif's driver will be updated and we need to make some modifications again to follow improvements made on it.

There is a nice guy @Rop Gonggrijp that already make a great job implementing the I2C_Manager [https://github.com/ropg/i2c_manager](url), please take a look.
 
Considering the nature of I2C interface to share devices on the same interface, we think that the espressif i2c driver should be thread-safe.

Thanks

grginho, 2024-04-25:
Hello everyone,

In the new driver for I2C documentation says:
Thread Safety
The factory function i2c_new_master_bus() and i2c_new_slave_device() are guaranteed to be thread safe by the driver, which means, user can call them from different RTOS tasks without protection by extra locks. Other public I2C APIs are not thread safe. which means the user should avoid calling them from multiple tasks, if user strongly needs to call them in multiple tasks, please add extra lock.

Does this mean that new I2C driver is thread safe, or that only i2c_new_master_bus() and i2c_new_slave_device() are thread safe?

mythbuster5, 2025-01-13:
@grginho docs updated.

To be honest, this topic is a little out of date, and we have a set of new i2c interface since 5.2 and it guarantee thread safe. So if there is no following topic, i will close it soon.

closed 2025-01-14 by AdityaHPatwardhan