espressif/esp-idf #7236: Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment) caused by ODR violation.

b1ackviking, 2021-07-06:
## Environment

- Development Kit:  custom PCB
- Module or chip used: ESP32-S2
- IDF version: v4.4-dev-1849-g8e3e65a47
- Build System: CMake
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 3.3V

## Problem Description

[This commit](https://github.com/espressif/esp-idf/commit/64750acbb0a0f5268844211ec6485bb94e3d416f) (found with `git bisect`) defines the `__cxx_eh_arena_size_get` function which is defined in `$HOME/.espressif/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/lib/no-rtti/libstdc++.a(eh_alloc.o)`.

This is a violation of the One Definition Rule, and it causes a crash due to unaligned memory access in the `aes_ll_write_key` function when I use external SPI RAM (ESP-PSRAM64H).

Reverting the commit fixes the bug.

### Expected Behavior

Everything works.

### Actual Behavior

The crash happens inside the `aes_ll_write_key` function during 1) WiFi initialization 2) establishing a secure network connection (with TLS).

### Steps to reproduce

1. Take ESP-IDF v4.4-dev-1849-g8e3e65a47
1. Take devkit or PCB with ESP32-S2 and ESP-PSRAM64H.
1. Activate PSRAM in the menuconfig and force malloc to use it (see `sdkconfig.defaults` below)
1. Run the example located in `examples/protocols/mqtt/ssl_mutual_auth` (see also https://test.mosquitto.org/ssl/index.php)

### Code to reproduce this issue

The example located in `examples/protocols/mqtt/ssl_mutual_auth`.

## Debug Logs

NOTE: the line `I (24) boot: ESP-IDF v4.4-dev-1849-g8e3e65a47-dirty 2nd stage bootloader` says `dirty` because of the modified `client.key` and `client.crt` in the example. There are no other changes.

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x173c
load:0x4004c000,len:0xa04
load:0x40050000,len:0x2b54
entry 0x4004c1f4
I (24) boot: ESP-IDF v4.4-dev-1849-g8e3e65a47-dirty 2nd stage bootloader
I (24) boot: compile time 14:31:14
I (24) boot: chip revision: 0
I (37) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : DIO
I (38) boot.esp32s2: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=204d0h (132304) map
I (120) esp_image: segment 1: paddr=000304f8 vaddr=3ffc77e0 size=03120h ( 12576) load
I (123) esp_image: segment 2: paddr=00033620 vaddr=40024000 size=0c9f8h ( 51704) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=40080020 size=866b0h (550576) map
I (249) esp_image: segment 4: paddr=000c66d8 vaddr=400309f8 size=06ddch ( 28124) load
I (256) esp_image: segment 5: paddr=000cd4bc vaddr=50000000 size=00010h (    16) load
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
[... 94 lines cut ...]

0x40032809: sta_rx_cb at ??:?

0x40035d15: ppRxPkt at ??:?

0x40034173: ppTask at ??:?

0x4002eab5: vPortTaskWrapper at /home/b1ack/projects/esp-idf/components/freertos/port/xtensa/port.c:168
```

## Other items if possible

The `sdkconfig.defaults` file.

```
CONFIG_ESP32S2_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=0
```

ESP-Marius, 2021-07-07:
Hi @b1ackviking thanks for reporting this. 

libstdc++ __cxx_eh_arena_size_get is defined as weak function so this is probably not related to any ODR violation. Seems like buffer ends up unaligned.

We'll take a look at it.

Alvin1Zhang, 2021-08-30:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/3907634d208347370e0e6f5abb51cb58a70fd1b4, feel free to reopen.

b1ackviking, 2021-08-30:
Hi @Alvin1Zhang and @ESP-Marius,
Thanks for fixing the issue with unaligned memory access in the `aes_ll`. However, there is a similar issue in another module:
```
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.

Core  0 register dump:
PC      : 0x40167393  PS      : 0x00060f30  A0      : 0x8011bf26  A1      : 0x3ffea890
0x40167393: xor_data at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:113

A2      : 0x3f7b8f14  A3      : 0x3f7b4bf9  A4      : 0x00000000  A5      : 0x00000010
A6      : 0x00000010  A7      : 0x00000001  A8      : 0x8011a25b  A9      : 0x3ffea870
A10     : 0x3f4c004c  A11     : 0x00000000  A12     : 0x3ffe3220  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000009
EXCVADDR: 0x3f7b4bf9  LBEG    : 0x3ffe3220  LEND    : 0x00000000  LCOUNT  : 0x40026f39
0x40026f39: _xt_user_exc at /home/b1ack/projects/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:633

Backtrace:0x40167390:0x3ffea8900x4011bf23:0x3ffea8b0 0x4011c24e:0x3ffea8e0 0x4011c2f2:0x3ffea940 0x4011c36b:0x3ffea960 0x40110509:0x3ffea9e0 0x4010958d:0x3ffeaa20 0x4010978d:0x3ffeaab0 0x40109868:0x3ffeaad0 0x40109a7a:0x3ffeaaf0 0x4010c873:0x3ffeab10 0x40108e01:0x3ffeab30 0x40108e3d:0x3ffeab50 0x4010583e:0x3ffeab70 0x40104be9:0x3ffeab90 0x4010501c:0x3ffeabb0 0x40105337:0x3ffeac30 0x40106bdb:0x3ffeac60 0x40166331:0x3ffeac90 0x400cf658:0x3ffeacb0 0x40031b65:0x3ffeace0
0x40167390: xor_data at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:110

0x4011bf23: esp_gcm_ghash at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:299

0x4011c24e: esp_aes_gcm_update at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:464

0x4011c2f2: esp_aes_gcm_crypt_and_tag_partial_hw at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:536

0x4011c36b: esp_aes_gcm_crypt_and_tag at /home/b1ack/projects/esp-idf/components/mbedtls/port/aes/esp_aes_gcm.c:569

0x40110509: mbedtls_cipher_auth_encrypt at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/cipher.c:1073

0x4010958d: ssl_encrypt_buf at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:1624

0x4010978d: mbedtls_ssl_write_record at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3495

0x40109868: mbedtls_ssl_write_handshake_msg at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3418

0x40109a7a: mbedtls_ssl_write_finished at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:6615

0x4010c873: mbedtls_ssl_handshake_client_step at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:3875

0x40108e01: mbedtls_ssl_handshake_step at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8189

0x40108e3d: mbedtls_ssl_handshake at /home/b1ack/projects/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8213

0x4010583e: esp_mbedtls_handshake at /home/b1ack/projects/esp-idf/components/esp-tls/esp_tls_mbedtls.c:138

0x40104be9: esp_tls_handshake at /home/b1ack/projects/esp-idf/components/esp-tls/esp_tls.c:76

0x4010501c: esp_tls_low_level_conn at /home/b1ack/projects/esp-idf/components/esp-tls/esp_tls.c:423 (discriminator 15)

0x40105337: esp_tls_conn_new_sync at /home/b1ack/projects/esp-idf/components/esp-tls/esp_tls.c:485

0x40106bdb: ssl_connect at /home/b1ack/projects/esp-idf/components/tcp_transport/transport_ssl.c:119

0x40166331: esp_transport_connect at /home/b1ack/projects/esp-idf/components/tcp_transport/transport.c:181

0x400cf658: esp_mqtt_task at /home/b1ack/projects/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1381

0x40031b65: vPortTaskWrapper at /home/b1ack/projects/esp-idf/components/freertos/port/xtensa/port.c:159
```

The problem is in the `xor_data` function: https://github.com/espressif/esp-idf/blob/b63ec47238fd6aa6eaa59f7ad3942cbdff5fcc1f/components/mbedtls/port/aes/esp_aes_gcm.c#L108-L117

Should be:
```cpp
/* Function to xor two data blocks */
static void xor_data(uint8_t *dst, const uint8_t *src)
{
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
    *dst++ ^= *src++;
}
```

Will you be able to fix that ASAP yourself, or would it be better if I open a PR for this fix?

b1ackviking, 2021-08-30:
Also, @Alvin1Zhang, I can not reopen the issue because of the GitHub rules (I am not the person who closed it). Will you reopen it, or should I create another one and add a reference?

ESP-Marius, 2021-08-30:
@b1ackviking Ah, thanks. We'll have to take a more thorough look then and some more tests to root out all of these issues.

Sorry for your inconvenience.

AxelLin, 2021-09-06:
> Thanks for reporting, fix is available [3907634](https://github.com/espressif/esp-idf/commit/3907634d208347370e0e6f5abb51cb58a70fd1b4), feel free to reopen.

Hi @ESP-Marius 

I'm wondering if v4.3 branch requires this fix or not.
(The fix cannot cleanly apply to v4.3, so if v4.3 needs the fix I need to wait for the backport.)

ESP-Marius, 2021-09-06:
@AxelLin Yes, v4.3 will have the same issue, we are backporting it. This only seems to cause issues when using large PSRAMs though, so if you are not using this then you shouldn't see any issues.

AxelLin, 2021-09-06:
> @AxelLin Yes, v4.3 will have the same issue, we are backporting it. This only seems to cause issues when using large PSRAMs though, so if you are not using this then you shouldn't see any issues.

ok, thanks for the clarify. (I don't use PSRAM).

AxelLin, 2021-11-11:
> @AxelLin Yes, v4.3 will have the same issue, we are backporting it. This only seems to cause issues when using large PSRAMs though, so if you are not using this then you shouldn't see any issues.

@ESP-Marius 
Just curious why it took such long time to backport the fix to v4.3? (maybe it got lost?)

mahavirj, 2021-11-11:
@AxelLin Remaining fix (per https://github.com/espressif/esp-idf/issues/7236#issuecomment-908030029) was recently added with https://github.com/espressif/esp-idf/commit/1cb07a128868c3e6e37e2087f45e8eead3c00401. We missed out on adding reference to this issue. Backport MRs combining both these fixes are already in final approval stage, they should appear on github in next few days.

Alvin1Zhang, 2021-11-25:
Thanks for reporting, sorry for slow turnaround, fix on release/4.3 is available at https://github.com/espressif/esp-idf/commit/79b59947d8fc6ca7e1d620529b63c74872e99c0b and fix on release/4.2 is available at https://github.com/espressif/esp-idf/commit/a7a495f7962749df788c56d76c941190bc0e99b2, feel free to reopen.

closed 2021-11-25 by Alvin1Zhang

ugurakas, 2023-02-02:
Did anyone find a solution for this problem [](https://github.com/espressif/esp-idf/issues/7236#issue-937659087) ı'm taking the same error at version 5.0

ugurakas, 2023-02-02:
Did anyone find a solution for this problem, ı'm getting the same error at version 5.0