espressif/esp-idf #7237: ot_br example: how to add on-mesh prefix and ping external network

cahty82008, 2021-07-06:
Is it possible to add  on-mesh prefix to the esp32 border router and ping the external network(google etc.) from the joiner device's cli ? Could you please give some suggestions of how to do this ? 
@gjc13

gjc13, 2021-07-06:
It's possible to directly call `otBorderRouterAddOnMeshPrefix` to add an on-mesh prefix.

For pinging Google IPv6 address there will be more prerequisities:

* Use [DHCPv6-pd](https://datatracker.ietf.org/doc/html/rfc8415) to allocate the prefix from your WiFi router so that the Internet knows how to forward the packets to the Thread devices. Your ISP and router need to support this protocol and have full IPv6 internet access.
* The added on-mesh prefix shall have `mDefaultRoute` set to true.
* Turn on `LWIP_IPV6_FORWARD` in menuconfig.
* Set the dns server to an IPv6 dns server address. Google DNS server `2001:4860:4860::8888` shall be fine.

cahty82008, 2021-07-08:
Thanks for your suggestions

I have another question, how could I test if the local routing between the OpenThread network and a normal network works?

chshu, 2023-09-11:
@sihan6 Now we have detailed docs on how to test bi-directional Connectivity: https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/connectivity.html

closed 2023-09-11 by chshu