espressif/esp-idf #7238: Enable/Disable BT Classic SPP memory leak

Alextrimux, 2021-07-06:
## Environment

- Development Kit:     custom PCB
- Module or chip used:  ESP32-WROOM-32UE
- IDF version: v4.3
- Build System: CMake 3.16.4
- Compiler version: 8.4.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes - ESP-IDF Eclipse (from esp-idf-tools-setup-online-2.9)
- Power Supply: external 3.3V

## Problem Description

Potential memory leak detected when turning esp_spp_init/deinit. The leak was detected by using xPortGetFreeHeapSize. According to the logs, the leak is 36 bytes.

### Expected Behavior

```
I (4301) MAIN_memory: free: 193500
W (5301) SPP_ACCEPTOR_DEMO: Enable bluedroid...
I (5301) esp_spp_cb: ESP_SPP_INIT_EVT
I (5301) esp_spp_cb: ESP_SPP_START_EVT
I (5311) esp_bt_gap_cb: event: 10
I (5311) MAIN_memory: free: 190148
W (8311) SPP_ACCEPTOR_DEMO: Disable bluedroid...
I (8311) esp_spp_cb: ESP_SPP_UNINIT_EVT
I (8311) MAIN_memory: free: 193500
```

### Actual Behavior

```
I (4301) MAIN_memory: free: 193500
W (5301) SPP_ACCEPTOR_DEMO: Enable bluedroid...
I (5301) esp_spp_cb: ESP_SPP_INIT_EVT
I (5301) esp_spp_cb: ESP_SPP_START_EVT
I (5311) esp_bt_gap_cb: event: 10
I (5311) MAIN_memory: free: 190148
W (8311) SPP_ACCEPTOR_DEMO: Disable bluedroid...
I (8311) esp_spp_cb: ESP_SPP_UNINIT_EVT
I (8311) MAIN_memory: free: 193464
```

### Steps to reproduce

1. esp_bt_controller_init()
2. esp_bt_controller_enable()
3. esp_bluedroid_init()
4. esp_bluedroid_enable()
5. esp_spp_register_callback()
6. xPortGetFreeHeapSize()
7. esp_spp_init()
8. esp_spp_deinit()
9. xPortGetFreeHeapSize()

### Code to reproduce this issue

https://gist.github.com/Alextrimux/36fa821fce5dc18db6df5fe7c1af313b

## Other items if possible

- The default sdkconfig file from the bt_spp_acceptor example was used.

xiongweichao, 2021-07-27:
Hi, @Alextrimux 

Can you try the patch attached below and check if it solves your issue?  Let me know if it solves the issue.

Thanks
[fix_memory_leak.zip](https://github.com/espressif/esp-idf/files/6883217/fix_memory_leak.zip)

Alextrimux, 2021-07-30:
> Hi, @Alextrimux
> 
> Can you try the patch attached below and check if it solves your issue? Let me know if it solves the issue.
> 
> Thanks
> [fix_memory_leak.zip](https://github.com/espressif/esp-idf/files/6883217/fix_memory_leak.zip)

Yep, I applied your patch and it fixed the memory leak. Thanks a lot.

closed 2021-07-30 by Alextrimux