espressif/esp-idf #7242: Deadlock in BLE (BT?) message queues

MartinTJDK, 2021-07-07:
I have observed, that BLE (BT?) stack can enter a deadlock, when it received a lot of BLE advertisements and ESP32 is busy (eg. using Console heavily).

When enabling "CONFIG_BLE_HOST_QUEUE_CONGESTION_CHECK" the issue only occurs very seldom, but it is still happens.

When the issue happens, the error message about congestion is reported "forever", and the BLE never recovers.

It is pretty easy to trigger the issue, if you change the "BT_QUEUE_CONGEST_SIZE" to 20 in file "bt_common.h".

What I see is, that the task handling BTU ("btuT") never runs, and the BTU queue is never processed.

When I change priority of "hciH4T", "btuT" and "BTC_TASK" tasks to use same priority, the BLE does recover.

I am aware of possibilities to use filtering etc. but I think it is not good, that something ends up in a deadlock - no matter what...

I believe some issues regarding stalled BLE stack - reported by other users (#4001) - might be related to this finding.

HeFeng1947, 2022-04-06:
Is there any update on this issue? @MartinTJDK Martin, have you made any new progress on this issue? I'm sure I have the same problem

Indastri, 2025-01-21:
Since there was no response from the user - closing this issue. If the issue persists reopen this or create a new issue.

Cheers!

closed 2025-01-21 by Indastri

pschlan, 2025-01-22:
> Since there was no response from the user - closing this issue. If the issue persists reopen this or create a new issue.
> 
> Cheers!

I don't understand - there was no question for him, except for one user who has the same problem who is asking if there is any progress on fixing it.
So why is this being closed?

Indastri, 2025-01-27:
This was closed as it is a stale issue that could not be reproduced on our side and even as the question was coming from the other user side, the actual issue creator did not respond. As I stated before, If the issue persists with the new iterations please reopen this or create. a new issue.

Kind regards.