espressif/esp-idf #7244: esp_wifi_get_tsf_time takes between 0 and 120 us  to return

leragequit, 2021-07-07:
## Environment

- Development Kit:      ESP32-DevKitC ( Azdelivery Amazon board )
- Kit version
- Module or chip used:  ESP32-WROOM-32
- IDF version 4.3
- Build System:         |idf.py
- Compiler version xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows10 PowerShell
- Using an IDE?: VS code
- Power Supply:   USB

## Problem Description
ESP32 in station mode without traffic or heavy computation on any of the cores.
When using `esp_wifi_get_tsf_time` to get a timestamp, the function takes between 0 and 120 microseconds to return, making the timestamp useless for anything that needs < 100 microsecond accuracy.

While I would love the function to return quickly, having a more constant return time would probably be just as beneficial.

### Expected Behavior
Have `timer_set_counter_value` return in the same number of microseconds each time.
### Actual Behavior
**timer_set_counter_value** returns in about 60us with big variance from 0 to 120us, some outliers in the 400us range.
### Steps to reproduce

1. run this code multiple times
`    int64_t tsf0 = esp_wifi_get_tsf_time(WIFI_IF_STA);    
    int64_t tsf1 = esp_wifi_get_tsf_time(WIFI_IF_STA);
    printf("%lld;%lld:%lld\n", tsf0, tsf1, tsf1-tsf0);`

2. take a look at the monitor
`178652272007;178652272117:110
178652372002;178652372108:106
178652472002;178652472112:110
178652572021;178652572131:110
178652671934;178652672024:90
178652771952;178652772065:113
178652871981;178652872087:106
178652971941;178652972050:109
178653071992;178653072098:106`

3. run this code multiple times
`    GPIO.out_w1ts = GPIO_OUTPUT_0_INT;
    int64_t tsf0 = esp_wifi_get_tsf_time(WIFI_IF_STA);    
    GPIO.out_w1tc = GPIO_OUTPUT_0_INT;`

4. see this on your scope
[image]

### Code to reproduce this issue

https://gist.github.com/leragequit/2c398e91d13649e1d106a8f46429116e

sorry for the mess, didn't have the time to clean it all up

mhdong, 2022-01-22:
hi @leragequit 
Because we use the task scheduling mechanism internally, there will be inconsistent delays in function calls. But the time obtained by esp_wifi_get_tsf_time is accurate according to you want to synchronized.

jack0c, 2022-05-20:
@mhdong @leragequit I think the time is not correctly provided.
AP have a TBTT time to send beacon which is times beacon interval, for instance TBTTn=n*102400, n = 0,1,2,3,...
However, AP can't send the beacon in time, for instance sent at TSF, which is TSF.
Station receive the beacon at local time T0, which is record by the hardware.
If you can get TSF,T0, and your now the current time, then you can sync time among the stations connected to the same AP.

1. TBTT=TSF - TSF%Beaon_interval
2. T1 - T0 + TSF%Beaon_interval means time since AP TBTT
3. If you want to make TBTT corresponding to T2, then current time should be T2+T1-T0+TSF%Beaon_interval
4. then the delta is  T2-T0+TSF%Beaon_interval, you need increase your local time by T2-T0+TSF%Beaon_interval
5. all devices connect to the same AP should have same time

sherinds, 2022-10-14:
hi @leragequit 

How did you achieve the wireless timing sync finally? Did you get the exact tsf?

linhz0hz, 2023-07-12:
Bumping this issue. I also want to use the TSF time to get synchronization across different devices. 
Also, from this post [https://esp32.com/viewtopic.php?t=21119](url) it seems that esp_mesh_get_tsf_time is faster. Are they consistent?

leragequit, 2023-07-31:
@sherinds @linhz0hz sorry for the long silence. I did my initial investigation, did a proof of concept that worked great. for the real application the long delay and jitter will be an issue and I will invest some time the moment the project becomes hot again. I will report back maybe end of 2023.

Sherry616, 2024-03-18:
Hi @leragequit, could you please share the latest status of this issue? Thanks.

Sherry616, 2024-06-07:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product!

closed 2024-06-07 by Sherry616

Magdenen, 2024-06-15:
> @sherinds @linhz0hz sorry for the long silence. I did my initial investigation, did a proof of concept that worked great. for the real application the long delay and jitter will be an issue and I will invest some time the moment the project becomes hot again. I will report back maybe end of 2023.

Hello. Is there any news on this topic? Did you manage to get a ready-made mechanism for synchronizing time between several devices? Please share an example