espressif/esp-idf #7247: Watchdog on CPUx Idle task is not being triggered

carlessole, 2021-07-07:
- Development Kit: ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER
- IDF version: v4.3
- Build System: CMake/idf.py (usingEclipse Plugin)
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows
- (Windows only) environment type: -.
- Using an IDE?: Yes. Eclipse + ESP-IDF plugin
- Power Supply: USB

## Problem Description

When creating a new freeRTOS with minimum priority, **altough never call vTaskDelay or sched_yield()** functions, CPUx Idle Task Watchdog are not being triggered

### Expected Behavior

CPUx Idle Task Watchdog is triggered. Something like:

```
<ESC>[0;31mE (15317) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:<ESC>[0m
<ESC>[0;31mE (15317) task_wdt:  - IDLE (CPU 1)<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: Tasks currently running:<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: CPU 0: IDLE<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: CPU 1: test_miscellaneous_task<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: Print CPU 0 (current core) backtrace<ESC>[0m

Backtrace:0x400D7DE3:0x3FFB08B0 0x40082AB9:0x3FFB08F0 0x400E57B1:0x3FFB6ED0 0x400D5F05:0x3FFB6EF0 0x400D5419:0x3FFB6F10 0x40089D1C:0x3FFB6F40 

<ESC>[0;31mE (15317) task_wdt: Print CPU 1 backtrace<ESC>[0m

Backtrace:0x40084E09:0x3FFB0EB0 0x40082AB9:0x3FFB0EF0 0x400D5EC8:0x3FFB8B20 
```

### Actual Behavior

CPUx Idle Task Watchdog is **NEVER** triggered, and so device is not resetting

### Steps to reproduce

1. Configure the application to enable de CPUx Idle task watchdogs:
[image]

2. Create a task with priority '1':

```cpp
    rsp = xTaskCreate(test_miscellaneous_task, "test_miscellaneous_task", 4096, NULL, 1, &pTestMiscellaneousTaskHandle);
    if (rsp != pdPASS)
    {
        printf("'test_miscellaneous_task' can not be created\r\n");
    }
```

3. Make this task be stuck in an infinite loop:
```cpp
void test_miscellaneous_task(void *arg)
{
    uint32_t dummy_counter = 0;
    while (1)
    {
        dummy_counter++;
        if (dummy_counter > 0xFFFFFF)
        {
            printf("'test_miscellaneous_task'->dummy_counter = 0x%X\r\n", dummy_counter);
            dummy_counter = 0;
        }
        // Without calling 'sched_yield' device should be stuck in this loop and watchdog will trigger a system reboot
        // sched_yield();
    }

    // This code should never be reached
    vTaskDelete(pTestMiscellaneousTaskHandle);
}
```

4. Run the application and check that CPUx Idle Task Watchdog is triggered:
```
<ESC>[0;31mE (15317) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:<ESC>[0m
<ESC>[0;31mE (15317) task_wdt:  - IDLE (CPU 1)<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: Tasks currently running:<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: CPU 0: IDLE<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: CPU 1: test_miscellaneous_task<ESC>[0m
<ESC>[0;31mE (15317) task_wdt: Print CPU 0 (current core) backtrace<ESC>[0m

Backtrace:0x400D7DE3:0x3FFB08B0 0x40082AB9:0x3FFB08F0 0x400E57B1:0x3FFB6ED0 0x400D5F05:0x3FFB6EF0 0x400D5419:0x3FFB6F10 0x40089D1C:0x3FFB6F40 

<ESC>[0;31mE (15317) task_wdt: Print CPU 1 backtrace<ESC>[0m

Backtrace:0x40084E09:0x3FFB0EB0 0x40082AB9:0x3FFB0EF0 0x400D5EC8:0x3FFB8B20 
```

5. Change task priority to the lowest (0), same as IDLE tasks:

```cpp
    rsp = xTaskCreate(test_miscellaneous_task, "test_miscellaneous_task", 4096, NULL, 0, &pTestMiscellaneousTaskHandle);
    if (rsp != pdPASS)
    {
        printf("'test_miscellaneous_task' can not be created\r\n");
    }
```
6. Run the application again and check that Watchdog is never triggered

## Other items if possible

I have a summary of my tasks and they are as expected:

[image]

[image]

ESP-Marius, 2021-07-08:
Hi @carlessole 

This seems like expected behavior to me, esp-idf's freertos scheduler is by default preemptive and multiple tasks with the same priority will alternate running in a round-robin kind of way.

So even if your task is not yielding the scheduler will still preempt it and switch to the idle task at the same priority.

carlessole, 2021-07-08:
Thanks @ESP-Marius .
So I assume that is something like that: https://www.freertos.org/FreeRTOS_Support_Forum_Archive/February_2018/freertos_Preemption_by_an_equal_priority_task_c8cb93d0j.html

And so, I already have two questions about that:

1) Scheduler will just yield to switch to the ide task, or it will switch to any other task with same priority?

2)There is a way to configure my ESP-IDF project to avoid this preemptive behaviour?

Lot of thanks

ESP-Marius, 2021-07-08:
@carlessole Not exactly like that, because in IDF we have `configUSE_TIME_SLICING 1`.

-  It will switch between all tasks of the same priority, not just the idle task.
-  You could make FreeRTOS do cooperative scheduling instead of preemptive scheduling by setting `configUSE_PREEMPTION 0`, but I am not sure how well it will work. I haven't heard of anyone running IDF this way, and we do not test this internally.

Is there any specific reason that you want to run IDF this way?

carlessole, 2021-07-08:
Thanks again @ESP-Marius .

> Is there any specific reason that you want to run IDF this way?

The reason is that we are porting/reusing an old existing code where the current RTOS on that platform acts this way and so, all our code has been developed with that on mind.

> You could make FreeRTOS do cooperative scheduling instead of preemptive scheduling by setting configUSE_PREEMPTION 0, but I am not sure how well it will work. I haven't heard of anyone running IDF this way, and we do not test this internally.

I tried changing the value here:
[image]

But Wathcodg is still not being triggered.. :/

Thanks

carlessole, 2021-07-08:
Hi @ESP-Marius .
I think that it was because scheduler automatically canges my task from one core to another. Because now if forcing my task to be pinned to a core:
```cpp
    rsp = xTaskCreatePinnedToCore(test_sched_yield_task, "test_sched_yield_task", 4096, NULL, 0, &pTestSchedYieldTaskHandle, 0);
    if (rsp != pdPASS)
    {
        printf("'test_sched_yield_task' can not be created\r\n");
    }
```

And setting:

`#define configUSE_PREEMPTION			0`

Watchdog is being triggered! :)

negativekelvin, 2021-07-08:
What is the advantage in making your minimum task priority 0 instead of 1?

carlessole, 2021-07-08:
> 
> 
> What is the advantage in making your minimum task priority 0 instead of 1?

Hi @negativekelvin, as I mentioned before, we have a lot of code that we want to reuse and in lot of the threads (tasks) present on this code, just 'sched_yield' are used, no delays.

If I set priority 1 and just call sched_yield, IDLE tasks (with priority 0) will never be executed and then watchdog will be triggered.

Thanks

ESP-Marius, 2021-07-09:
@carlessole Good to hear that the issue got solved. But as I mentioned earlier, setting `configUSE_PREEMPTION 0` is not something we officially support, I'm pretty sure you will run into issues that might be hard to solve. My recommendation is still to adapt your code for use with a preemptive scheduler.

Please help close the issue if you consider your question answered :smile:

carlessole, 2021-07-09:
Hi @ESP-Marius.

Thanks to you I understand a little better the behaviour **but** the issue is still not solved, as you don't recommend me set `configUSE_PREEMPTION 0`.

Moreover when I have tried to set that, watchdog is still not triggered:
> I tried changing the value here:
> [image]
> 
> But Wathcodg is still not being triggered.. :/
> 
> May it be because of the configUSE_TIME_SLICING 1? Or maybe because of the Round Robin Scheduling?

Thanks again

negativekelvin, 2021-07-09:
Have you considered adding your priority 0 tasks to the watchdog instead?

carlessole, 2021-07-09:
> 
> 
> Have you considered adding your priority 0 tasks to the watchdog instead?

Of course I have all my tasks added to a watchdog.

My problem is not the watchdog of my created tasks, is about understanding WHY the IDLE tasks watchdog is not triggered, and that means that it is being executed, although I never do a sched_yield on my tasks (in this test)

igrr, 2021-07-09:
@carlessole are you able to try [tracing your program execution using SystemView](See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#app-trace-system-behaviour-analysis-with-segger-systemview)? This way you will be able to see the timeline of task switching, and will also find the events that may cause the task switch. You need a JTAG connection to use SystemView.

For example, a call to `printf` may cause the task to get blocked on a mutex which protects access to the console output channel. If the task blocks, another task will be able to run.

carlessole, 2021-07-09:
Hi @igrr ,
Thanks for answering.
Later I will try the `SystemView`. But I think that I can conclude that printf and so are not the problem. I have tried with a infinite dummy loop

```cpp
while(1)
{
    ;
}
```

What I have seen is that if I change the freeRTOS config to that:

`#define configUSE_PREEMPTION   0`

If I just create 1 thread with low priority without sched_yield, somehow both IDLE tasks are able to reset the watchdog and device is not rebooting.

But if I create three threads with priority 0 that are not performing the sched_yield, then YES! device is rebooting:

```
'test_scheduler_task_A' has been called 16777216 times in 1794410us. Period = 0us<CR>
'test_scheduler_task_B' has been called 16777216 times in 1783818us. Period = 0us<CR>
'test_scheduler_task_A' has been called 16777216 times in 1785181us. Period = 0us<CR>
'test_scheduler_task_B' has been called 16777216 times in 1783818us. Period = 0us<CR>
<ESC>[0;31mE (16319) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:<ESC>[0m
<ESC>[0;31mE (16319) task_wdt:  - IDLE (CPU 1)<ESC>[0m
<ESC>[0;31mE (16319) task_wdt: Tasks currently running:<ESC>[0m
<ESC>[0;31mE (16319) task_wdt: CPU 0: test_scheduler_task_B<ESC>[0m
<ESC>[0;31mE (16319) task_wdt: CPU 1: test_scheduler_task_A<ESC>[0m
<ESC>[0;31mE (16319) task_wdt: Print CPU 0 (current core) backtrace<ESC>[0m

Backtrace:0x400D5494:0x3FFB07E0 0x40082319:0x3FFB0800 0x400D4070:0x3FFBA5D0 0x400877CD:0x3FFBA610 

<ESC>[0;31mE (16319) task_wdt: Print CPU 1 backtrace<ESC>[0m

Backtrace:0x40083A81:0x3FFB0DE0 0x40082319:0x3FFB0E00 0x400D406A:0x3FFB95D0 0x400877CD:0x3FFB9610 

'test_scheduler_task_A' has been called 16777216 times in 1794409us. Period = 0us<CR>
'test_scheduler_task_B' has been called 16777216 times in 1822352us. Period = 0us<CR>
'test_scheduler_task_A' has been called 16777216 times in 1785181us. Period = 0us<CR>
```

Note that the third custom task(test_scheduler_task_C) is never called neither (as expected).

If `#define configUSE_PREEMPTION   1`, then as expected device is not resetting because IDLE task are able to run.

So, there is just one thing that I don't understand, why when just having 1 task, IDLE0 or IDLE1 watchdog are not being triggered?

Thanks

ESP-Marius, 2021-07-09:
@carlessole Are you pinning your task to a core? If not then your one task will probably end up migrating between cores, letting both idle task run enough that the watchdog doesnt trigger (when your task goes to core 1, idle 0 runs and vice versa)

carlessole, 2021-07-09:
> 
> 
> @carlessole Are you pinning your task to a core? If not then your one task will probably end up migrating between cores, letting both idle task run enough that the watchdog doesnt trigger (when your task goes to core 1, idle 0 runs and vice versa)

No, I am not pinning the task to any core.

That was what I thought.  But I am not quite sure because in my task I was continously checking in what core was running and it always returns the same coreID. Buy maybe I was not checking properly :/

Thanks

ESP-Marius, 2021-08-05:
Closing the issue as there doesnt seem to be any issues with IDF here. For more general questions our [forum](https://www.esp32.com/) can also provide help.

closed 2021-08-05 by ESP-Marius