espressif/esp-idf #7255: esp_wifi_80211_tx() crashes when sending custom data frames

doragasu, 2021-07-09:
## Environment

- Development Kit: ESP-WROOM-32 NodeMCU
- Module or chip used: ESP32-WROOM-32
- IDF version: v4.3 (c9646ff0beffc86d2c6d1bfbad34da16e328e0e3)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

There is a regression in the `esp_wifi_80211_tx()` function causing it to crash the program with a LoadProhibited exception.

I need to send custom raw data packets. For that, I build a custom packet, filling all the required fields (Frame Control data, addresses, LLC data and payload) and send it using `esp_wifi_80211_tx()`. This was working with commit `0bbc721a633f9afe4f3ebe648b9ca99e2f2f6d5f`, but I have updated the SDK to `v4.3` tag and now when the `esp_wifi_80211_tx()` function is called, the device crashes.

I have been debugging the problem, and I have discovered that the crash does **not** happen if I set the raw packet to transmit as a Beacon frame (setting the very first byte of the raw data packet to value `0x80`). But if the packet is configured as a data frame (with the very first byte set to value `0x08`), the program crashes immediately after calling `esp_wifi_80211_tx()`.

### Expected Behavior

After calling `esp_wifi_80211_tx()`, the raw data packet is sent and the program continues working normally.

### Actual Behavior

After calling `esp_wifi_80211_tx()`, the program crashes with a LoadProhibited exception.

### Steps to reproduce

1. Build a custom data frame (the first byte must have value `0x08`).
2. Send the frame using `esp_wifi_80211_tx()` function.

## Debug Logs

When the crash happens, the following backtrace is printed to the log:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40092987  PS      : 0x00060a30  A0      : 0x80090e02  A1      : 0x3ffc0b50  
0x40092987: rcUpdateTxDone at ??:?

A2      : 0x00000000  A3      : 0x3ffbaf50  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x3ffb2238  A8      : 0x3ffb75c0  A9      : 0x00000000  
A10     : 0x3ffb2238  A11     : 0x3ffbaf3c  A12     : 0x3ffb2580  A13     : 0x00060623  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000b  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x40092984:0x3ffc0b50 0x40090dff:0x3ffc0b70 0x4009107d:0x3ffc0ba0 0x40144ccf:0x3ffc0be0 0x401454c8:0x3ffc0c10 0x40092682:0x3ffc0c50 0x4008d139:0x3ffc0c80
0x40092984: rcUpdateTxDone at ??:?

0x40090dff: lmacTxDone at ??:?

0x4009107d: lmacEndFrameExchangeSequence at ??:?

0x40144ccf: lmacProcessTxSuccess at ??:?

0x401454c8: lmacProcessTxComplete at ??:?

0x40092682: ppTask at ??:?

0x4008d139: vPortTaskWrapper at /home/esp/dev/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: a4d23a070ff95f53

Rebooting...
```

Alvin1Zhang, 2021-08-02:
Thanks for reporting, we had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

Alvin1Zhang, 2021-08-04:
Thanks for reporting, fix on master is at https://github.com/espressif/esp-idf/commit/5578909ffba70eca3973f1303109b9077ad4cbf0, we are back porting the fix to release/4.3, will keep updated.

doragasu, 2021-08-04:
That's great news, thanks!

AxelLin, 2021-08-06:
> Thanks for reporting, fix on master is at [5578909](https://github.com/espressif/esp-idf/commit/5578909ffba70eca3973f1303109b9077ad4cbf0), we are back porting the fix to release/4.3, will keep updated.

The fix is already in v4.3 of esp32-wifi-lib https://github.com/espressif/esp32-wifi-lib/commit/0bbc969738fac352a6bce8261f0130403d0078d9.
But the esp-idf needs to update sub-modules.
I'm wondering is it safe for users to manually checkout v4.3 branch of esp32-wifi-lib to verify the fix

AxelLin, 2021-08-23:
Fix for release/4.3 by 1fd8fdcf7de7b25f9980e7cc973da2953784c273.

Alvin1Zhang, 2021-08-24:
Thanks for reporting and updates, feel free to reopen.

closed 2021-08-24 by Alvin1Zhang

chegewara, 2022-03-31:
I know its old issue and most likely solved, so i dont want to open new issue.
Currently i have not much time to confirm it with any newer version, but it happen with this commit (arduino v2.0.0):
https://github.com/espressif/esp-idf/tree/1cb31e50943bb757966ca91ed7f4852692a5b0ed
when i call `esp_wifi_80211_tx` before wifi connect to AP in STA mode then it crash with no backtrace. Wifi is properly configured, just AP is not available. When AP is available and esp32 is connected then i dont see crash, even after disconnecting from AP, but there has to be at least initial connection some time in the past.

If this can be confirmed by someone else thats good, if not thats fine, because i cant update to anything further, so i dont expect any fix.

Thanks