espressif/esp-idf #7256: Scheduler: Unexpected behaviour on 'sched_yield' when task priority is 0 (minimum) and 'configIDLE_SHOULD_YIELD = 1'

carlessole, 2021-07-09:
- Development Kit: ESP32-Wrover-Kit
- Kit version: v4
- Module or chip used:  ESP32-WROVER
- IDF version: v4.3
- Build System: CMake/idf.py (usingEclipse Plugin)
- Compiler version: esp-2020r3-8.4.0
- Operating System: Windows
- (Windows only) environment type: -.
- Using an IDE?: Yes. Eclipse + ESP-IDF plugin
- Power Supply: USB

I have found these 3 configuration parameters that I consider very important on deciding the scheduler behaviour. Their default values on ESP-IDF are:

```
configUSE_PREEMPTION = 1
configUSE_TIME_SLICING = 1
configIDLE_SHOULD_YIELD = 0
```

With that default configuration if we create 3 tasks with minimum priority (0) that just perform a sched_yield (yes I have a lot of code duplications on 'test_scheduler_task_X' but I want these way to make debugging better:

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "intesis_main.h"

TaskHandle_t pTestSchedulerTaskAHandle = NULL;
TaskHandle_t pTestSchedulerTaskBHandle = NULL;
TaskHandle_t pTestSchedulerTaskCHandle = NULL;

void test_scheduler_task_A(void *arg)
{
    int64_t us_since_boot_previous = esp_timer_get_time();
    uint64_t task_executions_counter = 0;

    sched_yield();
    while (1)
    {
        task_executions_counter++;
        if (task_executions_counter >= 1000)
        {
            int64_t us_since_boot_current = esp_timer_get_time();

            printf("'test_scheduler_task_A' has been called %llu times in %lldus. Period = %lldus\r\n",
                   task_executions_counter, (us_since_boot_current - us_since_boot_previous),
                   (us_since_boot_current - us_since_boot_previous) / task_executions_counter);

            task_executions_counter = 0;
            us_since_boot_previous = esp_timer_get_time();
        }

        sched_yield();
    }

    // This code should never be reached
    vTaskDelete(pTestSchedulerTaskAHandle);
}

void test_scheduler_task_B(void *arg)
{
    int64_t us_since_boot_previous = esp_timer_get_time();
    uint64_t task_executions_counter = 0;

    sched_yield();
    while (1)
    {
        task_executions_counter++;
        if (task_executions_counter >= 1000)
        {
            int64_t us_since_boot_current = esp_timer_get_time();

            printf("'test_scheduler_task_B' has been called %llu times in %lldus. Period = %lldus\r\n",
                   task_executions_counter, (us_since_boot_current - us_since_boot_previous),
                   (us_since_boot_current - us_since_boot_previous) / task_executions_counter);

            task_executions_counter = 0;
            us_since_boot_previous = esp_timer_get_time();
        }

        sched_yield();
    }

    // This code should never be reached
    vTaskDelete(pTestSchedulerTaskBHandle);
}

void test_scheduler_task_C(void *arg)
{
    int64_t us_since_boot_previous = esp_timer_get_time();
    uint64_t task_executions_counter = 0;

    sched_yield();
    while (1)
    {
        task_executions_counter++;
        if (task_executions_counter >= 1000)
        {
            int64_t us_since_boot_current = esp_timer_get_time();

            printf("'test_scheduler_task_C' has been called %llu times in %lldus. Period = %lldus\r\n",
                   task_executions_counter, (us_since_boot_current - us_since_boot_previous),
                   (us_since_boot_current - us_since_boot_previous) / task_executions_counter);

            task_executions_counter = 0;
            us_since_boot_previous = esp_timer_get_time();
        }

        sched_yield();
    }

    // This code should never be reached
    vTaskDelete(pTestSchedulerTaskCHandle);
}

void app_main(void)
{
    BaseType_t rsp = pdPASS;

    rsp = xTaskCreate(test_scheduler_task_A, "test_scheduler_task_A", 4096, &pTestSchedulerTaskAHandle, 0,
                      &pTestSchedulerTaskAHandle);
    if (rsp != pdPASS)
    {
        printf("'test_scheduler_task_A' can not be created\r\n");
    }
    vTaskDelay(500 / portTICK_PERIOD_MS);
    rsp = xTaskCreate(test_scheduler_task_B, "test_scheduler_task_B", 4096, &pTestSchedulerTaskBHandle, 0,
                      &pTestSchedulerTaskBHandle);
    if (rsp != pdPASS)
    {
        printf("'test_scheduler_task_B' can not be created\r\n");
    }
    vTaskDelay(500 / portTICK_PERIOD_MS);
    rsp = xTaskCreate(test_scheduler_task_C, "test_scheduler_task_C", 4096, &pTestSchedulerTaskCHandle, 0,
                      &pTestSchedulerTaskCHandle);
    if (rsp != pdPASS)
    {
        printf("'test_scheduler_task_C' can not be created\r\n");
    }

    while (1)
    {
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }
}
```
With freeRTOS above configuration and these code, these are the results:

'test_scheduler_task_A' has been called 1000 times in 9999992us. Period = 9999us<CR>
'test_scheduler_task_B' has been called 1000 times in 10000000us. Period = 10000us<CR>
'test_scheduler_task_C' has been called 1000 times in 10000000us. Period = 10000us<CR>
'test_scheduler_task_A' has been called 1000 times in 9999345us. Period = 9999us<CR>
'test_scheduler_task_B' has been called 1000 times in 9999865us. Period = 9999us<CR>
'test_scheduler_task_C' has been called 1000 times in 9999869us. Period = 9999us<CR>
'test_scheduler_task_A' has been called 1000 times in 9999872us. Period = 9999us<CR>
'test_scheduler_task_B' has been called 1000 times in 9999872us. Period = 9999us<CR>
'test_scheduler_task_C' has been called 1000 times in 9999872us. Period = 9999us<CR>

As it can be seen, our tasks are executed every 10ms. We don't like it because we need to return to the tasks faster. I assume that so is happening because

` configIDLE_SHOULD_YIELD = 0`

and so, IDLE task are not YIELDING until a tick happens. And as freeRTOS frequency is 100Hz -> Here are the 10ms. I know I can configure the freeRTOS tick up to 1000Hz -> period = 1ms but I would like to avoid being dependent on the freeRTOS tick.

So, as here (https://www.freertos.org/a00110.html#configIDLE_SHOULD_YIELD) is explained, I tried to set `configIDLE_SHOULD_YIELD = 1` (on `esp-idf\components\freertos\port\xtensa\include\freertos\FreeRTOSConfig.h`. So here are my current configuration parameters:

```
configUSE_PREEMPTION = 1
configUSE_TIME_SLICING = 1
configIDLE_SHOULD_YIELD = 1
```
Just doing this change on `configIDLE_SHOULD_YIELD` I see that task A is active every 20us (or less), that's great! But it seems like just task A is being executed with more latency (20us). The other two tasks are still running every 10ms.

```
'test_scheduler_task_A' has been called 50000 times in 1008372us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013367us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1008321us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013367us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013333us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1008370us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013333us. Period = 20us<CR>
'test_scheduler_task_B' has been called 1000 times in 9449897us. Period = 9449us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013155us. Period = 20us<CR>
'test_scheduler_task_C' has been called 1000 times in 9989909us. Period = 9989us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013355us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1008370us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013334us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1008372us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013331us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1008372us. Period = 20us<CR>
'test_scheduler_task_A' has been called 50000 times in 1013366us. Period = 20us<CR>

.
.
.
```
Is this the normal behviour?

Am I missing something?

Thanks

carlessole, 2021-07-12:
Hello again,
I have set again `configIDLE_SHOULD_YIELD = 0` (as default).
But this time I actively force a IDLE `sched_yield` when calling IDLE hook:

```cpp
    esp_err_t err = esp_register_freertos_idle_hook_for_cpu(my_idle_hook, 0);
    if (err != ESP_OK)
    {
        printf("'esp_register_freertos_idle_hook_for_cpu' on coreID 0 failed\r\n");
    }
    err = esp_register_freertos_idle_hook_for_cpu(my_idle_hook, 1);
    if (err != ESP_OK)
    {
        printf("'esp_register_freertos_idle_hook_for_cpu' on coreID 1 failed\r\n");
    }
```

```cpp
bool my_idle_hook(void)
{   
    sched_yield();
    return true;
}
```

Doing that the behaviour is the same as setting `configIDLE_SHOULD_YIELD = 1` (unexpected and non-desired)

Reading the documentation I see that on "esp_register_freertos_idle_hook_for_cpu":

```cpp
  * @warning Idle callbacks MUST NOT, UNDER ANY CIRCUMSTANCES, CALL
  *          A FUNCTION THAT MIGHT BLOCK.
```

So ESP-IDF is not prepared to make IDLE tasks to YIELD maybe?

Thanks

carlessole, 2021-10-15:
Thanks @Dazza0 ,
Both cores are enabled!
[image]

Carles

AxelLin, 2021-10-18:
@Dazza0 
Is it possible to post the fix here for testing?

AxelLin, 2021-10-19:
@Dazza0 

Just curious if https://github.com/espressif/esp-idf/pull/7351 is also an solution?

closed 2021-11-04 by espressif-bot

AxelLin, 2021-11-04:
@Dazza0 

I notice your commit removed taskENTER_CRITICAL_ISR(); and taskEXIT_CRITICAL_ISR(); around the code calculating ulRunTimeCounter
https://github.com/espressif/esp-idf/commit/8babb3bca6f206ea463d125e2e8ab174d94818a0#diff-dc94871a7cae5e76d227db279a7b26d27a9a648faf63359143d43ae3997281abL3394
This is not mentioned in the commit log so I'm not sure if it's intentional.

Also FYI, now the ulTotalRunTime is volatile in upstream code:
https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/e4e4fb01a19fa9bff3bec4b398c860aa92110c2e
https://forums.freertos.org/t/make-runtime-stats-working-with-compiler-optimization-enabled/9846

Dazza0, 2021-11-04:
@AxelLin

- Critical sections are still present in `vTaskSwitchContext()` which calls `taskSelectHighestPriorityTaskSMP()` when already in a critical section.
- We haven't run into `ulTotalRunTime` being optimized yet. But I don't see any harm in making it volatile. Could you file a separate GH issue?

AxelLin, 2021-11-17:
@Dazza0 
Just to remind this report is against v4.3, should backport the fix to release/v4.3.

AxelLin, 2021-12-24:
v4.3 branch still does not include this fix.

Alvin1Zhang, 2021-12-27:
@AxelLin Thanks for the attention, we have back ported this fix to 4.3, but still under internal reviewing, will sync up here once the fix is available on GitHub.