espressif/esp-idf #7257: esp32c3.rom.ld - no ets_write_char_uart()

teuthid, 2021-07-09:
## Environment

- Development Kit:  esp32-c3-devkitm-1
- Module or chip used:  ESP32-C3
- IDF version: 4.3
- Build System:  CMake
- Compiler version: riscv32-esp-elf, 8.4
- Operating System:  Linux
- Using an IDE: Visual Studio Code
- Power Supply: USB

## Problem Description

There is no symbol ets_write_char_uart in esp32c3.rom.ld

### Expected Behavior

### Actual Behavior

### Steps to reproduce

While linking:
undefined reference to `ets_write_char_uart'

### Code to reproduce this issue

#include "esp_log.h"

void app_main()
{
 ets_write_char_uart(any_character);
}

```

## Debug Logs

## Other items if possible

chegewara, 2021-07-09:
https://github.com/espressif/esp-idf/blob/639e7ad494d93fc82159f9fe854041bc43a96d5f/components/esp_rom/include/esp32c3/rom/ets_sys.h#L215

teuthid, 2021-07-09:
You didn't understand the report. There is no linker symbol 'ets_write_char_uart' in esp32c3.rom.ld.

igrr, 2021-07-09:
Hi @teuthid, even though ESP32-C3 ROM contains `ets_write_char_uart` function, it is not considered to be a public interface. Its location in ROM is not fixed and will change between ROM revisions, unlike that of interface functions. For this reason, it is not present in the linker script. The mistake that we made was not removing it from ets_sys.h header file for ESP32-C3.

If you need to output characters to the console in an ESP-IDF application, you can use the [UART driver](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/uart.html).

teuthid, 2021-07-10:
Hi @igrr, thanks for the detailed explanation. Unfortunately, I can't really use the UART component due to higher resource consumption. So I will probably need to use ets_printf() for ESP32-C3 (ugly hack).

atanisoft, 2021-07-12:
@teuthid would `uart_tx_one_char` work for your usecase?

teuthid, 2021-07-12:
@atanisoft Yes, it works. Also on ESP32 and ESP32-S2. Thank you very much! Hopefully this function will not be removed in the next versions of ESP-IDF ;)

teuthid, 2021-07-12:
Using uart_tx_one_char() instead of ets_write_char_uart().

closed 2021-07-12 by teuthid