espressif/esp-idf #7259: rx length > tx length in full duplex mode

ActivePeter, 2021-07-10:
[image]
Here is my code:
```c++
paErr paSPI::transmit(char spiId, unsigned char *data, unsigned int len)
{
    spi_transaction_t t;
    memset(&t, 0, sizeof(t));
    t.rxlength = 0;
    t.rx_buffer = NULL;
    t.tx_buffer = data;

    int sizePer = 64;
    while (len > 0)
    {
        if (len > sizePer)
        {
            t.length = (sizePer)*8;
        }
        else
        {
            t.length = len * 8;
        }

        // t.tx_data = data;
        // t.flags=SPI_TRANS_USE_TXDATA;

        /* code */
        // spi_device_get_trans_result()
        spi_device_polling_transmit(spi_handle, &t);
        len -= t.length / 8;
        t.tx_buffer += t.length / 8;
    }

    {
        // ESP_LOGI("App", "spi send succ");
        return E_Succ;
    }
}
```

negativekelvin, 2021-07-11:
//If set rxlength=0 in full_duplex mode, it will be automatically set to length

If you are going to loop you have to set it back to zero

ActivePeter, 2021-07-11:
> //If set rxlength=0 in full_duplex mode, it will be automatically set to length
> 
> If you are going to loop you have to set it back to zero

Thanks. It works.

closed 2021-07-11 by ActivePeter