espressif/esp-idf #7260: Eth2AP example in boot loop

robopp55, 2021-07-11:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:  ESP32-Ethernet-Kit_A_v1.2
- Kit version v1.2
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-1594-g1d7068e4b
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I built the Eth2AP example application and flashed my ESP32-Ethernet-Kit_A_V1.2 and it's stuck in a boot loop. The debug output is as follows:
```
ELF file SHA256: 7e58e01de8426c2c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6744
load:0x40078000,len:14236
ho 0 tail 12 room 4
load:0x40080400,len:3716
entry 0x40080680
I (29) boot: ESP-IDF v4.4-dev-1594-g1d7068e4b 2nd stage bootloader
I (29) boot: compile time 18:14:02
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=12bc0h ( 76736) map
I (140) esp_image: segment 1: paddr=00022be8 vaddr=3ffb0000 size=03a2ch ( 14892) load
I (147) esp_image: segment 2: paddr=0002661c vaddr=40080000 size=099fch ( 39420) load
I (163) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=5d4f4h (382196) map
I (302) esp_image: segment 4: paddr=0008d51c vaddr=400899fc size=0af34h ( 44852) load
I (320) esp_image: segment 5: paddr=00098458 vaddr=50000000 size=00010h (    16) load
I (331) boot: Loaded app from partition at offset 0x10000
I (331) boot: Disabling RNG early entropy source...
I (343) cpu_start: Pro cpu up.
I (343) cpu_start: Starting app cpu, entry point is 0x400810b4
I (329) cpu_start: App cpu up.
I (357) cpu_start: Pro cpu start user code
I (357) cpu_start: cpu freq: 160000000
I (357) cpu_start: Application information:
I (362) cpu_start: Project name:     eth2ap
I (367) cpu_start: App version:      1
I (371) cpu_start: Compile time:     Jul 11 2021 18:13:59
I (377) cpu_start: ELF file SHA256:  7e58e01de8426c2c...
I (383) cpu_start: ESP-IDF:          v4.4-dev-1594-g1d7068e4b
I (390) heap_init: Initializing. RAM available for dynamic allocation:
I (397) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (403) heap_init: At 3FFB6618 len 000299E8 (166 KiB): DRAM
I (409) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (415) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (422) heap_init: At 40094930 len 0000B6D0 (45 KiB): IRAM
I (429) spi_flash: detected chip: generic
I (433) spi_flash: flash io: dio
W (437) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (451) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (575) system_api: Base MAC address is not set
I (575) system_api: read default base MAC address from EFUSE
I (2495) eth_example: Ethernet Started
I (2495) eth_example: Ethernet Link Up
ESP_ERROR_CHECK failed: esp_err_t 0x3001 (ESP_ERR_WIFI_NOT_INIT) at 0x40087f88
file: "../main/ethernet_example_main.c" line 108
func: eth_event_handler
expression: esp_wifi_start()

abort() was called at PC 0x40087f8b on core 0

Backtrace:0x400d23a7:0x3ffbb5100x40087f99:0x3ffbb530 0x4008e9da:0x3ffbb550 0x40087f8b:0x3ffbb5c0 0x400d6733:0x3ffbb5e0 0x4012b8a5:0x3ffbb600 0x4012b4f2:0x3ffbb630 0x4012b5cd:0x3ffbb670 0x4008b03d:0x3ffbb690

```

### Expected Behavior

It doesn't go into a boot loop and serves up an access point.

### Actual Behavior

Boot loop

### Steps to reproduce

Build and burn using the idf python script

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

robopp55, 2021-07-11:
The boot loop only happens when I have the RTSP server running and boot the ESP32. If I allow the ESP32 to boot then connect the RTSP server, it boots fine, but I cannot connect to the AP with my iPhone.

ESP-YJM, 2021-07-12:
@robopp55 Could you please move `initialize_wifi()` calling before `initialize_ethernet()` in app_main and try.

ESP-YJM, 2021-07-22:
@robopp55 Is this patch could slove your issue? Or any update.

closed 2021-07-29 by espressif-bot