espressif/esp-idf #7262: Timer ISR is not executed on exact time when wifi is enabled

rahulbhivare, 2021-07-12:
We are generating square waves on a pin based on interrupt.
The square waves are generated properly if wifi is disabled.
But when wifi mode is enabled, then square wave duty cycle is not constant, which indicates issue in timer interrupt.

Below is explanation :
     1. Square wave output pin 1 : 35 .
     2. Interrupt pin  : 34

Algorithm:
      1. Timer is configured for fix time of 6 ms
      2. After every 10 ms, interrupt is generated
      3. On "Negative going pulse on Interrupt pin ", we:
           a. Make Square wave output pin "false"
           b. Start 6 ms timer
       This interrupt occurs at every 10 ms
      4. In Timer ISR (Which occurs after 6 ms of interrupt)
           a. Make Square wave output pin "true"
           b. enable flag to indicate "timer isr is executed"
      5. In an infinite loop, we stop the timer based on flag status explained in 3.b

Expected result:
A square wave of 6 ms OFF and 4 ms ON is generated

Observations/Actual result:
Square wave duty cycle varies when wifi is enabled. i.e. Timer isr is not getting called at 6 ms, as configured. But it occurs anytime between 5.5 ms to 6.5. ms

Note:
Tried to use high resolution timer (esp_timer) instead of timer0/1. Issue still persists

rahulbhivare, 2021-07-12:
Adding timer code.
This code is working well and generating square waves. 
It fails when we start WiFi
[demo_code.txt](https://github.com/espressif/esp-idf/files/6800840/demo_code.txt)

negativekelvin, 2021-07-13:
Use the ledc peripheral to generate your pulses

closed 2022-11-07 by Alvin1Zhang