espressif/esp-idf #7263: ESP32S2 USB support

abdulwasey7, 2021-07-12:
**USB implemetaion  on esp32s2**

I am searching for usb implemetaion on esp32 but couldn't find any implementaion so far.

kindly suggest if there is any working implementaion or not. thanks

chegewara, 2021-07-12:
What do you mean exactly by USB support?
There is tinyusb component for USB device, esp-idf console over USB, also some support for USB host. So, which USB support do you need?

https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/components/usb
https://github.com/espressif/esp-idf/blob/master/components/usb/maintainers.md
https://github.com/espressif/esp-idf/tree/master/components/tinyusb

abdulwasey7, 2021-07-15:
Thanks I have gone through the samples and it did what I want

closed 2021-07-15 by abdulwasey7