espressif/esp-idf #7267: BT_APPL: Unknown connection ID: 3 fail sending notification

BigSmileFace, 2021-07-13:
## Environment

- Module or chip used:  [ESP32-WROOM-32]
- IDF version :Release v4.2.2
- Build System:         [CMake]
- Operating System:     [Windows]
- Using an IDE: [eclipse]
- Power Supply:         [USB]

## Problem Description
 
Our application works as a BLE server, and it will send data to a BLE client(Android with BLE 5.0) by notification.
And we found if we turn off the BLE  on Android during the notification (not a normal operation), 
we can get error log ：

“BT_APPL: Unknown connection ID: 3 fail sending notification”
or
“BT_L2CAP: l2ble_update_att_acl_pkt_num not found p_lcb
BT_APPL: Unknown connection ID: 7 fail sending notification”

After the error, reconnect the device,  try to get the available pkg for notifacition  .
And you can find function"esp_ble_get_cur_sendable_packets_num()" will not return 10.

if you have this kind of error for 10 times,you can get 0 by "esp_ble_get_cur_sendable_packets_num()" ;

then we have to restart Esp32 

### Steps to reproduce
you can reproduce with the example "throughput_server"
1. add some log ：
`  case ESP_GATTS_CONNECT_EVT: {
       ...
        ESP_LOGI(GATTS_TAG, "***cur pkg = %d,total = %d ***",
        				esp_ble_get_cur_sendable_packets_num(param->connect.conn_id),
        				esp_ble_get_sendable_packets_num());

        break;
    }`
2. start notification for 2~3 mins
3. turn off the BLE on Android
4. repeat 1~3

xulongzhuang, 2021-07-16:
Hi, @BigSmileFace  
We can reproduce the problem you encountered, we are looking for the reason, and will reply you as soon as possible

BigSmileFace, 2021-07-16:
@xulongzhuang thanks！

xulongzhuang, 2021-11-26:
@BigSmileFace 
`esp_ble_get_cur_sendable_packets_num()` may still have some problems, can you use the ESP_GATTS_CONGEST_EVT event to determine whether it is blocked and whether it can be sent?

closed 2024-12-26 by Alvin1Zhang