espressif/esp-idf #7268: softAP won't assign IP Address when a pair of GPIOs is used for UART Communication

GeorgeFlorian, 2021-07-13:
Development Kit: ESP32-DevKit-V4
Module or chip used: ESP-WROVER-B
IDF version: ESP-IDF v4.4-dev-1849-g8e3e65a47
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
Operating System: Linux
Power Supply: USB

# Problem

I am trying to implement this project: https://github.com/DroneBridge/ESP32
I have tried using GPIOS 18,19,21 and 22 as RX/TX.

**Whenever I connect GPIOs to a Flight Controller that operates on 3v3, the softAP won't assign IP Addresses anymore.**

I have measured the voltage on the RX and TX pins on the Flight Controller and it is 3v3.
[image]

This is the verbose starting log:
```
I (1866) wifi:wifi firmware version: ff5f4ea
I (1866) wifi:wifi certification version: v7.0
I (1876) wifi:config NVS flash: enabled
I (1876) wifi:config nano formating: disabled
I (1886) wifi:Init data frame dynamic rx buffer num: 32
I (1886) wifi:Init management frame dynamic rx buffer num: 32
I (1896) wifi:Init management short buffer num: 32
I (1896) wifi:Init dynamic tx buffer num: 32
I (1906) wifi:Init static rx buffer size: 1600
I (1906) wifi:Init static rx buffer num: 10
I (1906) wifi:Init dynamic rx buffer num: 32
V (1916) esp_adapter: thread sem get: sem=0x3ffbb5e8
I (1916) wifi_init: rx ba win: 6
I (1926) wifi_init: tcpip mbox: 32
I (1926) wifi_init: udp mbox: 6
I (1926) wifi_init: tcp mbox: 6
I (1936) wifi_init: tcp tx win: 5744
I (1936) wifi_init: tcp rx win: 5744
I (1946) wifi_init: tcp mss: 1440
I (1946) wifi_init: WiFi IRAM OP enabled
I (1946) wifi_init: WiFi RX IRAM OP enabled
V (1956) esp_adapter: thread sem get: sem=0x3ffbb5e8
V (1956) esp_adapter: thread sem get: sem=0x3ffbb5e8
V (1966) esp_adapter: thread sem get: sem=0x3ffbb5e8
V (2466) esp_adapter: thread sem get: sem=0x3ffbb5e8
I (2466) wifi:set country: cc=XX schan=1 nchan=13 policy=1

V (2466) esp_adapter: thread sem get: sem=0x3ffbb5e8
D (2466) ADC: Wi-Fi takes adc2 lock.
I (2476) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
D (2476) phy_init: loading PHY init data from application binary
D (2486) nvs: nvs_open_from_partition phy 0
D (2486) nvs: nvs_get cal_version 4
V (2496) phy_init: phy_get_rf_cal_version: 4670

D (2496) nvs: nvs_get_str_or_blob cal_mac
D (2506) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (2506) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (2516) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (2526) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (2536) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
D (2536) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 0 bit
D (2546) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 16 bit
D (2556) nvs: nvs_get_str_or_blob cal_data
D (2566) nvs: nvs_close 4
D (2566) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 8 bit
D (2566) efuse: In EFUSE_BLK0__DATA2_REG is used 8 bits starting with 0 bit
D (2576) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 24 bit
D (2576) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 16 bit
D (2586) efuse: In EFUSE_BLK0__DATA1_REG is used 8 bits starting with 8 bit
[... 556 lines cut ...]
D (16476) wpa: my mic

D (16486) wpa: be 39 27 c8 ab 58 ab e1 5e 2f 4e fe f9 9a 11 4a 
D (16486) wpa: wpa_rx: free eapol=0x3ffc7ac0

D (16496) wpa: wpa_rx: new eapol=0x3ffc7ac0

D (16496) wpa: WPA: 58:96:1d:be:d4:30 WPA_PTK_GROUP entering state REKEYESTABLISHED

D (16506) wpa: WPA: 58:96:1d:be:d4:30 WPA_PTK_GROUP entering state IDLE

V (64056) esp_adapter: thread sem get: sem=0x3ffbcb2c
D (64056) esp_netif_sta_list: esp_netif_get_sta_list entered
D (64056) DB_CONTROL: 1 0.0.0.0 0.0.0.0 58:96:1d:be:d4:30
D (64306) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffc82b8
V (65256) esp_adapter: thread sem get: sem=0x3ffbcb2c
D (65256) esp_netif_sta_list: esp_netif_get_sta_list entered
D (65256) DB_CONTROL: 1 0.0.0.0 0.0.0.0 58:96:1d:be:d4:30
```

And then it keeps trying to connect.

I don't know what to make of these logs.

Why does WiFi get affected by GPIOs ?

xueyunfei998, 2021-08-05:
hi @GeorgeFlorian 
do you use the SoftAP example in IDF to test?
Example location：esp_idf/example/wifi/getting_started/softAP/

xueyunfei998, 2021-08-05:
From the failed log, the handshake has been repeated four times and the IP address has not been obtained.

I suggest testing with the latest IDF example

Alvin1Zhang, 2021-11-23:
@GeorgeFlorian Thanks for reporting, sorry for slow turnaround, would you please help share if any updates for the issue? Thanks.

GeorgeFlorian, 2021-11-23:
I don't remember.
I think that it either had something to do with the project that I was using or the fact that I was using pins related to the ESPS's WiFi module.

MaxwellAlan, 2022-03-29:
Hi @GeorgeFlorian 
Seems like that the GPIO will interference WiFi RF when there are some data bits stream trans in the GPIO.
Can you try when WiFi tx/rx , do not trans data in the GPIO at the same time?
Can you provide some more detail about what frequence that your GPIO tx/rx ?
Thanks.

xueyunfei998, 2022-04-07:
[uart_echo_example_main.txt](https://github.com/espressif/esp-idf/files/8439515/uart_echo_example_main.txt)

I wrote an example, configured serial port 2 with 19 and 18 pins, and the softap can be connected normally. Here are my test results.

[image]

hi @GeorgeFlorian 
Can you test it with my example? I look forward to your feedback.

GeorgeFlorian, 2022-04-07:
> 

Thank you for your answer, but I am not working on that project anymore.
If I will ever resume it I will come back to this.

closed 2022-04-07 by GeorgeFlorian