espressif/esp-idf #7272: wifi_apb80m_request or wdt(?) leads to crash and bricks device in field

thecashit, 2021-07-14:
- Module or chip used: ESP32-WROOM-32D
- IDF version v4.3
- Build System:  idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Power Supply: USB

## Problem Description
We encountered very unpleasant device behavior after upgrading to IDF 4.3. 
Moreover, this did not immediately manifest itself and the bug is floating.
Due to conditions that are still incomprehensible to us, sometimes the following output to the console appears:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c52a  PS      : 0x00060033  A0      : 0x80093898  A1      : 0x3ffefe60  
A2      : 0x3ffefe80  A3      : 0xffffffff  A4      : 0x0000004b  A5      : 0x3ffefeca  
A6      : 0x00000001  A7      : 0x00000004  A8      : 0x00000000  A9      : 0x3ffeffbc  
A10     : 0xffffffff  A11     : 0x3ffeffbc  A12     : 0x3fff0070  A13     : 0x3fff0050  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x4000c527:0x3ffefe60 0x40093895:0x3ffefe80 0x40094455:0x3ffeff00 0x40093434:0x3ffeff20 0x400933e6:0xffffffff |<-CORRUPTED
0x40093895: vPortExitCritical at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/port.c:472
0x40094455: vTaskSwitchContext at /Firmware/Toolchain/esp-idf/components/freertos/tasks.c:3290 (discriminator 3)
0x40093434: _frxt_timer_int at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:314
0x400933e6: _frxt_int_exit at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:185

ELF file SHA256: b3ff32802362285b

Rebooting...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40095acf  PS      : 0x00060433  A0      : 0x80082186  A1      : 0x3ffefc30  
0x40095acf: mwdt_ll_config_stage at /Firmware/Toolchain/esp-idf/components/hal/esp32/include/hal/mwdt_ll.h:99
 (inlined by) wdt_hal_config_stage at /Firmware/Toolchain/esp-idf/components/hal/wdt_hal_iram.c:121

A2      : 0xffffffff  A3      : 0x3801ff80  A4      : 0x000249f0  A5      : 0x3801fb80  
A6      : 0x00000001  A7      : 0x0000005f  A8      : 0xffffffff  A9      : 0x50d83aa1  
A10     : 0x00000000  A11     : 0x00000010  A12     : 0x00000008  A13     : 0x3ffefc68  
A14     : 0x3ffc22a3  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000063  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40095acc:0x3ffefc30 0x40082183:0x3ffefc50 0x400930ad:0x3ffefc90 0x40092b92:0x3ffefcb0 0x40092fda:0x3ffefd30 0x4009309d:0x3ffefd80 0x400827d6:0x3ffefda0 0x4000c527:0x3ffefe60 0x40093895:0x3ffefe80 0x40094455:0x3ffeff00 0x40093434:0x3ffeff20 0x400933e6:0xffffffff |<-CORRUPTED
0x40095acc: mwdt_ll_config_stage at /Firmware/Toolchain/esp-idf/components/hal/esp32/include/hal/mwdt_ll.h:99
 (inlined by) wdt_hal_config_stage at /Firmware/Toolchain/esp-idf/components/hal/wdt_hal_iram.c:121

0x40082183: esp_restart_noos at /Firmware/Toolchain/esp-idf/components/esp32/system_api_esp32.c:53
0x400930ad: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101
0x40092b92: esp_panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/panic.c:279 (discriminator 3)
0x40092fda: panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:191
0x4009309d: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101
0x400827d6: _xt_user_exc at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:640
0x40093895: vPortExitCritical at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/port.c:472
0x40094455: vTaskSwitchContext at /Firmware/Toolchain/esp-idf/components/freertos/tasks.c:3290 (discriminator 3)
0x40093434: _frxt_timer_int at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:314
0x400933e6: _frxt_int_exit at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:185

ELF file SHA256: b3ff32802362285b

Rebooting...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40095acf  PS      : 0x00060f33  A0      : 0x80082186  A1      : 0x3ffefa00  
0x40095acf: mwdt_ll_config_stage at /Firmware/Toolchain/esp-idf/components/hal/esp32/include/hal/mwdt_ll.h:99
 (inlined by) wdt_hal_config_stage at /Firmware/Toolchain/esp-idf/components/hal/wdt_hal_iram.c:121

A2      : 0xffffffff  A3      : 0x3801ff80  A4      : 0x000249f0  A5      : 0x3801ff80  
A6      : 0x00000001  A7      : 0x00000058  A8      : 0xffffffff  A9      : 0x50d83aa1  
A10     : 0x00000000  A11     : 0x00000010  A12     : 0x00000008  A13     : 0x3ffefa38  
A14     : 0x3ffc22a3  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000063  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40095acc:0x3ffefa00 0x40082183:0x3ffefa20 0x400930ad:0x3ffefa60 0x40092b92:0x3ffefa80 0x40092fda:0x3ffefb00 0x4009309d:0x3ffefb50 0x400827d6:0x3ffefb70 0x40095acc:0x3ffefc30 0x40082183:00x40095acc: mwdt_ll_config_stage at /Firmware/Toolchain/esp-idf/components/hal/esp32/include/hal/mwdt_ll.h:99
 (inlined by) wdt_hal_config_stage at /Firmware/Toolchain/esp-idf/components/hal/wdt_hal_iram.c:121

0x40082183: esp_restart_noos at /Firmware/Toolchain/esp-idf/components/esp32/system_api_esp32.c:53

0x400930ad: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101

0x40092b92: esp_panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/panic.c:279 (discriminator 3)

0x40092fda: panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:191

0x4009309d: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101

0x400827d6: _xt_user_exc at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:640

0x40095acc: mwdt_ll_config_stage at /Firmware/Toolchain/esp-idf/components/hal/esp32/include/hal/mwdt_ll.h:99
 (inlined by) wdt_hal_config_stage at /Firmware/Toolchain/esp-idf/components/hal/wdt_hal_iram.c:121

0x40082183: esp_restart_noos at /Firmware/Toolchain/esp-idf/components/esp32/system_api_esp32.c:53

x3ffefc50 0x400930ad:0x3ffefc90 0x40092b92:0x3ffefcb0 0x40092fda0x400930ad: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101

0x40092b92: esp_panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/panic.c:279 (discriminator 3)

0x40092fda: panic_handler at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:191

:0x3ffefd30 0x4009309d:0x3ffefd80 0x400827d6:0x3ffefda0 0x4000c50x4009309d: print_state at /Firmware/Toolchain/esp-idf/components/esp_system/port/panic_handler.c:101

0x400827d6: _xt_user_exc at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:640

27:0x3ffefe60 0x40093895:0x3ffefe80 0x40094455:0x3ffeff00 0x40090x40093895: vPortExitCritical at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/port.c:472

0x40094455: vTaskSwitchContext at /Firmware/Toolchain/esp-idf/components/freertos/tasks.c:3290 (discriminator 3)

3434:0x3ffeff20 0x400933e6:0xffffffff |<-CORRUPTED
0x400933e6: _frxt_int_exit at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:185

The main problem here that devices are already in field and we have already several reports that they was bricked.
Is this is a ESP-IDF bug or heap corruption?

I also attached sdkconfig if it helps to answer this question
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6815913/sdkconfig.txt)

thecashit, 2021-07-14:
Addition: after enabling heap debugging i can found that stack overflow(?) happen here in bt.c:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4000c52a  PS      : 0x00060033  A0      : 0x80093cb0  A1      : 0x3ffd4d10  
A2      : 0x3ffd4d30  A3      : 0xffffffff  A4      : 0x0000004b  A5      : 0x3ffd4d7a  
A6      : 0x00000001  A7      : 0x00000004  A8      : 0x00000000  A9      : 0x3ffd4e6c  
A10     : 0xffffffff  A11     : 0x3ffd4e6c  A12     : 0x3ffd4f50  A13     : 0x3ffd4f20  
A14     : 0x3ffc4634  A15     : 0x4011fc78  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
**0x4011fc78: queue_send_wrapper at /Firmware/Toolchain/esp-idf/components/bt/controller/esp32/bt.c:750**

EXCVADDR: 0xffffffff  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x4000c527:0x3ffd4d10 0x40093cad:0x3ffd4d30 0x4009486d:0x3ffd4db0 0x4009384c:0x3ffd4dd0 0x400937fe:0xffffffff |<-CORRUPTED
0x40093cad: vApplicationStackOverflowHook at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/port.c:487 (discriminator 3)

0x4009486d: vTaskSwitchContext at /Firmware/Toolchain/esp-idf/components/freertos/tasks.c:3290

0x4009384c: _frxt_dispatch at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:432

0x400937fe: _frxt_int_exit at /Firmware/Toolchain/esp-idf/components/freertos/port/xtensa/portasm.S:231

thecashit, 2021-07-14:
Addition:
It seems that such a stack overflow and device death occurs only on Mikrotik routers with the Power Management function enabled on the device.

Everything is fine on other routers

negativekelvin, 2021-07-14:
By brick do you mean it won't boot even after a power cycle?

thecashit, 2021-07-14:
@negativekelvin 
Exactly! 
Only one thing we can do in this case - physically disassemble device and reflash it

negativekelvin, 2021-07-15:
Did you analyze flash dump to identify corruption?

thecashit, 2021-07-15:
@negativekelvin not yet, planned for today.
I think that it might be divide into 2 issues:
1) Flash corruption ( I will try to find dump difference today)
2) Stack overflow somewhere on Wi-Fi internal task I guess. But happened only on specified routers. it feels like it happens 20-30 seconds after intense network work

thecashit, 2021-07-15:
@negativekelvin so... we found problem
It very rare thing and i didn't understand why it is arise only on Mikrotik routers.
So, the problem is invoke esp_partition_erase_range with 0 in third param.

As i might propose - it will lead to full flash erase.
Am i right?

negativekelvin, 2021-07-16:
Seems possible yes due to no zero check and underflow

https://github.com/espressif/esp-idf/blob/9d34a1cd42f6f63b3c699c3fe8ec7216dd56f36a/components/spi_flash/esp_flash_api.c#L501

Alvin1Zhang, 2021-07-29:
Thanks for reporting, we had a fix under internal reviewing. The issue will be closed automatically once the fix is available on GitHub. Thanks.

thecashit, 2021-07-29:
@Alvin1Zhang Thank you!

Alvin1Zhang, 2021-07-30:
Fix on master is available at https://github.com/espressif/esp-idf/commit/58022f8599401740979ac7657a57348285a70374, we are now back porting to other release branches, will keep updating.

negativekelvin, 2021-07-31:
The bricking part is fixed but cause of memory corruption still unknown whether in user code or idf code

AxelLin, 2022-01-16:
> Fix on master branch is available at [58022f8](https://github.com/espressif/esp-idf/commit/58022f8599401740979ac7657a57348285a70374), we are now back porting to other release branches, will keep updating.

It has been a while (> 5 months) , just wondering how is the status of the backport fix for release branches.

igrr, 2022-01-16:
Thanks for the reminder @AxelLin. My bad, I was supposed to backport this. I have created the backport to release/v4.3 just now, it should be possible to get it released in v4.3.3. Older releases don't have this bug (the loop directly checks the input argument `len` [here](https://github.com/espressif/esp-idf/blob/85a36b86d0dfa53e1a24cdf76cdd9843bd1f6377/components/spi_flash/esp_flash_api.c#L345)).

AxelLin, 2022-03-14:
This is marked as "Resolution: Done" for quite long time, but the fix for v4.3 is still not yet available.

igrr, 2022-03-22:
release/v4.3 backport has been merged in https://github.com/espressif/esp-idf/commit/f4852bc055350a2213035f052177f91c81113157. I'm sorry for the delay!

closed 2022-03-22 by igrr