espressif/esp-idf #7275: uart: uart_read_bytes(1194): uart data null from UART read while trying ti switch back and forth from 2 different tasks

GunaGShekar, 2021-07-15:
OS: Ubuntu-mate (20.04)
ESP-IDF version: ESP-IDF v4.3-dev-2587-gc15b1e9d5-dirty
RTOS: FreeRTOS (available with the IDF)
ESP MCU: ESP WROOM 32

Hello, 

I am doing a project which requires me to do switch between 2 different tasks. In one of the tasks, I have connected a GSM module to which I will communicate using AT commands via UART. 

The example code snippet is like below: 
`void task1(){
//A task that doesn't concern the issue
  if(condition){
      switch_to_task2();
  ]
//Rest of the code in the local task
}

void task2(){
//A task in which I am facing the issue
  if(condition){
        switch_to_task1();
   }
  else{
    //code for communicating with LTE
    switch(cond){
       case RESET:
        txBytes = uart_write_bytes(UART_NUM_2, RESET_cmd, len);
         delay();
        rxBytes = uart_read_bytes(UART_NUM_2, GSM_RX_BYTES, RX_BUF_SIZE, 1000 / portTICK_RATE_MS);
         //Rest of the code in this particular statement

       case X:
         break;

        case XX:
         break;

    }
  }
}
`
The configuration for UART_NUM_2 (which I am using for GSM) is as below:

`
uart_config_t uart_config_2 = {
	.baud_rate = 128000,
	.data_bits = UART_DATA_8_BITS,
	.parity = UART_PARITY_DISABLE,
	.stop_bits = UART_STOP_BITS_1,
	.flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
	//.rx_flow_ctrl_thresh = 122,
};
`
The Declaration of buffers are as below:
RX_BUF_SIZE = 4096
GSM_RX_BYTES = (char*) malloc(RX_BUF_SIZE+1);

The code works fine for the first 4 switches.

The problem comes at the next switch from task A to task B and in task B it executes the first statement (i.e., sending a command via UART). Now, I get **uart: uart_read_bytes(1194): uart data null** error. 

I tried to uninstall and reinstall the UART driver, freed the buffer variable, and reinitialized it. Nothing seems to be working.

Even if I try to reboot the device or I try to delete the task and create again, the problem persists.

If anyone knows the solution to this problem, please do let me know. Thanks in advance.

ginkgm, 2021-07-16:
Hi @GunaGShekar ,

Firstly, the error report looks very simple, and unrelated to thread safety. Could you please help check if you are changing GSM_RX_BYTES by the time, and is it always non-NULL? One thing you can do is that, replace the call of uart_read_bytes by an `assert(GSM_RX_BYTES  != NULL)` and see what it happens.

And I cannot find the IDF version (c15b1e9d5) in the IDF project. If we have to do further debug, we will need a valid IDF version.

GunaGShekar, 2021-07-16:
Hello @ginkgm ,

Thanks for the response :)

ESP-IDF version I am using is: ESP-IDF v4.3-dev-2587-gc15b1e9d5-dirty [when I type idf.py --version I get the mentioned version]

If you need additional info about the build environment, here it is:
Toolchain version: esp-2020r3
Compiler version: 8.4.0
esptool.py v3.1-dev

Also I tried the GSM_RX_BYTES becoming NULL or not by using the assert statement before the uart read operation. something like this: 
`
assert(GSM_RX_BYTES != NULL);
printf("assert statement has been given before uart read \n");
AT_rxBytes = uart_read_bytes(UART_NUM_2, GSM_RX_BYTES, RX_BUF_SIZE, 1000 / portTICK_RATE_MS);
`

AxelLin, 2021-07-16:
> ESP-IDF version I am using is: ESP-IDF v4.3-dev-2587-gc15b1e9d5-dirty [when I type **idf.py --version**, I get the mentioned version]
> 

The point is commit c15b1e9d5 does not exist in official esp-idf.
Looks like you have some commits on top of esp-idf.
You need to tell the latest commit before your changes.

> If you need additional info about the build environment, here it is:
> Toolchain version: esp-2020r3
> Compiler version: 8.4.0
> esptool.py v3.1-dev

Please always follow the template and provide above information which will save your time.

GunaGShekar, 2021-07-17:
Hello,

I have downloaded the ESP-IDF Version to **ESP V4.3** and tried the same source code. Still no luck.

Without switching the task everything works fine and I have stress tested it for more than 2 days. But with switching between the tasks, I am getting UART null. Even if I restart the device the error persists.

If I reset the GSM module, then I am able to send commands again indicating that the issue is not with the hardware connection

All the software build environments remain the same as mentioned in the above 2 threads except the ESP-IDF version.

The sample program also remains the same as I mentioned in the original question.

No luck in resolving the issue till now. :(

Any lead would be greatly appreciated.

ginkgm, 2021-07-17:
It still looks like a concurrency issue, which means even you can do it for once, you cannot ensure its safety. The final solution is to rewrite your code until you are able to easily understand the flow among the threads. Or in other words, if you cannot draw the synchronization relationship among the tasks, the code is not maintainable enough for you.

Use a stricter model when modifying the `GSM_RX_BYTES` variable. One thing you may try is: use a queue to send a request to the receiving task, and fetch the result from the task when it's done. So that the receiving task is the only one that may touch the variable, and the variable it get form the queue is always valid. 

Or you may even use a more strict model: put everything that may touch that variable into one task. This may not fix the issue, but it will avoid any random issue due to the asychronization of the `GSM_RX_BYTES` touching task and the receiving task.