espressif/esp-idf #7276: GPIO set

jigsurfer, 2021-07-16:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jigsurfer, 2021-07-16:
void app_main (void)
.....
    gpio_config_t io_conf = {};
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = GPIO_SEL_2 | GPIO_SEL_4 | GPIO_SEL_5 ;
    io_conf.pull_down_en = 1;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    gpio_pad_select_gpio(GPIO_INOUTPUT_IO_1);
    gpio_pad_select_gpio(GPIO_INOUTPUT_IO_2);
    gpio_pad_select_gpio(GPIO_INOUTPUT_IO_3);
 

    /* Set the GPIO pull mode */
    gpio_set_pull_mode(GPIO_INOUTPUT_IO_1, GPIO_PULLDOWN_ONLY);
    gpio_set_pull_mode(GPIO_INOUTPUT_IO_2, GPIO_PULLDOWN_ONLY);
    gpio_set_pull_mode(GPIO_INOUTPUT_IO_3, GPIO_PULLDOWN_ONLY);

Output:

Slave starting in 3
Slave starting in 2
Slave starting in 1
␛[0;32mI (6301) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (6301) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;32mI (6301) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 ␛[0m
␛[0;31mE (6431) gpio: gpio_set_pull_mode(249): GPIO number error␛[0m
␛[0;32mI (6441) Device1: Driver installed␛[0m
␛[0;32mI (6441) Device1: Driver started␛[0m

Hi,
i configured almost everything are right but it seem the gpio_set_pull_mode always notifying error. I tried to look for this issue online but unsuccessful.

␛[0;31mE (6431) gpio: gpio_set_pull_mode(249): GPIO number error␛[0m 

code running ok in order modules but only stage after bit_mask setting. What is wrong ? plz kindly help!

Alvin1Zhang, 2021-07-19:
@jigsurfer Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

Dazza0, 2021-09-27:
@jigsurfer 
```
[0;31mE (6431) gpio: gpio_set_pull_mode(249): GPIO number error␛[0m
```
Seems like you're getting a GPIO number error, meaning you're inputting an invalid GPIO number to `gpio_set_pull_mode()`. I suggest you double check that your `GPIO_INOUTPUT_IO_X` macros define to valid GPIOs. For specifics on valid GPIOs, see the `GPIO_IS_VALID_GPIO` macro [here](https://github.com/espressif/esp-idf/blob/3e370c4296247b349aa3b9a0076c05b9946d47dc/components/driver/include/driver/gpio.h#L50), or refer to the datasheet of the ESP32.

I don't think this is related to the GPIO driver so I'm closing this issue. Please feel free to reopen.

closed 2021-09-27 by Dazza0