espressif/esp-idf #7277: SPP_INITIATOR_DEMO / SPP_ACCEPTOR_DEMO ASSERT_WARN() when devices moved out of range

BenjaminWKidd, 2021-07-16:
## Environment

- Development Kit:     [ESP32-DevKitC|
- Kit version for [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version: [v4.2]
- Build System:         [CMake]
- Operating System:     [Windows]
- Using an IDE?: [Yes (eclipse)]
- Power Supply:         [USB]

## Problem Description

When SPP_ACCEPTOR_DEMO and SPP_INITIATOR_DEMO examples are loaded onto ESP32 DevKitC boards and the boards are moved to apart until the devices are out of range of one another, various ASSERT_WARN() and bta_dm_pm_btm_status hci_status errors appear. Devices appear to cease operation until either both devices are reset, or device running INITIATOR code is reset twice.

### Expected Behavior

I would expect devices to show disconnection warning and (ideally) reconnect and resume normal operation after they are moved to within range of one another (I realize this may require additional code, though I have made an initial attempt at this in my version of the code included below).

### Actual Behavior

Debug output shows "ASSERT_WARN(51 9), in lc_task.c at line 1298E (5587820) BT_APPL: bta_dm_pm_btm_status hci_status=32" and similar other ASSERT_WARN / bta_dm_pm_btm_status hci_status errors. Once these errors appear, only a hard reset (pushbutton or re-powering) allows them to resume normal operation as described above.

### Steps to reproduce

1. Load EXAMPLE_SPP_ACCEPTOR_DEMO bluetooth classic IDF v4.2 example project and (optionally) overwrite with my version of the EXAMPLE_SPP_ACCEPTOR_DEMO.c file, build and load onto an ESP32-WROOM-32D equipped ESP32-DevKitC board.
2. Do the same with the EXAMPLE_SPP_INITIATOR_DEMO example project.
3. Connect both boards to two different computers (and observe the debug output). Move one of the boards away from the other until previously reported output shows up. Note this distance was around 150 ft thru multiple walls in an indoor environment for me.

### Code to reproduce this issue

<script src="https://gist.github.com/BenjaminWKidd/edc9a463ca89194421b2d0d12adb5031.js"></script>

## Debug Logs (SPP_ACCEPTOR_DEMO)

W (5517460) BT_APPL: new conn_srvc id:26, app_id:255
I (5517470) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
I (5517720) SPP_ACCEPTOR_DEMO: speed(2283.754639s ~ 2284.009766s): 2824.946533 kbit/s
I (5517730) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:0
I (5518710) SPP_ACCEPTOR_DEMO: speed(2284.009766s ~ 2285.003662s): 1091.703247 kbit/s
I (5518720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-5
I (5519720) SPP_ACCEPTOR_DEMO: speed(2285.003662s ~ 2286.004883s): 1249.834351 kbit/s
I (5519720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-6
I (5520710) SPP_ACCEPTOR_DEMO: speed(2286.004883s ~ 2287.003662s): 1189.452026 kbit/s
I (5520720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-6
I (5521720) SPP_ACCEPTOR_DEMO: speed(2287.003662s ~ 2288.006104s): 1090.298096 kbit/s
I (5521720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-5
I (5522720) SPP_ACCEPTOR_DEMO: speed(2288.006104s ~ 2289.004883s): 1141.873901 kbit/s
I (5522720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-5
I (5523720) SPP_ACCEPTOR_DEMO: speed(2289.004883s ~ 2290.009766s): 1205.871948 kbit/s
I (5523720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-5
I (5524710) SPP_ACCEPTOR_DEMO: speed(2290.009766s ~ 2291.000977s): 1070.690430 kbit/s
I (5524720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-6
I (5525720) SPP_ACCEPTOR_DEMO: speed(2291.000977s ~ 2292.004883s): 1151.820801 kbit/s
I (5525720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-6
I (5526710) SPP_ACCEPTOR_DEMO: speed(2292.004883s ~ 2293.003662s): 1229.100342 kbit/s
I (5526720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-5
I (5527730) SPP_ACCEPTOR_DEMO: speed(2293.003662s ~ 2294.017334s): 1140.724121 kbit/s
I (5527730) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-3
I (5528720) SPP_ACCEPTOR_DEMO: speed(2294.017334s ~ 2295.005859s): 1217.813965 kbit/s
I (5528720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-2
I (5529710) SPP_ACCEPTOR_DEMO: speed(2295.005859s ~ 2296.002197s): 1009.537048 kbit/s
I (5529720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-2
I (5530720) SPP_ACCEPTOR_DEMO: speed(2296.002197s ~ 2297.003662s): 1170.445557 kbit/s
I (5530720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-7
I (5531720) SPP_ACCEPTOR_DEMO: speed(2297.003662s ~ 2298.007080s): 1199.739380 kbit/s
I (5531730) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-13
I (5532710) SPP_ACCEPTOR_DEMO: speed(2298.007080s ~ 2299.003418s): 985.689697 kbit/s
I (5532720) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-16
I (5533720) SPP_ACCEPTOR_DEMO: speed(2299.003418s ~ 2300.005859s): 963.886780 kbit/s
[... 171 lines cut ...]
I (33645) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-42
I (34515) SPP_INITIATOR_DEMO: speed(33.120911s ~ 34.000416s): 171.096222 kbit/s
I (34525) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-41
I (35865) SPP_INITIATOR_DEMO: speed(34.000416s ~ 35.347431s): 135.232315 kbit/s
I (35875) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-44
I (36815) SPP_INITIATOR_DEMO: speed(35.347431s ~ 36.298412s): 99.938889 kbit/s
I (36825) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-44
I (37555) SPP_INITIATOR_DEMO: speed(36.298412s ~ 37.035988s): 128.854599 kbit/s
I (37555) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-45
I (38605) SPP_INITIATOR_DEMO: speed(37.035988s ~ 38.089123s): 270.734528 kbit/s
I (38615) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-45
I (39515) SPP_INITIATOR_DEMO: speed(38.089123s ~ 39.000095s): 278.208160 kbit/s
I (39525) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-46
I (40535) SPP_INITIATOR_DEMO: speed(39.000095s ~ 40.015327s): 171.625778 kbit/s
I (40535) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-46
I (41575) SPP_INITIATOR_DEMO: speed(40.015327s ~ 41.059662s): 182.010666 kbit/s
I (41585) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-47
I (42745) SPP_INITIATOR_DEMO: speed(41.059662s ~ 42.225822s): 122.247314 kbit/s
I (42745) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-47
I (45635) SPP_INITIATOR_DEMO: speed(42.225822s ~ 45.120819s): 49.243584 kbit/s
I (45645) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-50
E (60245) BT_APPL: bta_dm_pm_btm_status hci_status=12
I (66925) SPP_INITIATOR_DEMO: speed(45.120819s ~ 66.405518s): 4.465180 kbit/s
I (66925) esp_bt_gap_cb(): ESP_BT_GAP_READ_RSSI_DELTA_EVT RSSI:-52
ASSERT_WARN(1 6), in lc_task.c at line 4658E (72135) BT_APPL: bta_dm_pm_btm_status hci_status=12

BetterJincheng, 2021-09-02:
Hi
I tried to follow your steps to reproduce the scenario but falied. Here is my LOGs:
```c
I (8161) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
I (10511) SPP_ACCEPTOR_DEMO: speed(7.669840s ~ 10.014300s): 695.904236 kbit/s
I (13501) SPP_ACCEPTOR_DEMO: speed(10.014300s ~ 13.006701s): 632.562256 kbit/s
I (16501) SPP_ACCEPTOR_DEMO: speed(13.006701s ~ 16.004412s): 782.036865 kbit/s
I (19511) SPP_ACCEPTOR_DEMO: speed(16.004412s ~ 19.015560s): 991.595093 kbit/s
I (22491) SPP_ACCEPTOR_DEMO: speed(19.015560s ~ 22.000582s): 894.144287 kbit/s
W (28731) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
I (28731) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
```

I just booted two ESP32 and put one into the shielding box to act  as 'out of range'. But it did not cause any error logs.  Any suggestion ?
Thanks!

BenjaminWKidd, 2021-09-03:
Hi @BetterJincheng,

Unfortunately its been a while since I posted this, so I'll have to dig it back up and see if I can make it happen again. I'm pretty sure if you just take the \esp-idf-2\examples\bluetooth\bluedroid\classic_bt\bt_spp_acceptor and bt_spp_initiator and put enough distance between the two so that they go out of range, that you will see the assert(), but I will have to double check that I can reproduce it. I've since moved on to trying out esp-now for this application.

closed 2021-12-23 by Alvin1Zhang