espressif/esp-idf #7278: uploading bin file to the file server

Benedito821, 2021-07-17:
I am using the example **Simple HTTP File Server Example** to send the blink example .bin file through wi-fi. In readme is shown how to upload a HTML file. I changed the file extension in the example to .bin like this: 
_curl -X POST --data-binary @blink.bin 192.168.43.130:80/upload/dev/ttyUSB0/blink.bin_ 
(192.168.43.130:80 just for example).  The file is uploaded successfuly but on the terminal it shows that it has 0 bytes. What I am missing?

mahavirj, 2021-07-19:
>  The file is uploaded successfuly but how to run the blink example on my board?

If you would like to upload and execute new firmware then you can try OTA examples as specified at https://github.com/espressif/esp-idf/blob/master/examples/system/ota/README.md

Benedito821, 2021-08-11:
Hello, @mahavirj !The example you have suggested has worked. But how could I upload my firmware more easily , that is, using a web page(IP address) where I just upload the bin and then flash them? Is there an example that can flash this way?

closed 2021-08-22 by Benedito821