espressif/esp-idf #7279: Route CRITICAL RESET errors as ESP_LOGE

tsctrl, 2021-07-17:
hi espressif dev,

core dump generation has two option, either to print in uart or flash. both option are not feasible due to require physical access to the device.

current stack errors that cause by any guru meditation errors are displayed to uart stdout, but probably is using different approach to push to uart than vprintf.

similar with dump generation to uart0 stdout. it does print in uart but is not captured by the esp_log_set_vprint.

requesting to add the stack errors and core dump generation to be captured by esp_log_set_vprint so remote error reporting and error detection is possible.

cc: @igrr, @mahavirj

igrr, 2021-07-17:
Hi @tsctrl,

> both option are not feasible due to require physical access to the device.

The "core dump to flash" option can be used to send the core dump file to a remote server after the crash. This is possible because you have programmatic access to coredump partition in flash, so you can read the core dump data and send it over a protocol of your choice.

There are a couple of managed services which provide  functionality for sending, collecting, and analyzing core dumps in the cloud. ESP Insights has been announced by Espressif just recently: https://github.com/espressif/esp-insights, https://medium.com/the-esp-journal/introducing-esp-insights-7d85a1383d21.
There is also Memfault, which can be used on Espressif chips with ESP-IDF: https://github.com/memfault/memfault-firmware-sdk, https://docs.memfault.com/docs/mcu/esp32-guide.

tsctrl, 2021-07-17:
Hi,

this is more than required. what i think to add which as simple as printing the hardware reset cause and capturing dump in plain text. 
uploading dump to server will take additional process, hardware and cost with limitation only for device connected to network. what i plan to have is ability to log the dump as plain text in current device or flash, user will be able to download the dump and submit the dump for analysis. if we able to read and output dump from the flash than this could be the approach i am going to proceed but yet this require dedicated partition then forwarding the uart output to a plain text.

for now i am not able to capture error dump from the callback function i created using esp_log_set_vprint 
i will check the managed service solution you shared which could be the possible use cases that i can use in the future.

thanks @igrr!

igrr, 2021-07-17:
> what i plan to have is ability to log the dump as plain text in current device or flash

Unfortunately decoding of the core dump still cannot fully be done on the device, as that requires access to DWARF debug information not normally added into the firmware binary. This is why we need to keep the core dump in flash and upload it to cloud for decoding.

When you enable core dump to UART and see the decoded core dump text on crash, that decoding is not actually done on the device. The device dumps the base64 core dump file to UART, and idf_monitor.py detects and decodes that core dump, showing you only the decoded result.

tsctrl, 2021-07-17:
Hi @igrr,

thats to conclude the only way i saw other then uploading to the cloud service is to expose restful api for user to download the dump file for analysis. this will be the goto approch i would use for now. 

thanks for clarification and your effort to answer the question. 

thanks!

closed 2021-07-18 by igrr