espressif/esp-idf #7281: ESP32-C3 I2C read failure results in crash

eyewy, 2021-07-17:
On ESP32-C3, if for whatever reason I2C read fails, program crashes with following error code:

E (11861) task_wdt: Task watchdog got triggered. The following tasks did not res
et the watchdog in time:
E (11861) task_wdt:  - IDLE (CPU 0)
E (11861) task_wdt: Tasks currently running:
E (11861) task_wdt: CPU 0: my_i2c_service

eyewy, 2021-07-17:
closing to gather more info about the issue

eyewy, 2021-07-18:
As stated above, I tried many combinations and have been able to isolate the issue to I2c.

o-marshmallow, 2021-11-17:
Hello @eyewy ,
Do you mean that you specified `my_i2c_service` as the callback for `xTaskCreate`?
Can you please provide me a small snippet of code to reproduce this?

o-marshmallow, 2021-11-22:
@eyewy Any update?

AxelLin, 2022-02-13:
@eyewy 
Is this still an issue?
It has been a while since you reported, could you try recently updated esp-idf version.

BTW, it's hard to find the exactly commit of the "latest SP-IDF master" on 17 Jul 2021,
please always report esp-idf version using "git describe --tags" command.

Sampath-Science, 2022-03-17:
I am also facing similar issue. I2C is not working with ESP32C3. I had tried with the i2c_tools example to scan i2c devices but it ended with the below message 
"E (889558) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (889558) task_wdt:  - IDLE (CPU 0)
E (889558) task_wdt: Tasks currently running:
E (889558) task_wdt: CPU 0: console_repl
"

Version: IDF 4.4

AxelLin, 2022-04-08:
@Sampath-Science 
https://github.com/espressif/esp-idf/issues/8120

Alvin1Zhang, 2022-05-16:
Thanks for reporting and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/0b00831703ec351c4c3b1040b004b358e4396067, we are also back porting to release branches. 

Feel free to reopen.

closed 2022-05-16 by Alvin1Zhang