espressif/esp-idf #7282: Incorrect return values for pow(-1,nan) and tgamma(-inf)

dpgeorge, 2021-07-18:
## Environment

- Module or chip used:  ESP32-PICO-D4 and ESP32C3
- IDF version: v4.4-dev-2128-gf45b60b28
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
- Operating System: Linux

## Problem Description

The return value of `pow(-1, NAN)` and `tgamma(-INFINITY)` are incorrect (according to standard C function semantics).

### Expected Behavior

- `pow(-1, NAN)` should return `NAN`; see https://pubs.opengroup.org/onlinepubs/9699919799/functions/pow.html
- `tgamma(-INFINITY)` should return `NAN`; see https://pubs.opengroup.org/onlinepubs/9699919799/functions/tgamma.html

### Actual Behavior

- `pow(-1, NAN)` returns `-1.000`
- `tgamma(-INFINITY)` returns `INFINITY`

### Steps to reproduce / code

Create a simple program (eg based on hello_world) and add these lines:
```c
    printf("pow(-1, nan) = %f\n", pow(-1, NAN));
    printf("tgamma(-inf) = %f\n", tgamma(-INFINITY));
```

Then compile, flash, run, and inspect output.

0xjakob, 2021-11-15:
@dpgeorge I think I found the issue with the `pow()` function. It seems to be a toolchain bug in newlib which has been fixed in newer versions than the one IDF uses. I'll check `tgamma()` as well. We'll probably provide a quick fix for IDF versions and/or include the fix in the newer IDF toolchain releases.

0xjakob, 2021-11-16:
It's been fixed, but only very recently: https://sourceware.org/git/?p=newlib-cygwin.git;a=commit;h=bb25dd1b0f39f343b764fd0db861cb9a30441407

zikalino, 2022-02-24:
@dpgeorge we should have a fix for this issue very soon, as we will be upgrading toolchains to 2022r1.

closed 2022-06-11 by espressif-bot

0xjakob, 2022-11-14:
@dpgeorge Note that the fix for `tgamma()` is only available on v5.0 (will be released soon). The other versions are stuck with an older toolchain version which brings an older newlib version that can't easily be fixed. In contrast, `pow()` has been fixed on v4.4 as well.

dpgeorge, 2022-11-14:
OK, thanks for fixing!