espressif/esp-idf #7283: unable to read coredump from flash in windows

tsctrl, 2021-07-19:
Hi Espressif Dev,

i am unable to read generated coredump in the flash using espcoredump.py

i have core dump generated as elf:
```
I (13504) esp_core_dump_flash: Save core dump to flash...
I (13578) esp_core_dump_flash: Erase flash 16384 bytes @ 0x510000
I (13839) esp_core_dump_flash: Write end offset 0x3384, check sum length 4
I (13839) esp_core_dump_flash: Core dump has been saved to flash.
```

this is my core dump partition:
coredump,   data, coredump, 0x510000,  0x190000,

read flash command: 
`esptool.py --chip esp32 --port COM11 --baud 115200 read_flash 0x510000 0x190000 coredump.elf `

there is two possible command to run is to read from flash or to read from elf? both not shownmg any details:

1. read from flash directly, i not sure why i need to supply apps '.elf' because we just need to read elf from flash and decode:
> python espcoredump.py -p COM11 info_corefile build/apps.elf
espcoredump.py v0.4-dev
INFO: Invoke parttool to read image.

result: stuck for sometime and then below error was returned:
```
subprocess.CalledProcessError: Command '[python.exe', 'esptool.py', '--port', 'COM11', 'read_flash', '32768', '3072', 'C:\\Users\\user\\AppData\\Local\\Temp\\tmp5w0vyodr']' returned non-zero exit status 2.
ERROR: parttool script execution failed with err 1
INFO: b"esptool.py v3.1-dev\r\nSerial port COM11\r\nConnecting........_____....._____....._____....._____....._____....._____....._____\r\n\r\nA fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header\r\
raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> tot_len
stream read less than specified amount, expected 4, found 0
```
i am not using devkit board where i have to push button to enter the boot loader/flash modes, this script expect we have the auto program board?

2. read core elf files, when you read the flash using esptools and have the core dump ready, i not sure should i include elf from build folder and not sure what is prog location means, should use esp prog?

in documentation say: <prog> Path to program ELF file.
>%IDF_PATH%\components\espcoredump\python espcoredump.py info_corefile -t elf --off 5308416 --core coredump.elf prog 
```
raise ConstError("parsing expected %r but parsed %r" % (self.value, obj), path=path)
construct.core.ConstError: Error in path (parsing) -> elf_header -> e_ident -> EI_MAG
parsing expected b'\x7fELF' but parsed b'$6\x00\x00'
```

other than this issue what is the optimal size for core dump partition? 
documentation: core dump data structure introduces constant overhead of 20 bytes and per-task overhead of 12 bytes. This overhead does not include size of TCB and stack for every task. So partition size should be at least 20 + max tasks number x (12 + TCB size + max task stack size) bytes.

from what the doc mention, should count all task and figure out internal task size,  and not sure what is TCB. can put this term to layman english?

esp-idf-v4.3.1

thanks!

AxelLin, 2021-07-19:
esp-idf-v4.3.1 is not yet available, is that version a typo?

tsctrl, 2021-07-19:
Hi @AxelLin,

sorry, a typo

Operating System: windows 10
Java Runtime Version: 15.0.2+7-27
Eclipse Version: 4.19.0.v20210303-1800
Eclipse CDT Version: 10.2.0.202012191711
IDF Eclipse Plugin Version: 2.1.2.202106231127
ESP-IDF v4.3.0
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.9.0

negativekelvin, 2021-07-19:
https://github.com/espressif/esp-idf/blob/9d34a1cd42f6f63b3c699c3fe8ec7216dd56f36a/components/freertos/tasks.c#L256

igrr, 2021-07-19:
@tsctrl you might have a confusion between ELF file, firmware binary, and core dump.

1. The application ELF file is produced by the linker during the build process. This file is located in your project build directory and has `.elf` extension. This file contains all the debug information (in DWARF format) necessary to perform core dump decoding.
2. The application binary file is produced by `esptool.py elf2image` command during the build process, is located in your project build directory and has `.bin` extension. This file contains everything necessary for the program to run on the chip, but doesn't contain debug information. When you flash the program to the chip, this file gets written to flash, in addition to the bootloader and the partition table.
3. Core dump binary is produced by the program running in the chip when a crash happens. This file _happens_ to use `ELF` as a container format, but it is different from your program ELF file. The core dump binary can be read from Flash and then decoded into a human-readable form by espcoredump.py. In addition to the core file to perform the decoding the application ELF file is also necessary.

To give you some practical advice on working with core files:

1. If you want to read the core file from flash and process it, the easiest way is to use `espcoredump.py` and specify the serial port, for example:
   ```
   $ espcoredump.py --port /dev/ttyUSB0 --baud 921600 info_corefile build/your_app_name.elf
   ```
   This will read the partition table from the device, find the core dump partition, read the core file, decode it, and print the output in human-readable format.
2. If you want to split "reading coredump from flash" and "decoding coredump" into two separate stages, then the workflow is as follows:
   1. Read coredump from flash. Here is an example of doing this with esptool.py, but you can also read the contents of coredump partition in your app and send it to cloud, or retrieve it from the device by REST API, etc. Make sure to change the offset and size to match your partition table.
      ```
      esptool.py --port /dev/ttyUSB0 read_flash 0x110000 0x10000 /tmp/core.bin
      ```
   2. Decode the coredump using espcoredump.py:
       ```
       espcoredump.py info_corefile --core /tmp/core.bin --core-format raw build/your_app_name.elf 
       ```
       If everything was done correctly, you should get the same output at this step as you got in option (1), where espcoredump did both steps.

tsctrl, 2021-07-19:
Hi @igrr ,

thanks for the explanation. i have run the command in 1. as below:

`%IDF_PATH%\components\espcoredump>python espcoredump.py --port COM11 --baud 921600 info_corefile build\app_2021-1-1-0.elf`

i have the same error as mention above. The command stop at:
espcoredump.py v0.4-dev
INFO: Invoke parttool to read image.

after few minutes it return the same error as above, below:
```
Traceback (most recent call last):
  File "components\partition_table\parttool.py", line 365, in <module>
    main()
  File "components\partition_table\parttool.py", line 358, in main
    op(**common_args)
  File "components\partition_table\parttool.py", line 193, in _read_partition
    target.read_partition(partition_id, output)
  File "components\partition_table\parttool.py", line 170, in read_partition
    self._call_esptool(['read_flash', str(partition.offset), str(partition.size), output] + self.esptool_read_args)
  File "components\partition_table\parttool.py", line 141, in _call_esptool
    raise e
  File "components\partition_table\parttool.py", line 138, in _call_esptool
    subprocess.check_call(esptool_args, stdout=out, stderr=subprocess.STDOUT)
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe', 'components\\esptool_py\\esptool\\esptool.py', '--port', 'COM11', 'read_flash', '5308416', '1638400', 'AppData\\Local\\Temp\\tmpphfnne66']' returned non-zero exit status 2.
ERROR: parttool script execution failed with err 1
INFO: b"esptool.py v3.1-dev\r\nSerial port COM11\r\nConnecting........_____....._____....._\r\nDetecting chip type... ESP32\r\nChip is ESP32-D0WD-V3 (revision 3)\r\nFeatures: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None\r\nCrystal is 40MHz\r\nMAC: c4:dd:57:66:9b:8c\r\nUploading stub...\r\nRunning stub...\r\nStub running...\r\n3072 (100 %)\r\n3072 (100 %)\r\nRead 3072 bytes at 0x8000 in 0.3 seconds (87.0 kbit/s)...\r\nHard resetting via RTS pin...\r\nesptool.py v3.1-dev\r\nSerial port COM11\r\nConnecting........_____....._____....._____....._____....._____....._____....._____\r\n\r\nA fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header\r\nRunning .espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe components\\esptool_py\\esptool\\esptool.py --port COM11 read_flash 32768 3072 AppData\\Local\\Temp\\tmpgomo_8pj...\r\nRunning .espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe components\\esptool_py\\esptool\\esptool.py --port COM11 read_flash 5308416 1638400 AppData\\Local\\Temp\\tmpphfnne66...\r\nAn exception: ** Command '['.espressif\\\\python_env\\\\idf4.3_py3.8_env\\\\Scripts\\\\python.exe', 'components\\\\esptool_py\\\\esptool\\\\esptool.py', '--port', 'COM11', 'read_flash', '5308416', '1638400', 'AppData\\\\Local\\\\Temp\\\\tmpphfnne66']' returned non-zero exit status 2. ** occurred in _call_esptool.\r\n"
ERROR: Error during the subprocess execution
Traceback (most recent call last):
  File "espcoredump.py", line 295, in <module>
    info_corefile()
  File "espcoredump.py", line 98, in info_corefile
    core_filename, loader = core_prepare()
  File "espcoredump.py", line 56, in core_prepare
    loader = ESPCoreDumpFlashLoader(args.off, port=args.port, baud=args.baud)
  File "components\espcoredump\corefile\loader.py", line 426, in __init__
    self._reload_coredump()
  File "components\espcoredump\corefile\loader.py", line 160, in _reload_coredump
    _header = EspCoreDumpV1Header.parse(coredump_bytes)  # first we use V1 format to get version
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 1982, in _parse
    subobj = sc._parsereport(stream, context, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 2440, in _parse
    return self.subcon._parsereport(stream, context, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 1020, in _parse
    data = stream_read(stream, self.length, path)
  File ".espressif\python_env\idf4.3_py3.8_env\lib\site-packages\construct\core.py", line 92, in stream_read
    raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> tot_len
stream read less than specified amount, expected 4, found 0
```

for 2nd approach 1st step still the same as command that i use before. this is using coredump partition with elf format dump 
i am able to generate the bin file, the file wass created in the project folder using the esptools command as below:
[... cut ...]
espcoredump.py: error: unrecognized arguments: app.elf

```

so, both approach didnt work. 
some other question is do i read the partition programmatically like reading the SPIFFS partition or OTA partition will work? i read in forum it is require some additional formatting and magic byte skipping to make it to work. could you provide some example to read the partition programmatically. i use littlefs to read the partition and try to return a stream but did not work. silly but yes is not formatted the same way, but i did that too :D

thanks

igrr, 2021-07-19:
Regarding the first error:
> after few minutes it return the same error as above, below:

Looks like the error is here:

> A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

Please check https://github.com/espressif/esptool#troubleshooting and open an issue in esptool projects if you can't find the reason.

Regarding the 2nd part:

> components\espcoredump>python espcoredump.py info_corefile core.bin --core-format raw app.elf

You are missing `--core` before `core.bin`. The command should be:
```
python espcoredump.py info_corefile --core core.bin --core-format raw app.elf
```

igrr, 2021-07-19:
> some other question is do i read the partition programmatically like reading the SPIFFS partition or OTA partition will work?

No, you need to use the [partition API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash.html#partition-table-api).

You can find examples of using this API in https://github.com/espressif/esp-idf/tree/master/examples/storage/partition_api/partition_ops.

In https://github.com/espressif/esp-idf/blob/166c30e7b2ed1dcaae56179329540a862915208a/examples/storage/partition_api/partition_ops/main/main.c#L27, instead of searching the partition table by name, you can use `ESP_PARTITION_SUBTYPE_DATA_COREDUMP`, and specify `NULL` instead of name.

tsctrl, 2021-07-19:
Hi @igrr,

thank you. the second approach works, i can see the error messages.

one more before closing this, 
if application app.elf contains the debug information and required to debug the prod devices. it means we should keep the elf release version in pair with the bin release?
so incase we deploy new ota updates we also need to keep the app.elf in sync with the ota bin version?
release 1.0: "app_v1.bin -> app_v1.elf", 
release 1.1: "app_v1.bin -> app_v1_1.elf", 
so on..

> The partition API also includes esp_partition_mmap function which can map the partition into the CPU address space. After that, you can work with the coredump as an array in memory.

thanks for the tips!

igrr, 2021-07-19:
> if application app.elf contains the debug information and required to debug the prod devices. it means we should keep the elf release version in pair with the bin release?

That's correct! If you re-build the program ELF file, the program layout will change, and core dump produced for the older ELF file will no longer work with the newer one. So you need to keep the ELF files for your releases.

If `CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF` is enabled in menuconfig (it is enabled by default), we embed the SHA256 of the program ELF file into the core dump. Then espcoredump.py script will compare SHA256 value of the program ELF file you pass to it with the SHA256 embedded into the core dump. If they don't match, [an error will be returned](https://github.com/espressif/esp-idf/blob/166c30e7b2ed1dcaae56179329540a862915208a/components/espcoredump/corefile/loader.py#L264-L274).

tsctrl, 2021-07-20:
thank you so much for the explanation igrr, closing this issue

closed 2021-07-20 by tsctrl

nicoxx-xx, 2023-01-06:
> @tsctrl you might have a confusion between ELF file, firmware binary, and core dump.
>
> 1. The application ELF file is produced by the linker during the build process. This file is located in your [...]

nicoxx-xx, 2023-01-06:
Hi @igrr,
regarding the question of read and decoding coredump partition, I can suggest to make available to Arduino ESP32 Core users an executable version of espcoredump.py, as for esptool.py. Could be an easier way to read and decode coredump, without the need for esp-idf environment.