espressif/esp-idf #7285: Com Port won't change

mparker-landt, 2021-07-19:
I was able to program a Lyrat Dev Board a few days ago. I recently tried flashing the same board and a problem occurred where it would not flash. I can see it's saying Serial port COM2 when the device is on COM3.
I have tried changing the port to COM3 which it says it does but the terminal always says COM2. I have also tried updating idf and managing the existing extension setup with the same result.

The only thing I can think of that's changed is that I updated IDF through Github Desktop before flashing.

mparker-landt, 2021-07-19:
I have managed to get around this problem by rolling back the idf extension to the last update.

Alvin1Zhang, 2021-11-23:
Thanks for reporting, sorry for slow turnaround, good to hear the issue got around, feel free to reopen.

closed 2021-11-23 by Alvin1Zhang