espressif/esp-idf #7286: WiFi intermittently disconnects due to MIC failure; takes up to a minute to reconnect

mbratch, 2021-07-19:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-PICO-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-PICO-D4
- IDF version (run ``git describe --tags`` to find it): v4.3
 - Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (Eclipse with ESP-IDF plugin)
- Power Supply:        USB, or External with DC-DC regulator to 5V

## Problem Description

I have 4 or 5 ESP32 Pico Kit devices running with WiFi and all running the same application I have written. I have a server sending each of them a REST API request every 2 seconds.

This test runs consistently for several hours. However, approximately once a day (it varies somewhat), the WiFi will disconnect and my event handler will require 40 to 60 seconds in order to reconnect again. The reason for the original disconnect appears to be MIC Failure. Subsequent attempts to reconnect result in authentication failure until finally connection is successful. 

### Expected Behavior

The tests should be able to run continuously with no interruption due to intermittent WiFi disconnections.

### Actual Behavior

WiFi connections are interrupted intermittently on a daily basis.

### Steps to reproduce

1. Start application
2. Start test which sends REST API command (w/response) to device every 2 seconds
3. After several hours or up to about 1 day, the wifi will disconnect and have to reconnect itself

## Debug Logs

```
I (8317882) wifi:state: run -> init (ec0)
I (8317882) wifi:pm stop, total sleep time: 7601484796 us / 8316383274 us

I (8317882) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (8317892) BD-PTB: wifiEvent: wifi disconnect occurred (14); attempting reconnect
I (8317932) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (8317932) wifi:state: init -> auth (b0)
I (8317932) wifi:state: auth -> init (600)
[... 158 lines cut ...]
W (8374072) BD-PTB: wifiEvent: wifi disconnect occurred (205); attempting reconnect
I (8374142) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (8374152) wifi:state: init -> auth (b0)
I (8374152) wifi:state: auth -> init (600)
I (8374152) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (8374152) BD-PTB: wifiEvent: wifi disconnect occurred (202); attempting reconnect
W (8376212) BD-PTB: wifiEvent: wifi disconnect occurred (205); attempting reconnect
I (8376222) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (8376222) wifi:state: init -> auth (b0)
I (8376222) wifi:state: auth -> init (600)
I (8376222) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
W (8376232) BD-PTB: wifiEvent: wifi disconnect occurred (202); attempting reconnect
W (8378292) BD-PTB: wifiEvent: wifi disconnect occurred (205); attempting reconnect
I (8378302) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (8378302) wifi:state: init -> auth (b0)
I (8378302) wifi:state: auth -> assoc (0)
I (8378312) wifi:state: assoc -> run (10)
I (8378322) wifi:connected with Antioch, aid = 5, channel 11, BW20, bssid = 20:c9:d0:20:04:e7
I (8378322) wifi:security: WPA2-PSK, phy: bgn, rssi: -41
I (8378332) wifi:pm start, type: 1

I (8378342) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (8379662) esp_netif_handlers: sta ip: 192.168.1.95, mask: 255.255.255.0, gw: 192.168.1.1
I (8379662) BD-PTB: wifiEvent: wifi obtained IP
```

negativekelvin, 2021-07-19:
https://github.com/espressif/esp-idf/issues/6920

mbratch, 2021-07-26:
This error continues to happen on a daily basis. Last night it happened twice in a row for the first time, with a failure to authorize in between. So the wifi connection was down for two minutes.

```
I (18557072) wifi:state: run -> init (ec0)
I (18557072) wifi:pm stop, total sleep time: 15769641637 us / 16493873934 us

I (18557072) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18557082) BD-PTB: wifiEvent: wifi disconnect occurred (reason 14); attempting reconnect
I (18557122) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18557122) wifi:state: init -> auth (b0)
I (18557132) wifi:state: auth -> init (600)
I (18557132) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18557132) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18559192) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18559272) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18559272) wifi:state: init -> auth (b0)
I (18559272) wifi:state: auth -> init (600)
I (18559272) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18559282) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18561342) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18561352) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18561352) wifi:state: init -> auth (b0)
I (18561352) wifi:state: auth -> init (600)
I (18561352) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18561362) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18563412) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18563422) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18563422) wifi:state: init -> auth (b0)
I (18563432) wifi:state: auth -> init (600)
I (18563432) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18563432) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18565492) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18565502) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18565502) wifi:state: init -> auth (b0)
I (18565502) wifi:state: auth -> init (600)
I (18565502) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18565512) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18567572) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18567582) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18567582) wifi:state: init -> auth (b0)
I (18567582) wifi:state: auth -> init (600)
I (18567582) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18567592) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18569652) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18569712) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18569712) wifi:state: init -> auth (b0)
I (18569722) wifi:state: auth -> init (600)
I (18569722) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18569722) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18571782) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18571802) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18571802) wifi:state: init -> auth (b0)
I (18571812) wifi:state: auth -> init (600)
I (18571812) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18571812) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18573872) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18573882) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18573882) wifi:state: init -> auth (b0)
I (18573882) wifi:state: auth -> init (600)
I (18573882) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18573892) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18575952) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18575962) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18575962) wifi:state: init -> auth (b0)
I (18575962) wifi:state: auth -> init (600)
I (18575962) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18575972) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18578032) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18578042) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18578052) wifi:state: init -> auth (b0)
[... 289 lines cut ...]
I (18674852) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18674852) wifi:state: init -> auth (b0)
I (18674862) wifi:state: auth -> init (600)
I (18674862) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
W (18674862) BD-PTB: wifiEvent: wifi disconnect occurred (reason 202); attempting reconnect
W (18676922) BD-PTB: wifiEvent: wifi disconnect occurred (reason 205); attempting reconnect
I (18676932) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
I (18676932) wifi:state: init -> auth (b0)
I (18676942) wifi:state: auth -> init (600)
I (18676942) wifi:new:<7,0>, old:<7,0>, ap:<255,255>, sta:<7,0>, prof:1
[... cut ...]
I (18679052) wifi:state: auth -> assoc (0)
I (18679052) wifi:state: assoc -> run (10)
I (18679062) wifi:connected with Antioch, aid = 10, channel 7, BW20, bssid = 20:c9:d0:20:04:e7
I (18679062) wifi:security: WPA2-PSK, phy: bgn, rssi: -57
I (18679072) wifi:pm start, type: 1

I (18679072) BD-PTB: wifiEvent: wifi connected
I (18679072) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (18681152) esp_netif_handlers: sta ip: 192.168.1.32, mask: 255.255.255.0, gw: 192.168.1.1
I (18681152) BD-PTB: wifiEvent: wifi obtained IP
```

AxelLin, 2021-07-30:
@mbratch 
Can you enable below settings to get more detail log:
https://github.com/espressif/esp-idf/issues/6920#issuecomment-824599813

mbratch, 2021-07-30:
> @mbratch
> Can you enable below settings to get more detail log:
> [#6920 (comment)](https://github.com/espressif/esp-idf/issues/6920#issuecomment-824599813)

Roger. I'm setting those options now.

mbratch, 2021-07-31:
@AxelLin oddly, since I set logging level to Debug a couple of days ago, this error has not occurred. Earlier today I added the extra options you suggested. I decided, momentarily, to set logging level to Info and within a few minutes I saw this same MIC error again, but there were no extra wifi logging outputs per the settings you suggested. Does log level need to be Debug for those settings to work? I've since set log level back to Debug and still keeping the other wifi options you mentioned set to continue observing.

mbratch, 2021-08-01:
> @mbratch
> Can you enable below settings to get more detail log:
> [#6920 (comment)](https://github.com/espressif/esp-idf/issues/6920#issuecomment-824599813)

I discovered that if I try to enable "Print debug messages from WPA Supplicant" I get a compilation error from ESP-IDF file that says MSG_EXCESSIVE is undefined.

AxelLin, 2021-08-01:
> I discovered that if I try to enable "Print debug messages from WPA Supplicant" I get a compilation error from ESP-IDF file that says MSG_EXCESSIVE is undefined.

Hi @mbratch 

I think you are using v4.3 tag.
There are several fixes **after** v4.3 tag.
Can you checkout release/v4.3 branch instead?

mbratch, 2021-08-01:
> 
> I think you are using v4.3 tag.
> There are several fixes **after** v4.3 tag.
> Can you checkout release/v4.3 branch instead?

@AxelLin I had done a `git pull` and `git status` showed I was on master. I've set this esp-idf folder aside and am re-retrieving the master branch again from scratch. I'll let the code run with the debug statements for a few days to see if I can trap the error.

mbratch, 2021-08-01:
Hello @AxelLin. I captured the error last night and have attached the log. I did not delete anything out of the log starting at boot-up so it would be complete. Early on, I sent a few HTTP requests to the device to check functions, then left it run overnight. There's one MIC error (reason 14) followed by about 60 seconds of attempts to reconnect, then it finally reconnects. Let me know if you need anything else.

[puttyB64-20210801a.zip](https://github.com/espressif/esp-idf/files/6912592/puttyB64-20210801a.zip)

AxelLin, 2021-08-01:
> Hello @AxelLin. I captured the error last night and have attached the log. I did not delete anything out of the log starting with boot-up so it would be complete. Early on, I sent a few HTTP requests to the device to check functions, then left it run overnight. There's one MIC error (reason 14) followed by about 60 seconds of attempts to reconnect, then it finally reconnects. Let me know if you need anything else.
> 
> [puttyB64-20210801a.zip](https://github.com/espressif/esp-idf/files/6912592/puttyB64-20210801a.zip)

Add @espxiehang 

Can you check above log?
thanks.

AxelLin, 2021-08-02:
Hi @espxiehang 
Is above log helpful? Or do you need additional information?

espxiehang, 2021-08-03:
During the multicast secret key exchange, the ESP32 received the EAPol key packet with an MIC error, causing the wifi connection to be disconnected.

AxelLin, 2021-08-03:
> During the multicast secret key exchange, the ESP32 received the EAPol key packet with an MIC error, causing the wifi connection to be disconnected.

In the original report, the behavior is consistent for esp32 devices because multiple esp32 device re-connect at the same time.
The question is: Such MIC failure is AP's problem or ESP32 device's problem?

Hi  @mbratch 
Do you observe other devices (non-esp32 devices) also have the reconnect issue?
(Just want to clarify if the MIC failure reconnect issue only happen on esp32 devices or not).

espxiehang, 2021-08-03:
The AP is suspected to be abnormal during this period.
It's best to capture the packet for analysis.
I will also try to reproduce this problem and capture packets.

AxelLin, 2021-08-03:
> The AP is suspected to be abnormal during this period.

At least 2 APs have the same issue:
In this report: Wifi Router: Apple Airport model A1408 (~10 years old)
In https://github.com/espressif/esp-idf/issues/6920: TPLink AC1750

mbratch, 2021-08-03:
> In the original report, the behavior is consistent for esp32 devices because multiple esp32 device re-connect at the same time.
> The question is: Such MIC failure is AP's problem or ESP32 device's [...]

Hi @AxelLin the problem only happens with my ESP32 devices. All four of them disconnect simultaneously.

AxelLin, 2021-08-03:
> Hi @AxelLin @espxiehang , the problem only happens with my four ESP32 devices. All of them disconnect simultaneously. My hunch is it has something to do with the combination of specific AP and the ESP32.

It's also mentioned in https://github.com/espressif/esp-idf/issues/5023
Sometimes, after a lot of retries the module manage to re-connect to the wifi network.
**Other peripherals (laptop, smartphone,...) are connected to the same WiFi network and works normally.**

mbratch, 2021-08-03:
> It's also mentioned in #5023
> Sometimes, after a lot of retries the module manage to re-connect to the wifi network.
> **Other peripherals (laptop, smartphone,...) are connected to the same WiFi [...]

My event handler will attempt reconnect every time a disconnect event occurs. For my case, re-connection is always successful within about 1 minute of this continual attempt to re-connect. I have seen a few times when reconnect occurs in about 30 seconds, but most of the time it takes about 1 minute.

negativekelvin, 2021-08-03:
It is standard for AP to enforce 60 second blackout after multiple MIC failures.  This could be poorly implemented on AP, counting multiple reports of a single multicast MIC failure as multiple failures.

mbratch, 2021-08-03:
> It is standard for AP to enforce 60 second blackout after multiple MIC failures. This could be poorly implemented on AP, counting multiple reports of a single multicast MIC failure as multiple failures.

Even when I have one ESP32 device, a single MIC failure occurs and then no reconnect for 60 seconds. However, blackout may still be the logical reason for the delay in reconnect. The big question is: why did the MIC failure occur with just the ESP32s?

negativekelvin, 2021-08-03:
You know only the esp32 clients disconnected based on AP logs?  It is a dual band AP so are the other clients on 5ghz?

mbratch, 2021-08-03:
> You know only the esp32 clients disconnected based on AP logs? It is a dual band AP so are the other clients on 5ghz?

I know the ESP32 clients disconnected because of the console log on one of them, and because the others immediately stop responding to HTTP requests (they all run on a 2 per second HTTP request test). It is a dual band AP, but most of the other clients are also on 2.4ghz. I have not checked the AP logs after a disconnect. I can certainly do that on the next disconnect that occurs.

xueyunfei998, 2021-08-06:
idf.py menuconfig
Component config  ---> Wi-Fi  ---> [*] Enable WiFi debug log
                                   [*]     WiFi debug log submodule
                                   [*]         WiFi Debug Log Submodule Conn
Component config  ---> Log output  ---> Default log verbosity (Debug) 
Component config  ---> Supplicant  ---> [*] Print debug messages from WPA Supplicant

hi @mbratch 
Please follow the steps above to provide a detailed log. 

Guess there's something wrong with the four handshakes

AxelLin, 2021-08-06:
> idf.py menuconfig
> Component config ---> Wi-Fi ---> [_] Enable WiFi debug log [_] WiFi debug log submodule
> [_] WiFi Debug Log Submodule Conn Component config ---> Log output ---> Default log [...]

@xueyunfei998 
I believe the log in https://github.com/espressif/esp-idf/issues/7286#issuecomment-890502760
already enable above settings. Could you take a look?

mbratch, 2021-08-06:
@xueyunfei998  the log I provided already had those settings enabled..

AxelLin, 2021-08-06:
> I noticed that using master the wifi debug settings no longer appear in menuconfig under Components > Wifi. Do you know why that would be?

It's because the change by https://github.com/espressif/esp-idf/commit/4702feeeebcd4ccb3090f4747f96aea6d719f432#diff-9ca68292e9dc2f35cf728bd200097744cd63d6647f9f2df2dbe27c65c504039a
Now the wifi_log_level is set by CONFIG_LOG_MAXIMUM_LEVEL setting.

mbratch, 2021-08-06:
@AxelLin great, thank you!

mbratch, 2021-08-07:
@xueyunfei998 here is another log I captured with this error.
[puttyB64-20210730a.zip](https://github.com/espressif/esp-idf/files/6950032/puttyB64-20210730a.zip)

AxelLin, 2021-08-08:
> @xueyunfei998 here is another log I captured with this error.
> [puttyB64-20210730a.zip](https://github.com/espressif/esp-idf/files/6950032/puttyB64-20210730a.zip)

@mbratch 
This is httpd log, seems not related to this issue?

mbratch, 2021-08-08:
> > @xueyunfei998 here is another log I captured with this error.
> > [puttyB64-20210730a.zip](https://github.com/espressif/esp-idf/files/6950032/puttyB64-20210730a.zip)
> 
> @mbratch
> This is httpd log, seems not related to this issue?

I apologize, it was the wrong log file. I edited my comment and uploaded the correct log.

mbratch, 2021-08-12:
@AxelLin @xueyunfei998 I captured another log (this error occurs daily), this time with Verbose log setting turned on.

[puttyB64-MIC-20210812a.zip](https://github.com/espressif/esp-idf/files/6978804/puttyB64-MIC-20210812a.zip)

negativekelvin, 2021-08-12:
There is a discontinuity in the sequence number before the deauth and there are no other discontinuities in the log

> V (20592017) wpa: WPA: RX EAPOL-Key - hexdump(len=147):[0m
> V (20592017) wpa: 01 03 00 8f 02 13 82 00 20 00 00 00 00 00 00 00 [0m
> V (20592027) wpa: **45** f8 f0 81 5e 3c ce 37 39 23 71 62 f6 d3 bd 49 [0m
> ...
> V (23199837) wpa: WPA: RX EAPOL-Key - hexdump(len=147):[0m
> V (23199847) wpa: 01 03 00 8f 02 13 82 00 20 00 00 00 00 00 00 00 [0m
> V (23199847) wpa: **49** b9 37 68 1c f4 2f ba 23 b7 83 91 ef 3c db da [0m
>

mbratch, 2021-08-13:
@negativekelvin  you're looking at an unedited and complete log. I don't know why there is such a large gap in time.

mbratch, 2021-08-17:
Ok this is not the same issue, and is less problematic since the wifi only goes down for a second or so then comes back. But in the last 5 days, instead of seeing any further MIC errors, I've been getting daily reason 15 disconnects (4WAY_HANDSHAKE_TIMEOUT). Log attached, in case it offers any further clues to the MIC issue. It's the same unit. No changes to anything.

[puttyB64-015-20210817a.zip](https://github.com/espressif/esp-idf/files/7003536/puttyB64-015-20210817a.zip)

mbratch, 2021-08-18:
One more MIC failure log for the road. Should be plenty of data. Let me know if you need any other information on this defect.

[puttyB64-015-20210817a.zip](https://github.com/espressif/esp-idf/files/7010355/puttyB64-015-20210817a.zip)

negativekelvin, 2021-08-18:
Last log file does not have any reason 14

mbratch, 2021-08-18:
@negativekelvin  crap I posted the wrong file and I may have deleted the right one. I'll let it run further. I'll get another one eventually. How many do you need?

mbratch, 2021-08-18:
@negativekelvin  sorry about that. Posted the wrong log file (I've got a few of them collected up). Try this one:

[puttyB64-MIC-20210818a.zip](https://github.com/espressif/esp-idf/files/7010798/puttyB64-MIC-20210818a.zip)

negativekelvin, 2021-08-19:
Looks different than the other log, no missed eapol packets and mic failure detected on esp32 side

```
D (20789017) wpa: TKIP MIC failure occur
D (20789027) wpa: WPA: Sending EAPOL-Key Request (error=1 pairwise=0 ptk_set=1 len=99)
```

mbratch, 2021-08-19:
@negativekelvin I have no explanation for that. The environment and configuration is the same as the prior log.

negativekelvin, 2021-08-19:
Probably not. If we assume mic failure can't be prevented due to rf environment then the question is can esp32 do anything to avoid the 1 minute blacklist or is your AP overzealous in applying it.

mbratch, 2021-08-19:
@negativekelvin what is it about the "RF environment" that might cause this? It's a fairly benign environment and I'm 15 feet away from the AP without any walls in between. Are there other possible causes? I'll do some research on how my particular AP handles blacklisting and see if I can find anything interesting.

negativekelvin, 2021-08-19:
I have read that it is possible to get mic failure from interference/corruption but not sure how it would pass FCS unless it was an active MIM attack.  Otherwise I guess it would have to be computational error in calculating mic.  I have not checked the calculations based on the binary data in the verbose logs to attempt to verify the calculations on esp32 side.

mbratch, 2021-08-21:
I have several devices on my network (as might be expected). The ESP32's are the only devices experiencing this issue.

negativekelvin, 2021-08-21:
You have logs from those non-esp32 devices that prove they do not disconnect?

mbratch, 2021-08-21:
@negativekelvin  one of those devices is the laptop I'm using to capture the logs and it is not disconnecting. It is connected to the network using the 2.4GHz band as well. I don't have logs from another unit pinging it, but if my laptop disconnected at the same time, it would have failed to capture the information.

negativekelvin, 2021-08-21:
I'm not totally following you, aren't you capturing the esp32 logs over usb?  But if you are sure the laptop has no disconnects because it has clean dmesg or equivalent then ok.

mbratch, 2021-08-21:
@negativekelvin  my bad, I've been running some other tests, some over network as well. But my point still stands: I've sat here with my laptop and observed the problem occurring over USB on the connected ESP32, and my laptop is still connected properly to the network at that time.

negativekelvin, 2021-08-22:
If you have looked at the logs from Windows event viewer or dmesg on the laptop and there are no disconnects then it is time to look more closely at esp32 so hopefully @xueyunfei998 can review the verbose logs

mbratch, 2021-08-22:
Ok the next time the issue occurs, I'll check the Windows events.

mbratch, 2021-08-24:
The MIC error occurred again late last night (around midnight? Not sure as my ESP32 logging is ms after boot and not sure of the exact time I booted). I looked at Windows Event Logs, `dmesg` on my server, and error counts in `ifconfig` on my network interface on my server` and saw no errors.

mbratch, 2021-09-01:
Yesterday, when the MIC event occurred on two of the ESP devices, it also disconnected my work laptop for 60 seconds, dropping me from a Teams meeting briefly.

negativekelvin, 2021-09-01:
Have you seen any logs of a non-ESP device reporting a mic failure?

mbratch, 2021-09-01:
Nope. Only ESP devices reported a MIC.

Scott--R, 2021-09-03:
I think I am also experiencing similar behaviour. I have Google WiFi in my home. I use the BluFi process to configure the WiFi credentials. I sometimes will configure as many as 20 ESP32 devices in rapid succession. I have noticed that they will not go online until about 30 - 60 seconds after I finish configuring the last one. Then suddenly ALL of them go online within 1-2 seconds of each other.

I find this to be true pretty consistently. It is extremely repeatable. I have done it hundreds of times.

Please note: I am NOT a software developer but if I can offer some assistance in debugging this I am happy to do so.

mbratch, 2021-09-03:
@Scott--R  it may be a different problem. In my case, it's not repeatable at will. It just happens approximately once per day while connected without doing any explicit network interaction with the devices, and via console log, it shows it is due to an asynchronous disconnection of the wifi link to the ESP32. Check the console log on one of your ESP32's to see what the errors are. You may need to add a message or two in the wifi event handler for clarity (which is what I did). From there you can display the precise error, if there is one.

mbratch, 2021-11-16:
A couple of weeks ago I replaced my Apple Airport Extreme model A1408 with a TP-Link AC-4000 and this problem no longer occurs. So it appears to be dependent upon wifi access point manufacturer & model.

jack0c, 2022-05-20:
@mbratch TKIP MIC error is used to detect attack: If a station get 2 TKIP MIC errors in a minute, you need to disconnect from AP, and wait a minute until it can connect to the AP again. This is part of IEEE80211 standard.

mbratch, 2022-05-20:
@jack0c  yes, I understand that. However, if it were working properly in this implementation, it should not continually disrupt normal operation of the device on the wifi network.

AxelLin, 2022-05-24:
@jack0c 
Any comments about disabling WiFi Multimedia Management (WMM) can avoid the issue?
This was reported quite long time ago, but seems no feedback: https://github.com/espressif/esp-idf/issues/6920#issuecomment-860688696

[11 more replies]

closed 2025-02-26 by Alvin1Zhang