espressif/esp-idf #7291: MQTT HEAP IN EVENT CONNECTION ERROR

romocitto88, 2021-07-20:
## Environment

- Development Kit: 
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: ESP32-Wroom-32E
- IDF version latest stable 4.3
- Operating System:     Windows
- Using an IDE?: PlatformIO
- Power Supply: external 5V

Problem Description:

I'm using the MQTT library, when the mqtt event error comes i have two situation: 

Whit this type of error the free heap is about 45k bytes

> [0;31mE (378561) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x10[0m
> [0;31mE (378561) esp-tls: Failed to open new connection[0m
> [0;31mE (378562) TRANS_SSL: Failed to open a new connection[0m
> [0;31mE (378569) MQTT_CLIENT: Error transport connect[0m
> MQTT_EVENT_ERROR
> Last error code reported from esp-tls: 0x8010
> Last tls stack error number: 0x10
> Free heap: 45400
> MQTT_EVENT_DISCONNECTED

Whit this other type of error the free heap is about 17k

> [0;33mW (49032) MQTT_CLIENT: Connection refused, bad username or password[0m
> MQTT_EVENT_ERROR
> Connection refused error: 0x4
> Free heap: 17928
> [0;31mE (49035) MQTT_CLIENT: MQTT connect failed[0m
> MQTT_EVENT_DISCONNECTED

And after some error of refused connection:

> assertion "xResult" failed: file "C:\.platformio\packages\framework-espidf\components\freertos\timers.c", line 787, function: > > > prvProcessReceivedCommands
> 
> abort() was called at PC 0x401a4a9f on core 0
> 
> Backtrace:0x40091e76:0x3ffd8030 0x400925bd:0x3ffd8050 0x40098c3e:0x3ffd8070 0x401a4a9f:0x3ffd80e0 0x400954ba:0x3ffd8110
> 
> ELF file SHA256: 6e62d34834c8aefe

There's a way to free the memory space?

david-cermak, 2021-07-21:
Hi @romocitto88 

One ssl client session requires approximately 40k of RAM. This would probably explain the first error:
> Whit this type of error the free heap is about 45k bytes
Since `MBEDTLS_ERR_MPI_ALLOC_FAILED=-0x0010` and the heap might be fragmented.

As for the other two issues:
> Whit this other type of error the free heap is about 17k
> �[0;33mW (49032) MQTT_CLIENT: Connection refused, bad username or password�[0m

and 
> And after some error of refused connection:
> assertion "xResult" failed: file "C:.platformio\packages\framework-espidf\components\freertos\timers.c", line 787, function: > >

I would think a memory corruption of some kind (assuming you're connecting to the same server with the same credentials), probably related to the shortage of heap and incorrect handling of a failure to allocate.

> There's a way to free the memory space?

Please check the [Performance guidelines](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/index.html) in the docs. There's a very useful chapter on how to minimize RAM usage: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html

romocitto88, 2021-07-22:
> I would think a memory corruption of some kind (assuming you're connecting to the same server with the same credentials), probably related to the shortage of heap and incorrect handling of a failure to allocate.

In my case it's the same server, with a password that changes every time. Anyway, I'm testing the case where my module fails to connect to the broker for wrong credentials, but I don't understand why after some refused connection I get this error.

> Please check the Performance guidelines in the docs. There's a very useful chapter on how to minimize RAM usage: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html

I enabled the "NANO FORMATTING" flag to try if something changes, and the free heap rised to 24k but the 
> assertion "xResult" failed: file "C:.platformio\packages\framework-espidf\components\freertos\timers.c", line 787, function: > > > prvProcessReceivedCommands
>
> abort() was called at PC 0x401a4a9f on [...]

Happened again.

david-cermak, 2021-07-23:
> I'm testing the case where my module fails to connect to the broker for wrong credentials,

Thanks for explaining, now I understand, the `Connection refused error: 0x4` is actually an expected behaviour, but you report the consequent assertion from FreeRTOS. As mentioned before, these are typically caused by a memory corruption. So I'd like to ask you for a minimal project where this issue could be reproducible.

I've played a bit with this error code, but haven't seen anything unexpected. We test the mqtt library for connection refusal, but only for the 0x5 error code (`not authorized`) in the CI:

https://github.com/espressif/esp-idf/blob/release/v4.3/tools/test_apps/protocols/mqtt/publish_connect_test/app_test.py#L313

Also, if injecting the error code `0x4=Connection refused error`, the mqtt library works as expected as it continues to the publish tests iterating all transports, qos'es, enqueue's with different payloads. That's why I believe the issue must from the user application rather then anything in IDF. Please help to share the minimal application, so we can doublecheck this was the case.

Alvin1Zhang, 2021-08-31:
@romocitto88 Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

romocitto88, 2021-09-01:
I didn't found the cause of this issue, so I decided to use fixed credentials to finish the project and at the end investigate.
At the beginnig I will try to reduce the RAM usage, in addition I will try to stop some task one by one to see the effects and finally I will share here the results.

Alvin1Zhang, 2021-11-23:
@romocitto88 Thanks for reporting and sharing all the details, feel free to reopen if any updates available. Thanks.

closed 2021-11-23 by Alvin1Zhang