espressif/esp-idf #7292: esp-idf/examples/storage/sd_card/sdspi Code giving error

rjrajbir, 2021-07-20:
I have connected all the pins accordingly but it's showing the same error

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
[0;32mI (27) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader[0m
[0;32mI (27) boot: compile time 15:05:42[0m
[0;32mI (27) boot: chip revision: 1[0m
[0;32mI (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (37) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (42) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (46) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (51) boot: Enabling RNG early entropy source...[0m
[0;32mI (56) boot: Partition Table:[0m
[0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (82) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (89) boot: End of partition table[0m
[0;32mI (94) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0a7b4 ( 42932) map[0m
[0;32mI (126) esp_image: segment 1: paddr=0x0001a7dc vaddr=0x3ffb0000 size=0x02174 (  8564) load[0m
[0;32mI (130) esp_image: segment 2: paddr=0x0001c958 vaddr=0x40080000 size=0x036c0 ( 14016) load[0m
[0;32mI (139) esp_image: segment 3: paddr=0x00020020 vaddr=0x400d0020 size=0x1e9b0 (125360) map[0m
[0;32mI (189) esp_image: segment 4: paddr=0x0003e9d8 vaddr=0x400836c0 size=0x082c0 ( 33472) load[0m
[0;32mI (211) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (211) boot: Disabling RNG early entropy source...[0m
[0;32mI (211) cpu_start: Pro cpu up.[0m
[0;32mI (215) cpu_start: Application information:[0m
[0;32mI (220) cpu_start: Project name:     template-app[0m
[0;32mI (225) cpu_start: App version:      1[0m
[0;32mI (230) cpu_start: Compile time:     Jul 20 2021 15:05:08[0m
[0;32mI (236) cpu_start: ELF file SHA256:  c7856376af568d61...[0m
[0;32mI (242) cpu_start: ESP-IDF:          v4.2.2-dirty[0m
[0;32mI (247) cpu_start: Starting app cpu, entry point is 0x40081714[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (258) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (270) heap_init: At 3FFB6A20 len 000295E0 (165 KiB): DRAM[0m
[0;32mI (277) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (283) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (289) heap_init: At 4008B980 len 00014680 (81 KiB): IRAM[0m
[0;32mI (296) cpu_start: Pro cpu start user code[0m
[0;32mI (314) spi_flash: detected chip: generic[0m
[0;32mI (314) spi_flash: flash io: dio[0m
[0;32mI (315) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (323) example: Initializing SD card[0m
[0;32mI (323) example: Using SPI peripheral[0m
[0;31mE (333) vfs_fat_sdmmc: slot init failed (0x103).[0m
[0;31mE (333) example: Failed to initialize the card (259). Make sure SD card lines have pull-up resistors in place.[0m
ets Jun  8 2016 00:22:57
```

Please help how to resolve this error.

nopnop2002, 2021-07-24:
This sample works correctly.
There are several possible causes.

1. The connection is incorrect.
2. The SD card is not formatted in FAT32.
3. The connection cable is too long.
4. There is No pull-up resistor.

rjrajbir, 2021-08-04:
No, I have changed the SPI Frequency from 20000 Hz to 5000 Hz and it worked.

closed 2021-08-04 by rjrajbir