espressif/esp-idf #7293: Modbus reporting error 0x108, but the slave received the data successfully

silardg, 2021-07-20:
## Environment

- Module or chip used:  ESP32-WROOM-32D

After sending data to the slave via the function ` mbc_master_send_request` the slave does it's thing, but the code returns ` MB_CONTROLLER_MASTER: mbc_master_send_request(131): SERIAL master send request failure error=(0x108) (ESP_ERR_INVALID_RESPONSE).`

I have tried increasing timeout, but no luck.

SDK config:
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
# CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
[... 1082 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

alisitsyn, 2021-07-22:
@silardg , 

In order to solve your issue could you let me know the version of ESP-IDF and send me the log of master? The issue is usually related to coexistence of Modbus with Wi-Fi tasks.
The proposals according to your sdkconfig:
1. Increase the `CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND` to the value > of your slave response time (currently = 150 ms, which is not enough).
2. Place the interrupt handlers into IRAM, `CONFIG_FMB_TIMER_ISR_IN_IRAM` = y;
3. Below is the patch to add affinity kconfig option for mod bus tasks:
[0001_freemodbus_add_affinity.diff.log](https://github.com/espressif/esp-idf/files/6148467/0001_freemodbus_add_affinity.diff.log)
Please try to apply this patch (it is for latest master branch) `git apply --check 0001_freemodbus_add_affinity.diff` and if there are no errors apply it without `--check` then try to set affinity for modbus tasks to other CPU in menuconfig (`CONFIG_FMB_PORT_TASK_AFFINITY`). The Modbus tasks and Wi-Fi tasks should run on different CPU.
4. Recompile and start your code to check.

Try to follow the recommendations above and let me know if you have any further issues or questions.
Thanks.

silardg, 2021-07-22:
ESP-IDF version is 4.2

I will try the  steps you have written down.

I'm using ethernet with Modbus, not sure if that is the same as WiFi. 

Will keep this issue open and get back to it as soon as I tested.

alisitsyn, 2021-07-22:
Thanks for update.

> I'm using ethernet with Modbus, not sure if that is the same as WiFi.

This can cause similar issues. Modbus and LWIP related tasks better run on different CPU.

I have also some comments, instructions and example about your previous issue which you closed:
https://github.com/espressif/esp-idf/issues/7220
We can continue with this after solving the current issue.

alisitsyn, 2021-07-27:
@silardg ,

Once you have updates related Modbus coexistence with Ethernet, just let me know and I can help you to fix this.

Alvin1Zhang, 2021-11-23:
@silardg Thanks for reporting, would you please help share if any updates for the issue? Thanks.

silardg, 2021-12-01:
No update yet, let's close it, if I have anything new, will reopen.

closed 2021-12-01 by silardg

alisitsyn, 2021-12-01:
@silardg , thank you report. Yes, feel free to reopen if you have any issues.