espressif/esp-idf #7294: How do I know the timing device is using WiFi or BLE?

Rokachy, 2021-07-20:
Our app implements a BLE to Wi-Fi gateway.
The gateway listen to BLE devices advertisements (BLE scanning) and every second reports the list of devices to system application via WiFi.
We are using "Software controls WiFi/Bluetooth coexistence" (HW control coexistence didn't works for us).
We are using 4.0.3 and v4.3 SDKs.
How do I know how much time the device is at BLE at WiFi ?  Is there is a status that can query to retrieve its state?  
How I can configure the timing to optimize WiFi/BLE for my needs?

xiongweichao, 2021-07-27:
Hi, @Rokachy 

`1. Is there is a status that can query to retrieve its state?`
 Do you want to debug or use it in your application?

`2. How I can configure the timing to optimize WiFi/BLE for my needs?`
The time used by WiFi/BLE is automatically configured.

Thanks

Rokachy, 2021-07-29:
Hi, 

1.  I want to use it in my application, this is the setup:
>I have a BLE device that sent advertisement every second
>Gateway (ESP32 based) report uart the received BLE advertisement but sometimes reports missed for 2-3 sec and then resume 
  I assume that ESP32 was on WiFi at the time reports missed but I don't know how to approve it.
2.  what do mean for automatically? is it fixed or dynamically changed? 

Yehuda

xiongweichao, 2021-07-29:
Hi, @Rokachy 

`what do you mean for automatically? is it fixed or dynamically changed?`
This time is dynamically changing.

Thanks

xiongweichao, 2021-08-02:
Hi, @Rokachy 

Sorry, ESP32 currently does not support this function, but this function may be added in the future.

Thanks

xiongweichao, 2024-04-10:
Hi @Rokachy 

Sorry for replying to you so late. You can get the wifi/ble status through the GPIO level.  Reference documentation https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/phy.html

Thanks

closed 2025-03-20 by Alvin1Zhang