espressif/esp-idf #7295: littlefs and coredump partition write erase issue

tsctrl, 2021-07-20:
ESP-IDF v4.3.0

issue:
littlefs stop working when coredump partition added in csv. littlefs not able to erase and write if coredump partition added in csv.
how to replicate:
add coredump partition details in csv.

details:
littlefs not able to do write and erase operation as below api return error if coredump partition added in csv 
example api:
```
esp_err_t err = esp_partition_erase_range(efs->partition, part_off, c->block_size);
```

sample sketch:
```
File fs = LITTLEFS.open("/data/fs.txt", "w");
if(fs){
	fs.println("write test.");
	fs.close();
}

```
error:
E (00:00:00.654) esp_littlefs_api: int_storage, 
E (00:00:00.706) esp_littlefs_api: failed to erase addr 0009b000, size 00001000, err 258
[   730][E][vfs_api.cpp:270] VFSFileImpl(): fopen(/spiffs/data/fs.txt) failed

partition:
```
nvs,              data, nvs,      0x9000,    0x6000,
phy_init,      data, phy,      0xf000,    0x1000,
factory,        app,  factory,  0x20000,   2M,
coredump,  data, coredump, 0x220000,  0x1F400,
int_storage, data, spiffs,   0x23f400,  0x1C0C00
```

thanks!

tsctrl, 2021-07-20:
if i put the coredump in the last partition i will have this issue:

```
I (2959) esp_core_dump_flash: Save core dump to flash...
I (3031) esp_core_dump_flash: Erase flash 16384 bytes 
E (3111) esp_core_dump_flash: Failed to erase flash (258)!
E (3185) esp_core_dump_elf: Failed to prepare core dump storage (258)!
E (3272) esp_core_dump_common: Core dump write binary failed with error=258
I (3363) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
```

negativekelvin, 2021-07-20:
Partitions have to be sector aligned (0x1000)

tsctrl, 2021-07-21:
i have figure out workaround for this issue:

1. coredump partition must be configured/located at the end of the flash.
2. coredump partition size must be large enough, i not sure the suitable size, documentation mention is 64K. but i change the size to 6.1mb and it write the core dump successfully. the above setup with 128K double the size from documentation did not work either.

@negativekelvin , not sure if i miss the alignment. it should show compilation error somehow if that was the issue. if i understand correctly alignment offset only applied to nvs and app partition only. 

thanks!

negativekelvin, 2021-07-21:
>if i understand correctly alignment offset only applied to nvs and app partition only.

No, anything you want to be able to erase has to be aligned to sector erase boundaries

After you get the coredump you should be able to check it's actual binary size

tsctrl, 2021-07-22:
sorry, I couldn't get what you mean. the fs get the erase address automatically and try to perform erase to that address. not sure if sector boundaries alignment needed for that.

the issue is:
1. any spiffs write or erase operation fails if the coredump partition is added before the last partition. is not related to erasing coredump.
2. the second issue is coredump write failed when the coredump partition is smaller than what size it is required.

negativekelvin, 2021-07-22:
```
coredump,  data, coredump, 0x220000,  0x1F400,
int_storage, data, spiffs,   0x23f400,  0x1C0C00
```

Think about it.  If you tried to erase the coredump sector at 0x23f000-0x23ffff you would also erase the first part of the spiffs partition.  So it will fail.

closed 2021-08-24 by espressif-bot