espressif/esp-idf #7297: hd_eeprom example don't work.

nopnop2002, 2021-07-21:
## Environment

```
Core is ESP32@160Mhz
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
portTICK_RATE_MS:10 msec/tick
lwIP version:2-1-3-0
ESP-IDF version:v4.4-dev-1183-g9d34a1cd4-dirty
chip model is 1, chip with 2 CPU cores, WiFi/BT/BLE
silicon revision 1
2MB external flash
package is ESP32D0WDQ6
```

## Problem Description
read data not correct.

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (334) main: PIN_NUM_MISO=18 PIN_NUM_MOSI=23 PIN_NUM_CLK=19 PIN_NUM_CS=13
I (344) main: Initializing bus SPI2...
I (344) main: Initializing device...
I (354) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (364) main: Write: Hello World!
I (364) main: Read: 
I (364) main: 0x3ffb62d1   ff ff ff ff ff ff ff ff  ff ff ff ff ff 00 00 00  |................|
I (374) main: 0x3ffb62e1   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (384) main: Example finished.
```

### the connections

|      | ESP32 | |
| ---- | ----- | 
| Host | HSPI  |
| VCC  | 3.3V  |
| GND  | GND   |
| DO   |  18    |
| DI   |  23    | 
| SK   | 19    | 
| CS   | 13    |
| ORG  | GND   |

nopnop2002, 2021-07-21:
I changed the development board.
solved.
It's a hardware problem.

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (334) main: PIN_NUM_MISO=18 PIN_NUM_MOSI=23 PIN_NUM_CLK=19 PIN_NUM_CS=13
I (344) main: Initializing bus SPI2...
I (344) main: Initializing device...
I (354) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (364) main: Write: Hello World!
I (384) main: Read: Hello World!
I (384) main: 0x3ffb6331   48 65 6c 6c 6f 20 57 6f  72 6c 64 21 00 00 00 00  |Hello World!....|
I (394) main: 0x3ffb6341   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (404) main: Example finished.
```

closed 2021-07-21 by nopnop2002

charmi4857, 2022-12-09:
> ## Environment
> ```
> Core is ESP32@160Mhz
> freeRTOS version:V10.2.1
> NEWLIB version:3.0.0
> portTICK_RATE_MS:10 msec/tick
> lwIP version:2-1-3-0
> ESP-IDF version:v4.4-dev-1183-g9d34a1cd4-dirty
> chip model is 1, chip with 2 CPU cores, WiFi/BT/BLE
> silicon revision 1
> 2MB external flash
> package is ESP32D0WDQ6
> ```
> 
> ## Problem Description
> read data not correct.
> 
> ```
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (334) main: PIN_NUM_MISO=18 PIN_NUM_MOSI=23 PIN_NUM_CLK=19 PIN_NUM_CS=13
> I (344) main: Initializing bus SPI2...
> I (344) main: Initializing device...
> I (354) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
> I (364) main: Write: Hello World!
> I (364) main: Read: 
> I (364) main: 0x3ffb62d1   ff ff ff ff ff ff ff ff  ff ff ff ff ff 00 00 00  |................|
> I (374) main: 0x3ffb62e1   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
> I (384) main: Example finished.
> ```
> 
> ### menuconfig
> [image]
> 
> ### the connections
> 93C46	ESP32
> Host	HSPI
> VCC	3.3V
> GND	GND
> DO	18
> DI	23
> SK	19
> CS	13
> ORG	GND

I got the same problem. Can you tell me, how to solve that.

nopnop2002, 2022-12-09:
@charmi4857 

I changed the development board.

charmi4857, 2022-12-10:
But i think, there is no problem in my development board. Did you change
anything in the hd_eeprom code.

charmi4857, 2022-12-10:
Okk. I understand now.