espressif/esp-idf #7298: With ESP32-C3 hd_eeprom example does not work

nopnop2002, 2021-07-21:
# Environment

```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.4-dev-1183-g9d34a1cd4-dirty
```

# Problem Description
With ESP32-C3 hd_eeprom example does not work

# Using ESP32
```
I (0) cpu_start: Starting scheduler on APP CPU.
I (334) main: PIN_NUM_MISO=18 PIN_NUM_MOSI=23 PIN_NUM_CLK=19 PIN_NUM_CS=13
I (344) main: Initializing bus SPI2...
I (344) main: Initializing device...
I (354) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (364) main: Write: Hello World!
I (384) main: Read: Hello World!
I (384) main: 0x3ffb6331   48 65 6c 6c 6f 20 57 6f  72 6c 64 21 00 00 00 00  |Hello World!....|
I (394) main: 0x3ffb6341   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (404) main: Example finished.
```

# Using ESP32S2
```
I (261) cpu_start: Starting scheduler on PRO CPU.
I (266) main: PIN_NUM_MISO=37 PIN_NUM_MOSI=35 PIN_NUM_CLK=36 PIN_NUM_CS=34
I (266) main: Initializing bus SPI2...
I (276) main: Initializing device...
I (276) gpio: GPIO[34]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (286) main: Write: Hello World!
I (316) main: Read: Hello World!
I (316) main: 0x3ffc2c51   48 65 6c 6c 6f 20 57 6f  72 6c 64 21 00 00 00 00  |Hello World!....|
I (316) main: 0x3ffc2c61   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (326) main: Example finished.
```

# Using ESP32C3
I (268) cpu_start: Starting scheduler.
I (273) main: PIN_NUM_MISO=2 PIN_NUM_MOSI=7 PIN_NUM_CLK=6 PIN_NUM_CS=10
I (283) main: Initializing bus SPI2...
E (293) spi: spi_bus_initialize(634): invalid dma channel, chip only support spi dma channel auto-alloc
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x403857ff
0x403857ff: _esp_error_check_failed at /home/nop/esp-idf/components/esp_system/esp_err.c:42

file: "IDF/examples/peripherals/spi_master/hd_eeprom/main/spi_eeprom_main.c" line 80
func: app_main
expression: ret

abort() was called at PC 0x40385803 on core 0
0x40385803: _esp_error_check_failed at /home/nop/esp-idf/components/esp_system/esp_err.c:43

Stack dump detected
Core  0 register dump:
MEPC    : 0x420009f0  RA      : 0x40385812  SP      : 0x3fc91c00  GP      : 0x3fc8d000
0x420009f0: panic_abort at /home/nop/esp-idf/components/esp_system/panic.c:365

0x40385812: __ubsan_include at /home/nop/esp-idf/components/esp_system/ubsan.c:294

TP      : 0x3fc8a448  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fc91c64  A0      : 0x3fc91c2c  A1      : 0x3fc91c62
A2      : 0x00000000  A3      : 0x3fc91c59  A4      : 0x00000001  A5      : 0x3fc8f000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000
```

ESP-Marius, 2021-07-21:
Hi @nopnop2002 

This was already fixed with 4de9cbcf0ff6fec37993b7edcc9f1c84d75b4a62 

On the latest master it's already using `SPI_DMA_CH_AUTO`, so it should be OK.

Seeing as this is already fixed i'll close the issue. But thanks for the report!

closed 2021-07-21 by ESP-Marius

kaibin1111, 2021-07-30:
> 
> 
> Hi @nopnop2002
> 
> This was already fixed with [4de9cbc](https://github.com/espressif/esp-idf/commit/4de9cbcf0ff6fec37993b7edcc9f1c84d75b4a62)
> 
> On the latest master it's already using `SPI_DMA_CH_AUTO`, so it should be OK.
> 
> Seeing as this is already fixed i'll close the issue. But thanks for the report!

I also encountered this problem, has it not been fixed?

ESP-Marius, 2021-07-30:
@kaibin1111 This specific problem should be fixed on master, which version/commit ID are you using?

nopnop2002, 2021-07-30:
@kaibin1111 
It's fixed with latest.
[image]

kaibin1111, 2021-07-30:
> 
> 
> @kaibin1111
> 它是最新的。
> [image]

Thank you. It seems my version is not up to date.

kaibin1111, 2021-07-30:
> 

> 
> 
> @kaibin1111
> It's fixed with latest.
> [image]

Thank you. It seems my version is not up to date.

kaibin1111, 2021-07-30:
> 
> 
> @kaibin1111 This specific problem should be fixed on master, which version/commit ID are you using?

Thank you. It seems my version is not up to date.