espressif/esp-idf #7300: Ethernet crashing randomly

silardg, 2021-07-21:
I'm getting the following log from a backtrace:
```
0x400eafb5: emac_esp32_init at /Users/silardg/.platformio/packages/framework-espidf@3.40200.210118/components/esp_eth/src/esp_eth_mac_esp32.c:324 (discriminator 2)
0x400898cd: _frxt_timer_int at /Users/silardg/.platformio/packages/framework-espidf@3.40200.210118/components/freertos/xtensa/portasm.S:314
```
I'm using `lan8720a` with `HanRun HR911105A`.

david-cermak, 2021-07-23:
Hi @silardg 

Could you please provide more details, following the issue template?
Specifically interested in
* IDF version (platformIO with *some*/ forked IDF?)
* which chip (ESP32?)
* your ethernet connection (lan8720a is an RMII device, but HR911105A seems to use the old good ENC28J60?)

Alvin1Zhang, 2021-08-18:
@silardg Thanks for reporting, would you please help provide more details per the issue template? Thanks.

Alvin1Zhang, 2021-08-31:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates.

closed 2021-08-31 by Alvin1Zhang