espressif/esp-idf #7301: Duplication of events in mode gattc_gatts_coex

hard-cat, 2021-07-21:
I'm trying to run BLE in coex mode. Individually, the client or server works fine.
In the coex mode, when connecting to the server, the callback is processed by both the server and the client.
uuid char server and client do not match.
I wrote based on the example gattc_gatts_coex
ESP-IDF v4.2.2
chip esp32-pico-v3-02, user board

Initialization:

    esp_err_t ret;
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }
    ret = esp_ble_gap_register_callback(esp_gap_cb);
    if (ret){
         ESP_LOGE(TAG, "%s gap register failed, error code = %x\n", __func__, ret);
        return;
    }
    ret = esp_ble_gatts_register_callback(gatts_event_handler);
    if (ret){
        ESP_LOGE(TAG, "gatts register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    ret = esp_ble_gatts_app_register(PROFILE_B_APP_ID);
    if (ret){
        ESP_LOGE(TAG, "gatts app register error, error code = %x", ret);
        return;
    }
    const uint8_t * addr = esp_bt_dev_get_address();
    if (addr) {
    } 
    set_ble_tx_power(ble_pwr);
    ret = esp_ble_gattc_register_callback(esp_gattc_cb);
    if(ret){
        ESP_LOGE(TAG, "%s gattc register failed, error code = %x\n", __func__, ret);
        return;
    }
    ret = esp_ble_gattc_app_register(GATTC_PROFILE_C_APP_ID);
    if (ret){
        ESP_LOGE(TAG, "%s gattc app register failed, error code = %x\n", __func__, ret);
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }

When connecting to the server, the client's callback is triggered, with a connection event.

I (15612) SERVER: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 6b:18:c1:36:87:44:
I (15613) SERVER: CONNECT_EVT, conn_id 0, remote 6b:18:c1:36:87:44:
I (15617) BLE_CLIENT: ESP_GATTC_CONNECT_EVT conn_id 0, if 5
I (15623) BLE_CLIENT: REMOTE BDA:
I (15627) BLE_CLIENT: 6b 18 c1 36 87 44
I (15760) SERVER: ESP_GATTS_MTU_EVT, MTU 500
I (15761) SERVER: ESP_GATTS_MTU_EVT, MTU 500
I (16029) SERVER: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
I (16585) SERVER: GATT_READ_EVT, conn_id 0, trans_id 2, handle 46
I (16675) SERVER: GATT_WRITE_EVT, conn_id 0, trans_id 3, handle 44
I (16675) SERVER: GATT_WRITE_EVT, value: hello bro!!!, len 12, value :
I (16679) SERVER: 68 65 6c 6c 6f 20 62 72 6f 21 21 21
W (16885) BT_APPL: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0013
I (16887) SERVER: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13
I (16890) BLE_CLIENT: ESP_GATTC_DISCONNECT_EVT, reason = 19

xiewenxiang, 2021-09-02:
When a connection event is reported, the parameter `param->connect.link_role` in gattc and gatts event handler callback function can be used to determine which side the connection initiated.

If the link_role equal 0, it means ESP32 send CONNECT_IND, it's link layer role is master.
If the link_role equal 1, it means ESP32 send ADV and received CONNECT_IND from remote device, it's link layer role is slave.

hard-cat, 2022-04-27:
exactly what is needed!