espressif/esp-idf #7302: Example for UDP broadcast receive

silardg, 2021-07-22:
I have scrambled the internet for UDP broadcast examples and I have this code for now and it's not working.
Any suggestions?

```

static void udp_client_task(void *pvParameters)
{
    char rx_buffer[128];

    while (1) {
        struct sockaddr_in dest_addr;
        dest_addr.sin_addr.s_addr = INADDR_BROADCAST;
        dest_addr.sin_family = AF_INET;
        dest_addr.sin_port = htons(3333);
        int addr_family = AF_INET;
        int ip_protocol = IPPROTO_UDP;

        int sock = socket(addr_family, SOCK_DGRAM, ip_protocol);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            break;
        }
        //ESP_LOGI(TAG, "Socket created");

        int err = bind(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
        if (err < 0) {
            ESP_LOGE(TAG, "Socket unable to bind: err %d", err);
        }
        int bc = 1;
        if (setsockopt(sock, SOL_SOCKET, SO_BROADCAST, &bc, sizeof(bc)) < 0) {
            ESP_LOGE(TAG, "Failed to set sock options: errno %d", errno);
            closesocket(sock);
            break;
        }

        while (1) {

            // int err = sendto(sock, "test", strlen("test"), 0, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
            // if (err < 0) {
            //     ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
            //     break;
            // }
            // ESP_LOGI(TAG, "Message sent");

            struct sockaddr_storage source_addr; // Large enough for both IPv4 or IPv6
            socklen_t socklen = sizeof(source_addr);
            int len = recvfrom(sock, rx_buffer, sizeof(rx_buffer) - 1, 0, (struct sockaddr *)&source_addr, &socklen);

            // Error occurred during receiving
            if (len < 0) {
                ESP_LOGE(TAG, "recvfrom failed: errno %d", errno);
                break;
            }
            // Data received
            else {
                rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string
                ESP_LOGI(TAG, "Received %d bytes", len);
                ESP_LOGI(TAG, "%s", rx_buffer);
                if (strncmp(rx_buffer, "OK: ", 4) == 0) {
                    ESP_LOGI(TAG, "Received expected message, reconnecting");
                    break;
                }
            }

            vTaskDelay(1000 / portTICK_PERIOD_MS);
        }

        if (sock != -1) {
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
            shutdown(sock, 0);
            close(sock);
        }
    }
    vTaskDelete(NULL);
}
```

felmue, 2021-07-22:
Hello @silardg 

from my understanding there is nothing special in the **receiver** part to be able to receive broadcast packets.

I am running the [UDP Server example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_server).

It reacts on UDP packets sent to its IP address (unicast) or to the broadcast address, e.g. last octet 255 or even when I send the packets to 255.255.255.255.

Thanks
Felix

ESP-YJM, 2021-07-23:
@silardg I think it is incorrect to bind address to `INADDR_BROADCAST`(`dest_addr.sin_addr.s_addr = INADDR_BROADCAST `). The bind operation is to bind socket to local address and port, which is your output or input local interface. You can use `dest_addr.sin_addr.s_addr = htonl(INADDR_ANY)` and try.

ESP-YJM, 2021-07-23:
@silardg I think it is incorrect to bind address to `INADDR_BROADCAST`(`dest_addr.sin_addr.s_addr = INADDR_BROADCAST `). The bind operation is to bind socket to local address and port, which is your output or input local interface. You can use `dest_addr.sin_addr.s_addr = htonl(INADDR_ANY)` or last octet 255 and try.

silardg, 2021-07-23:
Thanks guys for the recommendation. My issue was that I had multiple routers and mikrotiks connected and I guess it didn't have a connection, even when I was connected to the same network as my esp32.

Closing ☺️😁

closed 2021-07-23 by silardg