espressif/esp-idf #7303: SPP support with NimBLE

venkateshgr-ayla, 2021-07-22:
Hi All,
I wanted to know if SPP can be supported with NimBLE? Currently I am not using Bluedroid. I want to port SPP-Bluedroid to SPP-NimBLE.
Is it technically possible and recommended? 
Does ESP have any plans to support SPP over NimBLE?
Kindly help me with your suggestions on SPP feature with NimBLE.

Thanks,
GR.Venkatesh

RahulEspressif, 2021-07-22:
Hi @venkateshgr-ayla ,  

 SPP ( Serial Port profile ) is a basically a bluetooth classic profile. Nimble, is BLE only stack. 

 We can however achieve the same by implementing a custom LE service running on top of GATT and simulate the behaviour and achieve SPP functionality in BLE. 

Currently, there is no such custom service implemented for Nimble. We will check internally and plan if we can release this as part of the SDK. 

Thanks,
 Rahul

venkateshgr-ayla, 2021-07-23:
Hi Rahul,

Thanks for the response. Is it possible for this feature release? If so can I know approximate timeline?

Thanks,
GR.Venkatesh

rahult-github, 2021-08-30:
Hi @venkateshgr-ayla ,

 We are planning on including this feature and are currently  targeting to include this example in stable release in Q4' 21  quarter. 

Thanks,
 Rahul

IshaESP, 2021-10-14:
Hi @venkateshgr-ayla,
 
Can you try this patch?
[nimble_spp_patch.txt](https://github.com/espressif/esp-idf/files/7345808/nimble_spp_patch.txt)

Thanks,
Isha

closed 2021-11-30 by espressif-bot