espressif/esp-idf #7306: undefined reference to mbedtls_chachapoly_init #3426

ln-12, 2021-07-22:
I am referring  to [this issue](https://github.com/espressif/arduino-esp32/issues/3426) which lists all details. The main problem is the linker error `undefined reference to mbedtls_chachapoly_init`. In [this comment]( https://github.com/espressif/arduino-esp32/issues/3426#issuecomment-547918746) a maintainer suggested asking for help in this repo.

## Environment

- Module or chip used: ESP32-WROOM-32
- Build System: Arduino IDE with [esp32 boards](https://github.com/espressif/arduino-esp32)
- Operating System: macOS
- Using an IDE?: Yes, Arduino IDE
- Power Supply: USB

## Problem Description

The build fails because of the linker error `undefined reference to mbedtls_chachapoly_init`.

### Expected Behavior

No linker error.

### Actual Behavior

A linker error.

### Steps to reproduce

Compile the script below.

### Code to reproduce this issue

I want to test chacha using the following script from the mentioned issue:
```
#include "mbedtls/chachapoly.h"

void setup()
{
	uint8_t iv[12] =		{ 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, 0x2B, \
							  0x38, 0xE6, 0x9A, 0x83 };
	uint8_t key[32] =		{ 0xB3, 0xD9, 0x1B, 0xF8, 0xA2, 0x3F, 0xA7, 0xEB, \
							  0x3E, 0x26, 0xD5, 0x34, 0x40, 0xE9, 0x0A, 0x12, \
							  0x73, 0x03, 0x0C, 0x6E, 0xB8, 0x69, 0xA0, 0x7B, \
							  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
	uint8_t aad[21] =		{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0x04, 0x65, 0x0E, \
							  0xDD, 0x94, 0x97, 0xB3, 0x84 };
	uint8_t message[49]=	{ 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xFC, 0x6F, 0x3E, \
							  0xEE, 0x04, 0x5B, 0x35, 0xDE, 0x04, 0xF2, 0xCE, \
							  0x45, 0x31, 0x37, 0x26, 0xD9, 0xEB, 0xAA, 0x32, \
							  0x6B, 0x36, 0x9C, 0x78, 0x82, 0xCD, 0xD6, 0x71, \
							  0xFE, 0x79, 0xB9, 0xDA, 0x23, 0xF9, 0x8F, 0x77, \
							  0x96 };
	uint8_t tag[16] =			{ 0x74, 0xBF, 0x73, 0x37, 0x27, 0x58, 0xF8, 0x9E, \
							  0xD2, 0xF3, 0x84, 0xB6, 0x4B, 0xC5, 0xE9, 0x04 };
	uint8_t cryptMsg[49] = { 0xFF, 0x99, 0x90, 0xAA, 0xF8, 0xEC, 0xEC, 0xB1, \
							  0x2B, 0x38, 0xE6, 0x9A, 0x83, 0xA0, 0x0F, 0xCE, \
							  0xE0, 0x2A, 0xA0, 0xD0, 0x7C, 0x98, 0x79, 0x59, \
							  0xA2, 0x1E, 0x25, 0xC6, 0x53, 0x7D, 0xE7, 0x44, \
							  0x82, 0x62, 0x74, 0x32, 0x3E, 0xF4, 0xF3, 0x52, \
							  0xB3, 0x8A, 0xFA, 0x7F, 0x93, 0x3A, 0xB5, 0xAD, \
							  0x9E };
	uint8_t buffer[49];
	uint8_t tagBuffer[16];
	mbedtls_chachapoly_context ctx;

	Serial.begin (115200);

	mbedtls_chachapoly_init (&ctx);
	if (mbedtls_chachapoly_starts (&ctx, iv, MBEDTLS_CHACHAPOLY_ENCRYPT)) {
		Serial.printf ("Start error");
	}
	mbedtls_chachapoly_setkey (&ctx, key);
	if (mbedtls_chachapoly_encrypt_and_tag (&ctx, sizeof (message), iv, aad, sizeof (aad), message, buffer, tagBuffer)) {
		Serial.printf ("Encrypt error");
	}
	mbedtls_chachapoly_free (&ctx);

	Serial.print ("Output: ");
	for (int i = 0; i < sizeof (message); i++) {
		Serial.printf ("%02x ", buffer[i]);
	}
	Serial.print ("Tag: ");
	for (int i = 0; i < sizeof (tag); i++) {
		Serial.printf ("%02x ", tagBuffer[i]);
	}
}

void loop(){}
```

## Debug Logs

```
sketch/crypto.ino.cpp.o:(.literal._Z5setupv+0x14): undefined reference to `mbedtls_chachapoly_init'
sketch/crypto.ino.cpp.o: In function `setup()':
/Users/lneumann/ownCloud/Cloud/Bluetooth/crypto/crypto.ino:36: undefined reference to `mbedtls_chachapoly_init'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

atanisoft, 2021-07-22:
@ln-12 did you enable CONFIG_MBEDTLS_CHACHAPOLY_C in sdkconfig in your ESP-IDF example? Arduino-esp32 does not enable this by default.

ln-12, 2021-07-22:
Hello @atanisoft, thanks for your fast answer. Do you mean to edit the file under `/Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/sdkconfig`?  Adding `CONFIG_MBEDTLS_CHACHAPOLY_C=y` there doesn't help. 

Also after adding `#define MBEDTLS_CHACHAPOLY_C` to `/Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config/sdkconfig.h` the error remains. 

Or do I have to setup the `sdkconfig` file in my arduino project directory?

atanisoft, 2021-07-22:
@ln-12 Editing sdkconfig.h will have *ZERO* effect unless you also rebuild the pre-built binaries that the arduino-esp32 packages provide. In your issue filed against arduino-esp32 the need to recompile the precompiled binaries was mentioned but perhaps not clearly explained.

ln-12, 2021-07-23:
Thanks for the hint, I thought this would happen automatically . I used [this how to](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html) for the building process. Afterwards, I copied `libmbedcrypto.a`, `libmbedtls.a` and `libmbedx509.a` from `<project directory>/esp32-arduino-lib-builder/out/tools/sdk/esp32s2/lib` into `/Users/user/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib`, but the error remains. Using `./tools/copy-to-arduino.sh` does also not help. Do you have any idea what I am missing?

atanisoft, 2021-07-23:
That guide should have worked as is.