espressif/esp-idf #7308: assertion "pxQueue->uxItemSize == 0" failed: file "../esp-idf/components/freertos/queue.c", line 1445

vanderareosa, 2021-07-22:
----------------------------- Delete below -----------------------------

## Environment
Mac IOS 10.13
IDE: Arduino 1.8
BLE Libs:
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>
Development Kit: ESP32-C3-DevKitM-1

I'm using BLE to send a message to esp32-c3, when the message arrives, it is sent to a method that will save the contents to a file using SPIFFS. This procedure is working 100%, but when I finish where there is no longer any activity to be performed, I am getting the error:
assertion "pxQueue->uxItemSize == 0" failed: file "../esp-idf/components/freertos/queue.c", line 1445, function: xQueueSemaphoreTake

and after that esp32-c3 is restarted.

KaeLL, 2021-07-22:
I had that problem happen to me a few times due to buffer overflows corrupting the mutex.

vanderareosa, 2021-07-22:
what did you do to solve?

KaeLL, 2021-07-23:
Fixed the memory corruption caused by writing past the end of a global array.

Dazza0, 2021-09-29:
@vanderareosa Were you able to resolve your issue?

Alvin1Zhang, 2021-11-23:
@vanderareosa Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Dazza0, 2022-03-17:
@vanderareosa This issue is likely caused by memory corruption (e.g., due to a stack overflow). I'm closing this issue as there's not much actionable information we can go on. Please feel free to reopen this issue if the error occurs again.

Here are some things you could also try.
- Increase your task stack sizes and see if the issue goes away
- Check the addresses of the semaphore, your task stack's start and end addresses, and the `uxTaskGetStackHighWaterMark()` of each task, to see if any of them overlap.

closed 2022-03-17 by Dazza0