espressif/esp-idf #7309: ULP+Touch

yaroshd81, 2021-07-22:
Good day!
I want use ulp for read "touch" values. I know about `esp_sleep_enable_touchpad_wakeup();`, but i don't want wakeup if i have only 1 condition. Can you help me: how read touch from ulp (sorry, my english is bad: i learn germany :) )

ginkgm, 2021-07-23:
Hi @yaroshd81 , which chip are you using?

yaroshd81, 2021-07-23:
Sorry, ESP32 WROOM-32

ESP-Marius, 2024-11-22:
For anyone looking to achieve there is an external example [here](https://github.com/boarchuz/HULP/blob/master/src/hulp_touch.h), while for the ulp-riscv we have an example [here](https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/touch). 

We are probably unlikely to add any more examples for the old ULP-FSM at this point

closed 2024-11-22 by ESP-Marius