espressif/esp-idf #7310: BlueFi library esp_ble_gap_set_device_name can't accept more than 26 characters

raoulshih, 2021-07-23:
## Environment

- Module or chip used:  [ESP32-S by AI Thinker]
- IDF version (run ``git describe --tags`` to find it):
v3.3.1
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp32-2018r1) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [vscode]
- Power Supply:         [USB]

## Problem Description

I used BlueFi to create my own BLE to WiFi Setting, when I try to set the name of the ESP32 Device using esp_ble_gap_set_device_name, I got incorrect results.  

### Expected Behavior

I intend to set a 29-characters name for the device, expected to see 29 characters using the EspBluFi utility. 

### Actual Behavior

On the EspBluFi utility, I only see 26 characcters.

### Code to reproduce this issue

I use this to set the name of ESP32,
```cpp
BLUFI_DEVICE_NAME = "123456789012345678901234567890";
esp_ble_gap_set_device_name(BLUFI_DEVICE_NAME);
```
But I see only 26 characters on the EspBluFi utility "12345678901234567890123456"
[image]

xulongzhuang, 2021-08-16:
Hi，@raoulshih , 
I think the reason may be this: first, the length of adv_data is 31 bytes, and it contains multiple AD structures. When the AD_struture of device_name is too long, it will make the AD_struture arranged after it not displayed, and it will also The display is not complete due to length limitation, I think this is normal. The attachment is a device_name with 28 bytes, you can see that his original adv_data is only 26 bytes 
[image]

Alvin1Zhang, 2021-11-23:
@raoulshih Thanks for reporting, sorry for slow turnaround, would you please help share if any updates for the issue? Thanks.

raoulshih, 2021-12-02:
@Alvin1Zhang I just stuck with 26 characters, and it worked fine.

closed 2021-12-02 by raoulshih

Alvin1Zhang, 2021-12-02:
@raoulshih Thanks for sharing the updates.